MarlinFirmware/Marlin #14298: [BUG] - Possible Bug ? TMC2208 256 Micro Steps - ABL ONLY SMALL STEPS

SISLANGER, 2019-06-16:
### Description

ABL WORKS FINE UNTIL I TRY AND INCREASE MICROSTEPS ABOVE 16
I have set the correct steps and homing and moving around the bed works fine. However when I try and run G29. It like starts ABL about an 8th of the size.

### Steps to Reproduce

1) 
#if HAS_TRINAMIC

  #define HOLD_MULTIPLIER    0.5  // Scales down the holding current from run current
  #define INTERPOLATE       true  // Interpolate X/Y/Z_MICROSTEPS to 256

  #if AXIS_IS_TMC(X)
    #define X_CURRENT     950  // (mA) RMS current. Multiply by 1.414 for peak current.
    #define X_MICROSTEPS   256  // 0..256
    #define X_RSENSE     0.11
  #endif
and so on

2)
#define DEFAULT_AXIS_STEPS_PER_UNIT { 1611.68, 1614.08, 6359.68, 6588.48, 1533.76 }

3)
G28
G29

**Expected behavior:** [Normal ABL]

**Actual behavior:** [Micro ABL]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

gloomyandy, 2019-06-16:
It probably does not make sense to run with 256 microsteps on typical 3D printers. When moving at say 100mm/s (not that high a speed) you will need to generate 161400+ pulses per second which although perhaps possible with a 32bit cpu is probably not a good idea. Your distance per step 1/1614mm is way smaller than the mechanical precision of probably any printer out there. Also the large number of steps per mm (6000+) may be exposing issues with Marlin like overflow of variables when computing large moves etc. In addition you will almost certainly get smoother motion by allowing the driver to generate the extra steps to take say 16 microsteps up to 256 (using interpolation), than by asking Marlin to generate all of those steps. Do you have a reason for using these high step rates? If you really want to use higher microstep values I've seen reports of 64 and possibly 128 working, but they also tend to report poorer print quality as a result. 

Note I'm not saying that there are not be bugs in Marlin in this area, just that even if they get fixed there will probably not be any benefit from the results. There are probably more important things to sort out at the moment.

SISLANGER, 2019-06-16:
Reason using because I can, wanted to see the effects good and bad. 

There is no point just listening to people saying there is no point in doing this if you can’t learn this for yourself.

Not a moan just a reason.

SISLANGER, 2019-06-16:
The problem happens on 32 micro steps and above

gloomyandy, 2019-06-16:
I assume when you changed the microsteps you also changed the DEFAULT_AXIS_STEPS_PER_UNIT? Did you also remember to load firmware defaults and store them in eeprom (M502 followed by M500)?

SISLANGER, 2019-06-16:
@gloomyandy Yeah steps as in original post.Loaded & Stored Movement seems normal with the exception of ABL

SISLANGER, 2019-06-16:
Its not a major issue, just wanted to learn the effects first hand, but there is something odd going on with it. Even happens when set to 32 m steps

gloomyandy, 2019-06-16:
The steps in your original post are for 256 microsteps, if you want to use 32 microsteps you will need to divide those numbers by (256/32) so 8.

SISLANGER, 2019-06-16:
Yeah I know it was adjusted when tried 32 micro steps above is the example when trying 256

reepotah, 2019-06-19:
Which motherboard do you have? Maybe 1/32+ is already too much for your MCU.

I also just want to share my experience here:
1. Even at 128 microsteps TMC2208 + Ender 3 Pro stock steppers start skipping on travel moves _violently_, like 5cm in random direction, @ > 125mm/sec. Hybrid mode.
2. There was some bug, when you try to use Linear advance with TMCs, and they stop moving at all cause pulse width/frequency is now for whatever reason too much for them to handle. Was temp-fixed with `#define MINIMUM_STEPPER_PULSE 1`.

Someone said that at 128microsteps there's like only 1% of motor holding torque left.

I have an SKR1.3, and I settled with 1/32, with hybrid mode on every axis except E0. Nice, smooth and quiet.

SISLANGER, 2019-06-20:
@reepotah I’m using SKR SKR V1.3 tmc2208

SISLANGER, 2019-06-20:
I’m getting a lot of deep vibrations with the tmc2208s that I can’t narrow down at moment

AnHardt, 2019-06-20:
> Someone said that at 128microsteps there's like only 1% of motor holding torque left.

They are right - but that does not increase the chance for skipping steps.
The 1% could be about right for the angular difference of **one microstep**.
To get the full torque you have to displace rotor and field by one full step. 
1 full step / 128 microsteps is about 1%. 
But the stepper will only skip is the displacement is larger than **one full step** = 128 microsteps.

Those who tell you about the decreased torque are right. but this is not applikable when talking about lost steps.

[How do steppers work (some rarely described facts)](https://github.com/AnHardt/Marlin/issues/83)

boelle, 2019-07-21:
so its it wrong to say that is a user error more than a bug?

would his config files help us much?

boelle, 2019-09-24:
@SISLANGER is the issue still there?

SISLANGER, 2019-09-24:
Been working on another setup for a while not sure if fixed or if still an issue ? I will be going back to it when get a chance. 

Which doesn't help anyone so I am really and truly sorry.

boelle, 2019-10-21:
@SISLANGER had a chance to revisit this one? if not would you mind if we closed this one and you open a new one when and if you get time to play with it?

SISLANGER, 2019-10-21:
@boelle sadly not had time please close, and sorry to have been a pain. Thank you to all those who tried to help.

closed 2019-10-21 by SISLANGER

boelle, 2019-10-21:
hehe no pain, your case might be the famous needle in the haystack

so please if you get time to test let us know so we can reopen it

SISLANGER, 2019-10-21:
@boelle Thank you