MarlinFirmware/Marlin #14300: SKR 1.3 looses all communication after emergency stop in S3D

robnrie, 2019-06-16:
After calling an emergency stop command from S3D the SKR 1.3 board looses all communication and requires a reboot.

1. Set a homing command
2. Stop the homing with the emergency stop
3. Try to reconnect through S3D

It should reconnect and establish communication

Instead it requires a re powering of the board

Ludy87, 2019-06-16:
The error lies with S3D and not with Marlin!

thisiskeithb, 2019-06-16:
Doesn’t S3D issue an `M112` command for emergency stop? If so, [This is intentional](marlinfw.org/docs/gcode/M112.html).

robnrie, 2019-06-16:
On my other boards MKS Gen 1.4 and so on or my any cubic Chiron all with marlin 1.1.9 if the emergency stop is used you have the ability to reconnect from within S3d without having to power down and re power the board.
But with this board and marlin 2.0 bugfix you are required to power the board down first then reboot then reconnect and in the meantime if you have used the emergency because the nozzle has crashed into the bed then this takes too much time to get control again....

gloomyandy, 2019-06-16:
With Arduino based boards the USB serial port interface (which you will be using to talk to the board) is an external chip and a side effect of opening/closing the port is that it will cause the Arduino to perform a reset. On many/most 32bit based boards the USB serial interface is provided by software built into the main processor. So when you tell the processor to emergency stop you lose the USB connection and need to perform a hardware reset (as there is no external chip sitting there to do it for you). The only alternative would be to keep the USB code running after the M112 but this is potentially dangerous as allowing this code to run requires that interrupts and other code is running which could result in heaters etc. being activated. A further option would be to force a chip reset, but again this is potentially dangerous in some situations.

 I doubt if the way this is will get changed without a lot of discussion and thought. Almost certainly one for @thinkyhead

redone00, 2019-08-11:
hi get the same error in repetier and in both runs for quite some time before halting

thisiskeithb, 2019-08-11:
> hi get the same error in repetier and in both runs for quite some time before halting

Repetier Host also sends `M112` for emergency stop, so [this is intentional](http://marlinfw.org/docs/gcode/M112.html):
```
Send emergency stop to printer. You may need to reset the printer for a restart!
N46 M112 *51
```

closed 2019-08-16 by shitcreek