MarlinFirmware/Marlin #14303: Can not Compile (#14287)

Udo-Nbg, 2019-06-16:
# NO SUPPORT REQUESTS PLEASE

Support Requests posted here will be automatically closed!

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. See https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question.

For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org Marlin Forum http://forums.reprap.org/list.php?415
- Tom's 3D Forums https://discuss.toms3d.org/
- Facebook Group "Marlin Firmware" https://www.facebook.com/groups/1049718498464482/
- Facebook Group "Marlin Firmware for 3D Printers" https://www.facebook.com/groups/3Dtechtalk/
- Marlin Configuration https://www.youtube.com/results?search_query=marlin+configuration on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a Bug Report at https://github.com/MarlinFirmware/Marlin/issues/new/choose).

Udo-Nbg, 2019-06-16:
Fix SPI, SD for BIGTREETECH SKR Mini (#14287) 
will not compile,
Attaches Files: 
   Errors
   platformio.ini
   configurations

mostly default configurations...

[Marlin2_SKR_mini_Errors_Build_#14287.zip](https://github.com/MarlinFirmware/Marlin/files/3294149/Marlin2_SKR_mini_Errors_Build_.14287.zip)

3DSmitty, 2019-06-16:
I just compiled from scratch using minimal config with no issues.

Udo-Nbg, 2019-06-17:
Sorry no...
tried with ATOM & VSCODE but no complete compile.

I use exact the commds u suggest to me, but
Some type errors appear...

`In file included from C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/boards.h:38:0,
from C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:54,
from C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
from C:\users\udo\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_common.c:46:
C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish_types.h:65:9: error: unknown
type name 'bool'
typedef bool boolean;
^~~~
In file included from C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:54:0,
from C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
from C:\users\udo\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_common.c:46:
C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/boards.h:111:1: error: unknown type
name 'bool'; did you mean '_Bool'?
bool boardUsesPin(uint8 pin);
^~~~
_Bool
In file included from C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:58:0,
from C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
from C:\users\udo\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_common.c:46:
C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/ext_interrupts.h:90:6: error: conflicting types for 'attachInterrupt'
void attachInterrupt(uint8 pin, voidArgumentFuncPtr handler, void *arg,
^~~~~~~~~~~~~~~
C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/ext_interrupts.h:69:6: note: previous declaration of 'attachInterrupt' was here
void attachInterrupt(uint8 pin, voidFuncPtr handler, ExtIntTriggerMode mode);
^~~~~~~~~~~~~~~
In file included from C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:60:0,
from C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
from C:\users\udo\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_common.c:46:
C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish_math.h:50:6: error: conflicting types for 'random'
long random(long max);
^~~~~~
In file included from c:\users\udo\.platformio\packages\toolchain-gccarmnoneeabi@1.70201.0\arm-none-eabi\include\stdlib.h:11:0,
from C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:41,
from C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
from C:\users\udo\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_common.c:46:
c:\users\udo\.platformio\packages\toolchain-gccarmnoneeabi@1.70201.0\arm-none-eabi\include\stdlib.h:255:6: note: previous declaration of 'random' was here
long _EXFUN(random,(_VOID));
^
In file included from C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:60:0,
from C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
from C:\users\udo\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_common.c:46:
C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish_math.h:59:6: error: conflicting types for 'random'
long random(long min, long max);
^~~~~~
In file included from c:\users\udo\.platformio\packages\toolchain-gccarmnoneeabi@1.70201.0\arm-none-eabi\include\stdlib.h:11:0,
from C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:41,
from C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
from C:\users\udo\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_common.c:46:
c:\users\udo\.platformio\packages\toolchain-gccarmnoneeabi@1.70201.0\arm-none-eabi\include\stdlib.h:255:6: note: previous declaration of 'random' was here
long _EXFUN(random,(_VOID));
^
compilation terminated due to -fmax-errors=5.
*** [.pioenvs\BIGTREE_SKR_MINI\libc77\U8glib_ID7\clib\u8g_com_arduino_common.c.o] Error 1
In file included from C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/boards.h:38:0,
from C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:54,
from C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
from C:\users\udo\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_fast_parallel.c:69:
C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish_types.h:65:9: error: unknown
type name 'bool'
typedef bool boolean;
^~~~
In file included from C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:54:0,
from C:\users\udo\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
from C:\users\udo\.platformio\lib\U8glib_ID7\src\clib\u8g_com_arduino_fast_parallel.c:69:
[... 156 lines cut ...]
Environment DUE_USB                     [SKIP]
Environment DUE_debug                   [SKIP]
Environment LPC1768                     [SKIP]
Environment LPC1769                     [SKIP]
[... cut ...]
Environment black_stm32f407ve           [SKIP]
Environment teensy35                    [SKIP]
Environment malyanm200                  [SKIP]
Environment esp32                       [SKIP]
Environment fysetc_f6_13                [SKIP]
Environment linux_native                [SKIP]
============================================== [ERROR] Took 81.02 seconds ==============================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.`

RudyBenner, 2019-06-19:
No changes to files. Compile failed. Updated project libraries. Success.

boelle, 2019-07-21:
@Udo-Nbg can you confirm that updating project libs sorts the issue?

Udo-Nbg, 2019-07-21:
Sorry noc succsess,
using VS-Code and ATOM.
PlatformIO and Updates are updated.
VS-Code uninstalled automatically SintelliSense colliding Plugins/Libs. 
but no Success
also Atom is up to date (i need tor reinstall it)

I think there is somthing totally wrong.

I try to install to an other PC and report later again

Udo-Nbg, 2019-07-22:
@boelle 
(mion)
i can NOT compile the Firmware for SKR-MINI
but i can compile compile tje Firmware for SKR-PRO (new)
and now i'm missing SKR-1.3 entries (maybe that will be my next borard)

RudyBenner, 2019-07-22:
Can you compile the basic default sketch for RAMPS 1.4?

boelle, 2019-07-22:
as @RudyBenner says try with the standard setup for ramps 1.4, if that works i would investigate if all libs are there and if they are fully updated

Udo-Nbg, 2019-07-22:
OKAY her are my results
I use a fresh copy (unzip) of Marlin-bugfix-2.0.x_branch_4262
Absolute noch Chages on the source

   Arduino 1.8.9 Compile Succsess
      #define MOTHERBOARD BOARD_RAMPS_14_EFB

   VS-Code 1.36.1
      env_default  = megaatmega2560
      #define MOTHERBOARD BOARD_RAMPS_14_EFB

Warning! `env_default` configuration option in section [platformio] is deprecated and will be removed in the next release! Please use `default_envs` instead

Seriously **after the 3rd attempt** it will run straight through, i did not made any changes, nowhere.
It took quite a while, but **it compiled without any more warnings or errors**

Here are my results after the 4th Compile:

`> Executing task: C:\Users\Udo\.platformio\penv\Scripts\platformio.exe run <

Warning! `env_default` configuration option in section [platformio] is deprecated and will be removed in the next release! Please use `default_envs` instead
Processing megaatmega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR 1.14.0 > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
PACKAGES: toolchain-atmelavr 1.50400.190710 (5.4.0), framework-arduinoavr 4.0.0
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 16 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|-- <LiquidCrystal> 1.3.4
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.4.5
|   |-- <SoftwareSerial> 1.0
|   |-- <SPI> 1.0
|-- <Adafruit NeoPixel> 1.1.3
|-- <30aa480>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <c1921b4>
|   |-- <SPI> 1.0
|-- <SailfishLCD> #6f53c19
|-- <SailfishRGB_LED> #4d0c888
|   |-- <Wire> 1.0
|-- <SlowSoftI2CMaster> #3a18be5
|-- <Wire> 1.0
|-- <SPI> 1.0
Checking size .pioenvs\megaatmega2560\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [===       ]  30.2% (used 2478 bytes from 8192 bytes)
PROGRAM: [==        ]  21.0% (used 53248 bytes from 253952 bytes)
============================================= [SUCCESS] Took 11.32 seconds =============================================

====================================================== [SUMMARY] ======================================================
Environment megaatmega2560              [SUCCESS]
Environment megaatmega1280              [IGNORED]
Environment at90usb1286_cdc             [IGNORED]
Environment at90usb1286_dfu             [IGNORED]
Environment DUE                         [IGNORED]
Environment DUE_USB                     [IGNORED]
Environment DUE_debug                   [IGNORED]
Environment LPC1768                     [IGNORED]
Environment LPC1769                     [IGNORED]
Environment melzi                       [IGNORED]
Environment melzi_optiboot              [IGNORED]
Environment rambo                       [IGNORED]
Environment sanguino_atmega644p         [IGNORED]
Environment sanguino_atmega1284p        [IGNORED]
Environment STM32F1                     [IGNORED]
Environment fysetc_STM32F1              [IGNORED]
Environment BIGTREE_SKR_MINI            [IGNORED]
Environment STM32F4                     [IGNORED]
Environment STM32F7                     [IGNORED]
Environment ARMED                       [IGNORED]
Environment alfawise_U20                [IGNORED]
Environment mks_robin                   [IGNORED]
Environment mks_robin_mini              [IGNORED]
Environment mks_robin_nano              [IGNORED]
Environment JGAURORA_A5S_A1             [IGNORED]
Environment black_stm32f407ve           [IGNORED]
Environment BIGTREE_SKR_PRO             [IGNORED]
Environment teensy35                    [IGNORED]
Environment malyanm200                  [IGNORED]
Environment esp32                       [IGNORED]
Environment fysetc_f6_13                [IGNORED]
Environment linux_native                [IGNORED]
============================================= 1 succeeded in 11.34 seconds =============================================`

boelle, 2019-07-22:
Warning! is a warning, it does not mean fail

if you look close you got a success: [SUCCESS] Took 11.32 seconds

Udo-Nbg, 2019-07-22:
yes after the 1st complete compile (no cleanup) and then again compile.

but that is noch my reason.

i want to compile fo a SKR mini V1.1 and ther are some errors.

I try it again rigt now.

Udo-Nbg, 2019-07-22:
Her are me Settings to try a compile for SKR mini V 1.1
no other chages are made:

platformio.ini
```
[platformio]
src_dir      = Marlin
build_dir    = .pioenvs
lib_dir      = .piolib
libdeps_dir  = .piolibdeps
boards_dir   = buildroot/share/PlatformIO/boards
env_default  = BIGTREE_SKR_MINI
```

Configuration.h  CHANGES:
```
#ifndef MOTHERBOARD
  #define BOARD_BIGTREE_SKR_MINI_V1_1
#endif

```
```
> Executing task: C:\Users\Udo\.platformio\penv\Scripts\platformio.exe run <

Warning! `env_default` configuration option in section [platformio] is deprecated and will be removed in the next release! Please use `default_envs` instead
Processing BIGTREE_SKR_MINI (platform: ststm32; board: genericSTM32F103RC; framework: arduino)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 5.5.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190719 (1.0.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing c1921b4
Using cache: C:\Users\Udo\.platformio\.cache\8c\f244bfa38b7bfff5e9042f5d9f9f5c8c
Unpacking  [###########################---------]   77%  00:00:03
Unpacking...
IOError: [Errno 2] No such file or directory: 'd:\\___Marlin2\\Marlin-bugfix-2.0.x_branch_4262\\Basic_test_nochanges\\.piolibdeps\\BIGTREE_SKR_MINI\\_tmp_installing-5fyx8o-package\\TMC26XStepper-c1921b4856a5ef4b659b661b8cb5dd56f3075e57\\examples\\TMC26XMotorTester\\processing\\TMC26XMotorTest\\ChopperConfiguration.pde':
  File "C:\users\udo\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 126:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Udo\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Udo\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "C:\users\udo\.platformio\platforms\ststm32\builder\main.py", line 91:
    target_elf = env.BuildProgram()
  File "C:\Users\Udo\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\udo\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 124:
    _build_project_deps(env)
  File "C:\users\udo\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 49:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Udo\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\udo\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1043:
    project.install_dependencies()
  File "C:\users\udo\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 878:
    lm.install(uri)
  File "c:\users\udo\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 353:
    force=force)
  File "c:\users\udo\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 724:
    track=True)
  File "c:\users\udo\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 506:
    self.unpack(dlpath, tmp_dir)
  File "c:\users\udo\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 219:
    return fu.unpack(dest_dir, with_progress=False)
  File "c:\users\udo\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 118:
[... 22 lines cut ...]
Environment sanguino_atmega644p         [IGNORED]
Environment sanguino_atmega1284p        [IGNORED]
Environment STM32F1                     [IGNORED]
Environment fysetc_STM32F1              [IGNORED]
Environment BIGTREE_SKR_MINI            [FAILED]
Environment STM32F4                     [IGNORED]
Environment STM32F7                     [IGNORED]
Environment ARMED                       [IGNORED]
Environment alfawise_U20                [IGNORED]
Environment mks_robin                   [IGNORED]
Environment mks_robin_mini              [IGNORED]
Environment mks_robin_nano              [IGNORED]
Environment JGAURORA_A5S_A1             [IGNORED]
Environment black_stm32f407ve           [IGNORED]
Environment BIGTREE_SKR_PRO             [IGNORED]
Environment teensy35                    [IGNORED]
Environment malyanm200                  [IGNORED]
Environment esp32                       [IGNORED]
Environment fysetc_f6_13                [IGNORED]
Environment linux_native                [IGNORED]
======================================== 1 failed, 0 succeeded in 5.38 seconds ========================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

RudyBenner, 2019-07-22:
Take a close look, exact syntax is expected.

[image]

RudyBenner, 2019-07-22:
Warnings (as you know) are yellow, Errors are red. Generally, you can ignore the warnings, the first one(s) may be of interest. Errors are bad, generally, the first one is the one to dig into.

Always keep libraries updated. There are more drastic 'cures' when things go totally south.

Udo-Nbg, 2019-07-22:
Ooops, but this was not the issue
i tried again with the correct syntax (hopefully)
```
// Choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_BIGTREE_SKR_MINI_V1_1
#endif

```

```
> Executing task: C:\Users\Udo\.platformio\penv\Scripts\platformio.exe run <

Warning! `env_default` configuration option in section [platformio] is deprecated and will be removed in the next release! Please use `default_envs` instead
Processing BIGTREE_SKR_MINI (platform: ststm32; board: genericSTM32F103RC; framework: arduino)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 5.5.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190719 (1.0.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing c1921b4
Using cache: C:\Users\Udo\.platformio\.cache\8c\f244bfa38b7bfff5e9042f5d9f9f5c8c
Unpacking  [###########################---------]   77%
Unpacking...
IOError: [Errno 2] No such file or directory: 'd:\\___Marlin2\\Marlin-bugfix-2.0.x_branch_4262\\Basic_test_nochanges\\.piolibdeps\\BIGTREE_SKR_MINI\\_tmp_installing-od8fp7-package\\TMC26XStepper-c1921b4856a5ef4b659b661b8cb5dd56f3075e57\\examples\\TMC26XMotorTester\\processing\\TMC26XMotorTest\\ChopperConfiguration.pde':
  File "C:\users\udo\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 126:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Udo\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Udo\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "C:\users\udo\.platformio\platforms\ststm32\builder\main.py", line 91:
    target_elf = env.BuildProgram()
  File "C:\Users\Udo\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\udo\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 124:
    _build_project_deps(env)
  File "C:\users\udo\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 49:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Udo\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\udo\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1043:
    project.install_dependencies()
  File "C:\users\udo\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 878:
    lm.install(uri)
  File "c:\users\udo\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 353:
    force=force)
  File "c:\users\udo\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 724:
    track=True)
  File "c:\users\udo\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 506:
    self.unpack(dlpath, tmp_dir)
  File "c:\users\udo\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 219:
    return fu.unpack(dest_dir, with_progress=False)
  File "c:\users\udo\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 118:
    self._unpacker.extract_item(item, dest_dir)
  File "c:\users\udo\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 38:
    self._afo.extract(item, dest_dir)
  File "C:\Users\Udo\.platformio\python27\Lib\zipfile.py", line 1031:
    return self._extract_member(member, path, pwd)
  File "C:\Users\Udo\.platformio\python27\Lib\zipfile.py", line 1086:
    file(targetpath, "wb") as target:
============================================== [ERROR] Took 3.52 seconds ==============================================

====================================================== [SUMMARY] ======================================================
Environment megaatmega2560              [IGNORED]
[... 11 lines cut ...]
Environment sanguino_atmega644p         [IGNORED]
Environment sanguino_atmega1284p        [IGNORED]
Environment STM32F1                     [IGNORED]
Environment fysetc_STM32F1              [IGNORED]
Environment BIGTREE_SKR_MINI            [FAILED]
Environment STM32F4                     [IGNORED]
Environment STM32F7                     [IGNORED]
Environment ARMED                       [IGNORED]
Environment alfawise_U20                [IGNORED]
Environment mks_robin                   [IGNORED]
Environment mks_robin_mini              [IGNORED]
Environment mks_robin_nano              [IGNORED]
Environment JGAURORA_A5S_A1             [IGNORED]
Environment black_stm32f407ve           [IGNORED]
Environment BIGTREE_SKR_PRO             [IGNORED]
Environment teensy35                    [IGNORED]
Environment malyanm200                  [IGNORED]
Environment esp32                       [IGNORED]
Environment fysetc_f6_13                [IGNORED]
Environment linux_native                [IGNORED]
======================================== 1 failed, 0 succeeded in 3.55 seconds ========================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

RudyBenner, 2019-07-22:
I see a pattern.

"C:\users\udo\.platformio\

when you delete that whole directory, and restart VSCODE, it will take some time to rebuild, and you are expected to not touch anything.

that is one of the 'cures'.

boelle, 2019-07-22:
so is it fair to say that this is not a marlin error but more user error and misunderstanding?

Udo-Nbg, 2019-07-22:
maybe ?
but why i can compile
ATMEGA2560
and SKR-PRO

sl1pkn07, 2019-07-22:
tested in my linux local from scratch with the @3DSmitty config without problems

have a tons of warnings, but it is warnings

greetings

Udo-Nbg, 2019-07-22:
I will give Linux a chance ...
I have to update Linux on my other machine.
Maybe I'm able until the end of the week I will send new information.

PS: 
I compiled SKR-Pro firmware right now,
but only on the 3rd try did the program run until successful

Not so with the SKR-mini-V1.1

Udo-Nbg, 2019-07-22:
@sl1pkn07 where i can find / download the @3DSmitty config

sl1pkn07, 2019-07-22:
simply edit the files by hand

Udo-Nbg, 2019-07-22:
i see no difference to my files

sl1pkn07, 2019-07-22:
try to remove the folder `.pioenvs` `.piolibs` `.piolibdeps` in the marlin project and try again

golfromeo-fr, 2019-07-24:
somewhere there is ".platformio/" folder with the cache of all the libs / compiler, you can clean the folder

Udo-Nbg, 2019-08-04:
Marlin-bugfix-2.0.x branch 4350 solved my Problem.
Now i can successfull compile.

Thank you

closed 2019-08-04 by Udo-Nbg