MarlinFirmware/Marlin #14307: [BUG] Compile Error in src/lcd/menu/menu_temperature.cpp with laser enabled

MarcelMo, 2019-06-16:
Bugfix 2.0 branch!
My cpp skills are limited, but in my opinion this looks like parameters to MENU_ITEM_EDIT_CALLBACK have been changed, but this line of code (in function menu_spindle_laser) is missing an update!
After defining SPINDLE_LASER_ENABLE this is the error I get:

> In file included from Marlin\src\lcd\menu\menu_temperature.cpp:31:0:
Marlin\src\lcd\menu\menu_temperature.cpp: In function 'void menu_spindle_laser()':
Marlin\src\lcd\menu\menu.h:303:56: error: no matching function for call to 'TMenuItem<MenuItemInfo_int3>::action_edit(const char [12], uint8_t*, int, int, void (&)())'
MenuItem_##TYPE ::action ## VARIANT(__VA_ARGS__); \
^
Marlin\src\lcd\menu\menu.h:339:62: note: in expansion of macro '_MENU_ITEM_VARIANT_P'
#define MENU_ITEM_EDIT_CALLBACK(TYPE, LABEL, ...)            _MENU_ITEM_VARIANT_P(TYPE, _edit, false, PSTR(LABEL), PSTR(LABEL), ## __VA_ARGS__)
^~~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_temperature.cpp:332:9: note: in expansion of macro 'MENU_ITEM_EDIT_CALLBACK'
MENU_ITEM_EDIT_CALLBACK(int3, MSG_LASER_POWER, &spindle_laser_power, SPEED_POWER_MIN, SPEED_POWER_MAX, update_spindle_laser_power);
^~~~~~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:190:17: note: candidate: static void TMenuItem<NAME>::action_edit(const char*, TMenuItem<NAME>::type_t*, TMenuItem<NAME>::type_t, TMenuItem<NAME>::type_t, screenFunc_t, bool) [with NAME = MenuItemInfo_int3; TMenuItem<NAME>::type_t = short int; screenFunc_t = void (*)()]
static void action_edit(PGM_P const pstr, type_t * const ptr, const type_t minValue, const type_t maxValue, const screenFunc_t callback=nullptr, const bool live=false) {
^~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:190:17: note:   no known conversion for argument 2 from 'uint8_t* {aka unsigned char*}' to 'TMenuItem<MenuItemInfo_int3>::type_t* {aka short int*}'
Marlin\src\lcd\menu\menu.h:307:110: error: cannot convert 'uint8_t* {aka unsigned char*}' to 'MenuItemInfo_int3::type_t* {aka short int*}' for argument '5' to 'void draw_menu_item_edit_int3(bool, uint8_t, const char*, const char*, MenuItemInfo_int3::type_t*, ...)'
draw_menu_item ## VARIANT ## _ ## TYPE(encoderLine == _thisItemNr, _lcdLineNr, PLABEL, ## __VA_ARGS__); \
^
Marlin\src\lcd\menu\menu.h:339:62: note: in expansion of macro '_MENU_ITEM_VARIANT_P'
#define MENU_ITEM_EDIT_CALLBACK(TYPE, LABEL, ...)            _MENU_ITEM_VARIANT_P(TYPE, _edit, false, PSTR(LABEL), PSTR(LABEL), ## __VA_ARGS__)
^~~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_temperature.cpp:332:9: note: in expansion of macro 'MENU_ITEM_EDIT_CALLBACK'
MENU_ITEM_EDIT_CALLBACK(int3, MSG_LASER_POWER, &spindle_laser_power, SPEED_POWER_MIN, SPEED_POWER_MAX, update_spindle_laser_power);
^~~~~~~~~~~~~~~~~~~~~~~
*** [.pioenvs\LPC1768\src\src\lcd\menu\menu_temperature.cpp.o] Error 1

Ludy87, 2019-06-17:
share your config files with us

MarcelMo, 2019-06-17:
> share your config files with us

configuration.h
https://pastebin.com/epfPnJ5Q
configuration_adv.h
https://pastebin.com/yPUnkMUL

The relevant part for this problem should be:

> #define SPINDLE_LASER_ENABLE
#if ENABLED(SPINDLE_LASER_ENABLE)
  #define SPINDLE_LASER_PWM_PIN P2_00
  #define SPINDLE_LASER_ENA_PIN P1_01
  #define SPINDLE_LASER_ENABLE_INVERT   false  // set to "true" if the on/off function is reversed
  #define SPINDLE_LASER_PWM             true   // set to true if your controller supports setting the speed/power
  #define SPINDLE_LASER_PWM_INVERT      false   // set to "true" if the speed/power goes up when you want it to go slower
  #define SPINDLE_LASER_POWERUP_DELAY   0   // delay in milliseconds to allow the spindle/laser to come up to speed/power
  #define SPINDLE_LASER_POWERDOWN_DELAY 0   // delay in milliseconds to allow the spindle to stop
  #define SPINDLE_DIR_CHANGE            false   // set to true if your spindle controller supports changing spindle direction
  #define SPINDLE_INVERT_DIR            false
  #define SPINDLE_STOP_ON_DIR_CHANGE    false   // set to true if Marlin should stop the spindle before changing rotation direction

  /**
   *  The M3 & M4 commands use the following equation to convert PWM duty cycle to speed/power
   *
   *  SPEED/POWER = PWM duty cycle * SPEED_POWER_SLOPE + SPEED_POWER_INTERCEPT
   *    where PWM duty cycle varies from 0 to 255
   *
   *  set the following for your controller (ALL MUST BE SET)
   */

  #define SPEED_POWER_SLOPE    1
  #define SPEED_POWER_INTERCEPT  0
  #define SPEED_POWER_MIN     0
  #define SPEED_POWER_MAX    255    // SuperPID router controller 0 - 30,000 RPM

  //#define SPEED_POWER_SLOPE      0.3922
  //#define SPEED_POWER_INTERCEPT  0
  //#define SPEED_POWER_MIN       10
  //#define SPEED_POWER_MAX      100      // 0-100%
#endif

PR #14312 (2019-06-17): Fix Spindle/Laser compile error
Fixes #14307
closed 2019-06-17 by thinkyhead