MarlinFirmware/Marlin #14315: Can't compile Marlin 2.0 while enabling BLTouch - STM32

anm8tor, 2019-06-17:
**Description**

**Steps to Reproduce**
- Verify Marlin 2.x arduino 1.9.0-beta on enabling bltouch and BILENEAR

**Expected behavior: Verification valid.**

**Actual behavior: It shows an error:**
In file included from Marlin\src\HAL\shared\servo.cpp:59:0:
Marlin\src\HAL\shared\servo_private.h:51:4: error: #error "This library only supports boards with an AVR or SAM3X processor."
#error "This library only supports boards with an AVR or SAM3X processor."
^~~~~
Marlin\src\HAL\shared\servo_private.h:62:0: warning: "MAX_SERVOS" redefined
#define MAX_SERVOS   (_Nbr_16timers  * SERVOS_PER_TIMER)

In file included from Marlin\src\HAL\shared\servo.h:78:0,
from Marlin\src\HAL\shared\servo.cpp:58:
Marlin\src\HAL\shared\../HAL_STM32F1/HAL_Servo_STM32F1.h:30:0: note: this is the location of the previous definition
#define MAX_SERVOS 3Compiling .pioenvs\BIGTREE_SKR_MINI\src\src\feature\bedlevel\bedlevel.cpp.o

In file included from Marlin\src\HAL\shared\servo.cpp:59:0:
Marlin\src\HAL\shared\servo_private.h:62:23: error: '_Nbr_16timers' was not declared in this scope
#Cdefine MAX_SERVOS   (_Nbr_16timers  * SERVOS_PER_TIMER)ompiling .pioenvs\BIGTREE_SKR_MINI\src\src\feature\bedlevel\mbl\mesh_bed_leveling.cpp.o

^
Marlin\src\HAL\shared\servo_private.h:93:31: note: in expansion of macro 'MAX_SERVOS'
extern ServoInfo_t servo_info[MAX_SERVOS];
^~~~~~~~~~
In file included from Marlin\src\HAL\shared\servo.cpp:59:0:
Marlin\src\HAL\shared\servo_private.h:97:21: error: variable or field 'initISR' declared void
extern void initISR(timer16_Sequence_t timer);
^~~~~~~~~~~~~~~~~~
Marlin\src\HAL\shared\servo_private.h:97:21: error: 'timer16_Sequence_t' was not declared in this scope
Marlin\src\HAL\shared\servo_private.h:97:21: note: suggested alternative: 'timer_resume'
extern void initISR(timer16_Sequence_t timer);
^~~~~~~~~~~~~~~~~~
timer_resume
Marlin\src\HAL\shared\servo_private.h:98:20: error: variable or field 'finISR' declared void
extern void finISR(timer16_Sequence_t timer);
^~~~~~~~~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pioenvs\BIGTREE_SKR_MINI\src\src\HAL\shared\servo.cpp.o] Error 1

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3296722/Configuration.zip)

[image]

RudyBenner, 2019-06-19:
I don't think this board will support a servo

anm8tor, 2019-06-19:
> I don't think this board will support a servo

According to BTT tech reps, it will. They even have pictures showing how to hook a BLTouch up and modify the XMAX input to do so..... this is strictly a coding issue, that I lack knowledge in.

RudyBenner, 2019-06-19:
It would be nice if they shared that information.

tpruvot, 2019-06-21:
Fixed the F1 warnings in #14333 but i dunno for your issue

pinchies, 2019-06-23:
Can someone please clarify... does servo control work on STM32F1 at the moment? Is there a working bltouch example?

tpruvot, 2019-06-23:
yes on the F1 with such thing in the pins.h

//
// PWM
//
//#define NUM_SERVOS        1
#define SERVO0_TIMER_NUM    8 // General or Adv. timer to use for the servo PWM (2 & 5 are reserved)

#define SERVO0_PWM_OD
#define SERVO0_PIN       PD13 // Open drain PWM pin on the V0G (GND or floating 5V)

david123718, 2019-06-24:
Hello! 

I am getting this error and I am not able to compile when enabling BLTouch:

`In file included from Marlin\src\HAL\HAL_STM32F1\HAL_spi_STM32F1.cpp:44:0:
C:\Users\david\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\SPI\src/SPI.h:171:15: warning: '_spi3_this' defined but not used [-Wunused-variable]
static void (*_spi3_this);
^~~~~~~~~~
C:\Users\david\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\SPI\src/SPI.h:168:15: warning: '_spi2_this' defined but not used [-Wunused-variable]
static void (*_spi2_this);
^~~~~~~~~~
C:\Users\david\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\SPI\src/SPI.h:165:15: warning: '_spi1_this' defined but not used [-Wunused-variable]
static void (*_spi1_this);
^~~~~~~~~~
C:\Users\david\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\SPI\src/SPI.h:163:16: warning: 'ff' defined but not used [-Wunused-variable]
static uint8_t ff = 0XFF;
^~
In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:41:0,
from Marlin\src\HAL\HAL_STM32F1\HAL_Servo_STM32F1.cpp:26:
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:995:3: error: non-constant condition for static assertion
static_assert(FLOOR(float(X_PROBE_OFFSET_FROM_EXTRUDER)) == float(X_PROBE_OFFSET_FROM_EXTRUDER), "X_PROBE_OFFSET_FROM_EXTRUDER must be an integer!");
^~~~~~~~~~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfigPre.h:32:0,
from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:28,
from Marlin\src\HAL\HAL_STM32F1\HAL_Servo_STM32F1.cpp:26:
Marlin\src\HAL\HAL_STM32F1\../../inc/../core/macros.h:260:27: error: call to non-constexpr function 'float floorf(float)'
#define FLOOR(x)    floorf(x)
~~~~~~^~~
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:995:17: note: in expansion of macro 'FLOOR'
static_assert(FLOOR(float(X_PROBE_OFFSET_FROM_EXTRUDER)) == float(X_PROBE_OFFSET_FROM_EXTRUDER), "X_PROBE_OFFSET_FROM_EXTRUDER must be an integer!");
^~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:41:0,
from Marlin\src\HAL\HAL_STM32F1\HAL_Servo_STM32F1.cpp:26:
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:996:3: error: non-constant condition for static assertion
static_assert(FLOOR(float(Y_PROBE_OFFSET_FROM_EXTRUDER)) == float(Y_PROBE_OFFSET_FROM_EXTRUDER), "Y_PROBE_OFFSET_FROM_EXTRUDER must be an integer!");
^~~~~~~~~~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfigPre.h:32:0,
from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:28,
from Marlin\src\HAL\HAL_STM32F1\HAL_Servo_STM32F1.cpp:26:
Marlin\src\HAL\HAL_STM32F1\../../inc/../core/macros.h:260:27: error: call to non-constexpr function 'float floorf(float)'
#define FLOOR(x)    floorf(x)
~~~~~~^~~
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:996:17: note: in expansion of macro 'FLOOR'
static_assert(FLOOR(float(Y_PROBE_OFFSET_FROM_EXTRUDER)) == float(Y_PROBE_OFFSET_FROM_EXTRUDER), "Y_PROBE_OFFSET_FROM_EXTRUDER must be an integer!");
^~~~~
Marlin\src\HAL\HAL_STM32F1\HAL_Servo_STM32F1.cpp: In member function 'bool libServo::attach(int32_t, int32_t, int32_t)':
Marlin\src\HAL\HAL_STM32F1\HAL_Servo_STM32F1.cpp:101:11: warning: unused variable 'tchan' [-Wunused-variable]
uint8_t tchan = PIN_MAP[pin].timer_channel;
^~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:41:0,
from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:36:
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:995:3: error: non-constant condition for static assertion
static_assert(FLOOR(float(X_PROBE_OFFSET_FROM_EXTRUDER)) == float(X_PROBE_OFFSET_FROM_EXTRUDER), "X_PROBE_OFFSET_FROM_EXTRUDER must be an integer!");
^~~~~~~~~~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\HAL.h:39:0,
from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:34:
Marlin\src\HAL\HAL_STM32F1\../../core/macros.h:260:27: error: call to non-constexpr function 'float floorf(float)'
#define FLOOR(x)    floorf(x)
~~~~~~^~~
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:995:17: note: in expansion of macro 'FLOOR'
static_assert(FLOOR(float(X_PROBE_OFFSET_FROM_EXTRUDER)) == float(X_PROBE_OFFSET_FROM_EXTRUDER), "X_PROBE_OFFSET_FROM_EXTRUDER must be an integer!");
^~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:41:0,
from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:36:
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:996:3: error: non-constant condition for static assertion
static_assert(FLOOR(float(Y_PROBE_OFFSET_FROM_EXTRUDER)) == float(Y_PROBE_OFFSET_FROM_EXTRUDER), "Y_PROBE_OFFSET_FROM_EXTRUDER must be an integer!");
^~~~~~~~~~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\HAL.h:39:0,
from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:34:
Marlin\src\HAL\HAL_STM32F1\../../core/macros.h:260:27: error: call to non-constexpr function 'float floorf(float)'
[... 103 lines cut ...]
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:995:17: note: in expansion of macro 'FLOOR'
static_assert(FLOOR(float(X_PROBE_OFFSET_FROM_EXTRUDER)) == float(X_PROBE_OFFSET_FROM_EXTRUDER), "X_PROBE_OFFSET_FROM_EXTRUDER must be an integer!");
^~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:41:0,
from Marlin\src\HAL\HAL_STM32F1\watchdog_STM32F1.cpp:29:
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:996:3: error: non-constant condition for static assertion
[... cut ...]
*** [.pioenvs\JGAURORA_A5S_A1\src\src\HAL\HAL_STM32F1\persistent_store_sdcard.cpp.o] Error 1
*** [.pioenvs\JGAURORA_A5S_A1\src\src\HAL\HAL_STM32F1\watchdog_STM32F1.cpp.o] Error 1
*** [.pioenvs\JGAURORA_A5S_A1\src\src\HAL\HAL_STM32F1\u8g_com_stm32duino_fsmc.cpp.o] Error 1`

It is clear that the XY Offsets are whole integers. 
`#define X_PROBE_OFFSET_FROM_EXTRUDER 10 // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 10 // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER 0 // Z offset: -below +above  [the nozzle]`

david123718, 2019-06-24:
This is after I applied the fix that @tpruvot posted.

tpruvot, 2019-06-24:
you need to -std=gnu++14 changes to build with bltouch support, see PR #14366 for a sample

david123718, 2019-06-24:
didn't work, I applied the changed and I am still getting the same errors. 
New code: 
#
# JGAurora A5S A1 (STM32F103ZET6)
#
[env:JGAURORA_A5S_A1]
platform      = ststm32@5.3.0
framework     = arduino
board         = genericSTM32F103ZE
extra_scripts = buildroot/share/PlatformIO/scripts/jgaurora_a5s_a1_with_bootloader.py
build_flags   = !python Marlin/src/HAL/HAL_STM32F1/STM32F1_flag_script.py
  ${common.build_flags} -std=gnu++14
build_unflags = std=gnu++11
src_filter    = ${common.default_src_filter} +<src/HAL/HAL_STM32F1>
lib_deps      = ${common.lib_deps}
lib_ignore    = c1921b4
  libf3c
  lib066
  Adafruit NeoPixel_ID28
  Adafruit NeoPixel
  libf3e
  TMC26XStepper
lib_ldf_mode  = 1
monitor_speed = 250000

tpruvot, 2019-06-24:
typo in unflags

david123718, 2019-06-24:
@tpruvot fixed the typo, still getting the same error. 

[env:JGAURORA_A5S_A1]
platform      = ststm32@5.3.0
framework     = arduino
board         = genericSTM32F103ZE
extra_scripts = buildroot/share/PlatformIO/scripts/jgaurora_a5s_a1_with_bootloader.py
build_flags   = !python Marlin/src/HAL/HAL_STM32F1/STM32F1_flag_script.py
  ${common.build_flags} -std=gnu++14
build_unflags = -std=gnu++11
src_filter    = ${common.default_src_filter} +<src/HAL/HAL_STM32F1>
lib_deps      = ${common.lib_deps}
lib_ignore    = c1921b4
  libf3c
  lib066
  Adafruit NeoPixel_ID28
  Adafruit NeoPixel
  libf3e
  TMC26XStepper
lib_ldf_mode  = 1
monitor_speed = 250000

tpruvot, 2019-06-24:
maybe you need to remove @5.3.0 on the platform too.. and do a "pio update"

pinchies, 2019-06-24:
We added the 5.3.0 because we were having other build problems with the latest STM32 package.

david123718, 2019-06-24:
Removing the 5.3.0 platform and then doing "pio update" did not work either. Still receiving the same errors.

tpruvot, 2019-06-24:
so run platformio run --verbose to show the command lines

frezap, 2019-06-24:
Doing everything as mentioned here and setting ststm to 5.4.1 results in these outputs when using --verbose. I cant really read much into it unfortunatly.

https://pastebin.com/bPVMnZGr
As seen in davids errors.

And after commenting out the SanityChecks since they are kinda pointless, or at least should... I get the same error as in the original post.
https://pastebin.com/hthPG4ds

Hope it helps!

tpruvot, 2019-06-25:
I dont know, your source code is not normal.. shared/servo.cpp is not used/required on the F1 and even your pastebins seems corrupted

frezap, 2019-06-26:
I got Marlin to compile when doing this:

https://github.com/frezap/Marlin/commit/49d7bb2b763f6d21f5545978d6508948a342fa88

But servos still don't seem to work on our board. I couldn't figure out exactly why yet.
Maybe @anm8tor could give it a try?

anm8tor, 2019-06-26:
@frezap I have tried already. My problem is that I don't know coding very well at all. That's the reason I started this issue thread in the first place. I know I have the hardware set up correctly for the Mini, but am stuck beyond that

david123718, 2019-06-26:
> > I don't think this board will support a servo
> 
> According to BTT tech reps, it will. They even have pictures showing how to hook a BLTouch up and modify the XMAX input to do so..... this is strictly a coding issue, that I lack knowledge in.

What is BTT?

anm8tor, 2019-06-26:
@david123718 BigTreeTech, the maker of the SKR boards

anm8tor, 2019-07-07:
No progress on this makes me sad.... 32bit board going unused :(

AnHardt, 2019-07-07:
> No progress on this makes me sad.... 32bit board going unused :(

Why? There are that many kinds of probes not using servo code.

BL-Touch, with its trigger *pulse*, is *crap by design* anyway. Commercially successful closed source hardware is provoking incompatible clones. A commercially successful Kickstarter because (then) lying about Marlin compatibility  Maintenance became a  nightmare. And now they deliver "improved" incompatible, to themselves, probes without previous announcement. 

**Drop BL-Touch support to set a sign against closed source hardware in the RepRap universe.**
**Ban them from Marlin!**

anm8tor, 2019-07-07:
@AnHardt because I received the board through a contest and the BLTouch at no cost. My Anet is running fine and I won't swap out the board until I have that part supported out

AnHardt, 2019-07-07:
Gifted toys are still too expensive if proprietary. 
Support the RepRap idea!

frezap, 2019-09-24:
BLTouch works for me with this servo set up now:

#define NUM_SERVOS 1
#define SERVO0_TIMER_NUM 8
//#define SERVO0_PWM_OD
#define SERVO0_PIN PD7

sjasonsmith, 2019-10-12:
@tpruvot, how did you know to use timer 8 for the servo timer?
```
#define SERVO0_TIMER_NUM 8 // General or Adv. timer to use for the servo PWM (2 & 5 are reserved)
```

The HAL now defines timer 1 as the default. This caused many problems for me. When I look at the servo output with an oscilloscope, I see the expected pulse width briefly, then the output reverts to almost 100% high. I suspect something else is altering the timer registers after the BLTouch command is sent.

Changing it to use Timer 8 seems to fix everything, and I now have good clean PWM signals. The BLTouch commands remain active until a new command is sent, matching the behavior with my old AVR board.

I plan to do more investigation tomorrow to better understand the problems with timer 1. Based on your comments on this issue, I suspect you may already know the answer?

tpruvot, 2019-10-12:
i just used one not declared at all in marlin to be sure... didnt see any problem with timer 1 neither... even if that was set/changed without asking me... 1 & 8 are both advanced timers... maybe some tiny F1 
 chips only have the number 1... and not 8, unsure about that..

sjasonsmith, 2019-10-12:
Thanks, I’ll do some more research to find out why timer 1 is being corrupted for me. I can fix it by using 8, but I’d like to understand why that is necessary.

sjasonsmith, 2019-10-13:
@tpruvot, it turned out to be a conflict between the fan pin on all the SKR E3 boards and Timer 1. It looks like the SKR Mini doesn't have the same issue. Your post about using Timer 8 sure helped me fine the root-cause of this failure!
#15547

tpruvot, 2019-10-13:
Yes, the F103CB does not have it... Even if im unsure if this tiny chip is big enough for marlin 2...

```
extern struct timer_dev *const TIMER1;
extern struct timer_dev *const TIMER2;
extern struct timer_dev *const TIMER3;
extern struct timer_dev *const TIMER4;
#if defined(STM32_HIGH_DENSITY) || defined(STM32_XL_DENSITY)
extern struct timer_dev *const TIMER5;
extern struct timer_dev *const TIMER6;
extern struct timer_dev *const TIMER7;
extern struct timer_dev *const TIMER8;
#endif
#ifdef STM32_XL_DENSITY
extern struct timer_dev *const TIMER9;
extern struct timer_dev *const TIMER10;
extern struct timer_dev *const TIMER11;
extern struct timer_dev *const TIMER12;
extern struct timer_dev *const TIMER13;
extern struct timer_dev *const TIMER14;
#endif
```

sjasonsmith, 2020-05-12:
@thinkyhead, I assume this could be closed? BLTouch works in both STM32 and STM32F1 HALs. If issues remaining they are board-specific, but that would probably warrant new issues as-needed.

pinchies, 2020-05-12:
I've seen successful users on JGAurora STM32 use BLTouch. I second the above.

anm8tor, 2020-07-01:
Sorry,  still learning!

closed 2020-07-01 by anm8tor