MarlinFirmware/Marlin #14318: Cannot mount SD card on my computer with the latest Marlin, SKR 1.3

hudja, 2019-06-18:
Hi, 

I just installed the latest Marlin bugfix, the one with Square Wave feature, and I cannot mount SD card on my Mac anymore via USB. The board I am using is SKR 1.3. I plug SD card into the board card reader. I can connect to the printer using terminal (pronterface or s3d) and control it. The settings I am using worked fine with the previous release (downloaded on 29.05), and I also tried to downgrade firmware and can confirm that with old firmware SD support is back to normal. 

One thing I noticed is that with the latest release the card is automatically mounted in the printer, when I turn it on. On the previous release, I need to click 'Init SD card' to mount it in the printer. Releasing SD card in the printer menu on the latest release does not mount it on Mac.

The relevant code which worked fine with the previous release is:
configuration.h:
```
#define SERIAL_PORT -1

#define BAUDRATE 250000

```
configuration_adv.h
```
  #ifdef TARGET_LPC1768
    //#define LPC_SD_LCD          // Use the SD drive in the external LCD controller.
    #define LPC_SD_ONBOARD      // Use the SD drive on the control board. (No SD_DETECT_PIN. M21 to init.)
    //#define LPC_SD_CUSTOM_CABLE // Use a custom cable to access the SD (as defined in a pins file).
    //#define USB_SD_DISABLED     // Disable SD Card access over USB (for security).
    #if ENABLED(LPC_SD_ONBOARD)
      #define USB_SD_ONBOARD    // Provide the onboard SD card to the host as a USB mass storage device.
    #endif
  #endif
[archive.zip](https://github.com/MarlinFirmware/Marlin/files/3299595/archive.zip)

```

I also attach archive of both config files from previous (SD card mounted successfully) and current (faulty SD card) releases.

Thanks!

shitcreek, 2019-06-18:
Latest version has some bugs. The SD card auto mounting at startup is being worked on. This might be a related issue.
I'm using an older commit for now.

gloomyandy, 2019-06-18:
Trying to reproduce this now (and work out what is going wrong). I suspect that the USB sharing code has got part way through making the SD card available to the host when the card "disappears" due to it being mounted by the new init code. I'm not really sure why that should stop the Mac from trying to mount it again though when it is released. Can you try releasing the card and then unplugging/replugging  the USB cable, does the card show ip on the Mac then?

gloomyandy, 2019-06-18:
@shitcreek did you see this problem as well? Do you use a Mac or a PC?

gloomyandy, 2019-06-18:
Hmm I can't reproduce this problem using either a Windows PC or a Linux system (rPi). I do see on Linux that things get very messy when Marlin boots though and I suspect that perhaps the Mac sees this and just decides to ignore the drive from then on.

So if anyone has a Mac and is seeing this problem can you try something for me. In the file: Marlin/src/HAL/HAL_LPC1768/main.cpp can you locate the lines that look like this....
```C
  #if DISABLED(USB_SD_DISABLED)
    MSC_SD_Init(0);                         // Enable USB SD card access
  #endif
```
and change them to look like this...
```C
  #if DISABLED(USB_SD_DISABLED)
    //MSC_SD_Init(0);                         // Enable USB SD card access
  #endif
```
then rebuild and try this new firmware build. Let me know if this fixes the problem.

hudja, 2019-06-18:
Hi, I tried both release SD card and reconnect printer, and you code fix. Both does not work.

MarkMan0, 2019-06-18:
Try disabling `USB_SD_ONBOARD`. It works for me like that for some reason. 

```
  #ifdef TARGET_LPC1768
    //#define LPC_SD_LCD          // Use the SD drive in the external LCD controller.
    #define LPC_SD_ONBOARD      // Use the SD drive on the control board. (No SD_DETECT_PIN. M21 to init.)
    //#define LPC_SD_CUSTOM_CABLE // Use a custom cable to access the SD (as defined in a pins file).
    //#define USB_SD_DISABLED     // Disable SD Card access over USB (for security).
    #if ENABLED(LPC_SD_ONBOARD)
      //#define USB_SD_ONBOARD    // Provide the onboard SD card to the host as a USB mass storage device.
    #endif
  #endif
```

hudja, 2019-06-18:
MarkMan0, do I need to use the fixed code from above also?

Grogyan, 2019-06-18:
@MarkMan0 that would defeat the ability to use the onboard SD to be seen from external host programs, like OctoPrint

gloomyandy, 2019-06-18:
Actually disabling USB_SD_ONBOARD will end up with both Marlin and your PC/Mac accessing the SD card at the same time, which may work some of the time but will almost certainly at some point end up with a corrupt SD card. It is interesting that it works though.

Can any one else confirm that the above fix I posted does not work on a Mac? As far as I can see it should work fine. 

@hudja I am very surprised that releasing the SD card from the menu and then unplugging and  re-attaching the printer did not work. That situation is pretty much identical to what happened with the old code. Is there any chance you have other config changes?

Grogyan, 2019-06-18:
@MarkMan0 I'm surprised it even worked for you like this, it doesn't for me, thus, have to ensure the USB_SD_ONBOARD is enabled

hudja, 2019-06-18:
I can confirm, it also works for me with:

```
#ifdef TARGET_LPC1768
    //#define LPC_SD_LCD          // Use the SD drive in the external LCD controller.
    #define LPC_SD_ONBOARD      // Use the SD drive on the control board. (No SD_DETECT_PIN. M21 to init.)
    //#define LPC_SD_CUSTOM_CABLE // Use a custom cable to access the SD (as defined in a pins file).
    //#define USB_SD_DISABLED     // Disable SD Card access over USB (for security).
    #if ENABLED(LPC_SD_ONBOARD)
      //#define USB_SD_ONBOARD    // Provide the onboard SD card to the host as a USB mass storage device.
    #endif
  #endif
```

and

```
#if DISABLED(USB_SD_DISABLED)
    MSC_SD_Init(0);                         // Enable USB SD card access
  #endif

```

I can connect with Pronterface and see SD card.

gloomyandy, 2019-06-18:
Seriously that is not a very safe configuration. Please don't use it, it is certainly not a fix for this problem.

@hudja With your old (working firmware), can you confirm that doing the following works....
1. Boot your printer.
2. SD card should appear on your Mac
3. Choose init SD from the menu, SD card should disappear from the Mac
4. Choose Release Sd from the menu, SD card should again be available to the Mac
In particular please check step 4 does that work for you?

gloomyandy, 2019-06-18:
Oh and what if anything are you doing on the Mac to mount the SD card? Does it just appear or do you need to do something?

gloomyandy, 2019-06-18:
Also if you are doing this testing using the SD card that came free with the SKR board please try using a different SD card. That "free" card seems to cause all sorts of odd things to happen when you switch between Marlin and PC access.

hudja, 2019-06-18:
> Seriously that is not a very safe configuration. Please don't use it, it is certainly not a fix for this problem.
> 
> @hudja With your old (working firmware), can you confirm that doing the following works....
> 
> 1. Boot your printer.
> 2. SD card should appear on your Mac

Yes.

> 3. Choose init SD from the menu, SD card should disappear from the Mac

Yes.

> 4. Choose Release Sd from the menu, SD card should again be available to the Mac

No, it does not reappear. Reconnecting USB cable also does not help. I need to turn off/on the printer to get it back.

Concerning your other question, I do not need to do anything. Mac sees and mounts it automatically.

I am using other (not free) SD card, which was working fine for long time now.

hudja, 2019-06-18:
> > 1. Choose init SD from the menu, SD card should disappear from the Mac
> 
> Yes.

Correction. 
After turning printer on, I can see the SD card and contents.
I 'Init SD card'. I can still see SD card on Mac in Finder, but it is empty now, no contents. Only after some time I can see that Card was released on Mac.

gloomyandy, 2019-06-18:
@hudja If your drive does not re-appear at step 4 then something is not working correctly. That is the same problem that you are seeing now because with the current  code you will in effect always be going from the state of Marlin having the card mounted to Marlin no longer having it mounted.

If you have any other SD cards please try them with the above test, this is exactly the situation I have seen many times before with the "free" SD card. Most of the time it works fine but when you switch from Marlin access to pc/Mac access you get this sort of problem. Perhaps your current SD card is triggering a similar issue? I am not saying this is not a problem, but I'd rather try and solve one at a time!

hudja, 2019-06-18:
I am afraid to plug in my 128Gb and 256Gb cards into it :) Will buy a small new one later today and try it.

gloomyandy, 2019-06-18:
@hudja just seen your correction so if after doing the Init SD you wait until the card is no longer seen on the Mac, does doing a release on the printer make the drive come back again on the Mac or is it lost for ever?

Ah yes I wouldn't plug a large SD card in there! An 8Gb or 16Gb card is probably the best to try! Thanks for doing this.

hudja, 2019-06-18:
It behaves strangely. When I init, I can still see SD card in Finder, but it is empty. When I release, contents of SD card reappear in Finder (I can still see card as mounted all the time). And only after few init/release cycles, the Mac says that card was unsafely removed and card disappears from the Finder.

gloomyandy, 2019-06-18:
If you do an Init SD and then wait a few minutes before you do the release what happens?
If you do an Init SD and then do nothing for a long time do you ever get any sort of message? Does the card remain visible?

hudja, 2019-06-18:
It is behaving erratically. Now, it takes about 20-30 seconds on the first Init for Mac to throw out an error 'Disc not ejected properly'. Also it takes (and took before) quite long time to mount it after printer turn on, up to 30 seconds.

forkoz, 2019-06-19:
So this is the offending commit: https://github.com/MarlinFirmware/Marlin/commit/eb1c9113c23869b839abb63b7a43f84d13ed603c

Maybe 

`#if ENABLED(SDIO_SUPPORT) && !PIN_EXISTS(SD_DETECT)`

needs another qualifier for boards using the eeprom.dat file or just a straight setting is needed.

gloomyandy, 2019-06-19:
Hmm that is all very strange. 

If you have chance can you retest the latest software with the change that I gave above (comment out, the call to MSC_SD_Init(0); then rebuild and flash to the printer. Then can you try the following...
1.Boot the printer - what happens is the SD card visible in any way on the Mac?
2. Select release SD card - what happens? If you wait a few minutes does anything change?
3. Select Init SD - what happens does anything change?

Thanks again for your help.

gloomyandy, 2019-06-19:
@forkoz that line only relates to boards that use SDIO SD card readers which is not the case here. The "problem" is the changes made to ultralcd.cpp which basically means that on boot Marlin will always try and mount the SD card, before these changes if there was no SD detect pin then it would not try and mount the card until you told it to with Init SD or whatever.

But really going back to the way things were is not the best solution if we can avoid it. There are good reasons to mount the SD card if we can. Also it seems to work fine with Linux and Windows, unfortunately that is not the case with the Mac. Which of course is the one machine type I don't have easy access to!

gloomyandy, 2019-06-19:
So if there is anyone else with a Mac and an LPC1768 based printer out there I would very much appreciate some help testing this case.

hudja, 2019-06-19:
ok, will test after getting a new card. btw, why this is not the safe solution (I do not use octopi)?

#ifdef TARGET_LPC1768
    //#define LPC_SD_LCD          // Use the SD drive in the external LCD controller.
    #define LPC_SD_ONBOARD      // Use the SD drive on the control board. (No SD_DETECT_PIN. M21 to init.)
    //#define LPC_SD_CUSTOM_CABLE // Use a custom cable to access the SD (as defined in a pins file).
    //#define USB_SD_DISABLED     // Disable SD Card access over USB (for security).
    #if ENABLED(LPC_SD_ONBOARD)
      //#define USB_SD_ONBOARD    // Provide the onboard SD card to the host as a USB mass storage device.
    #endif
  #endif

gloomyandy, 2019-06-19:
Because you now have two file systems (the one in your printer and the one on your PC) accessing the disk at a block level at the same time. one will not know anything about the changes made by the other. So if you copy a file to the disk from your Mac it may not show up on the list of files on the printer unless you do a refresh. Even worse if your printer and your Mac both write to the disk (and your printer may do this if you delete a file, or are using something like power loss recovery) then you will have a corrupt SD card as both will have updated the low level file structure without knowing that the other has also done the same.

You may get away with it for a while, but it is not safe as a general solution.

thinkyhead, 2019-06-19:
Over at #14325 I have modified the patch so that Marlin won't try to mount the outboard SD card if the onboard SD card is at all USB-accessible. Please test the patch to make sure it does the right thing for your setup.

hudja, 2019-06-22:
Hi, sorry for the late reply. Took little bit longer to get a new card than expected. I am using new 16Gb sandisc ultra formatted in FAT. I can confirm previous behaviour. It takes about 20s until card is mounted on mac on printer boot up. When I click 'init' on printer, it takes up to a minute for Mac to show 'Disc not ejected properly' error. After I click 'release' on printer, the card is not re-mounted on Mac. I need to restart printer to re-mount it. I am using latest OS X Mojave.

thinkyhead, 2019-06-22:
Try a low-level format on your SD card to see if it then mounts more quickly.

hudja, 2019-06-22:
15 seconds to mount on boot. Could you please point me to exact change I need to do from https://github.com/MarlinFirmware/Marlin/pull/14325?

gloomyandy, 2019-06-22:
15 to 20s sounds about right. Windows/Mac/Linux systems all seem to like to read all of the system structures (like the file allocation table) when they mount an SD card, this can be many hundreds (sometimes thousands) of blocks. 

What version of Marlin did you use for the above tests? If you want to try the latest changes just use the current state of the code in  #14325.

Could you also try the following for me (on the Marlin version you used for the above tests and if possible on the PR version) ...
1. Boot Marlin (SD card should be mounted by your Mac)
2. On the Mac eject the SD card to unmount it.
3. On Marlin do Init SD
4. On Marlin do Release SD
What happens? Does the Mac remount the SD card? Do you get any sort of error message from the Mac?

Thanks again for your help with this.

VanessaE, 2019-07-02:
I finally picked up a LCD for my bot -- a common RepRap Discount Full Graphic 12864, which gave me a reason to pull out my SKR v1.3 and set things up.  No problems at all with the basics, but I've ran into problems with both SD cards (SKR onboard and on the LCD).

This is on my usual PC, running Debian Buster, with the SKR v1.3's copy of Marlin at commit c28e08c8.

Depending on what SD options I choose, I can get either the SKR's onboard SD or the LCD's SD to show up to Marlin (that is, accessible via the LCD menus), but even after reading through this issue and through the comments in #14325, I can find no combination of settings that mimics the way the SD works on my SKR v1.1 (commit 2513f6b5 there, with no LCD).

It doesn't matter whether or not the card is inserted when the SKR boots, nor does "Release SD card" in the menu have any effect at the PC end.  Looking at `dmesg`, there's literally zero reaction at all to inserting or removing the card, nor to the "Init" and "Release" items in the LCD menu.

I'd have expected Marlin to present the SKR's onboard card as a mountable volume on the PC, with the LCD's SD for printing via the usual G-code upload commands.

If it matters, I'm only running on USB power right now, since I'm still just tinkering.

Here are my configs:

<details><summary>Configuration.h</summary>

```
/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2019 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#define CONFIGURATION_H_VERSION 020000

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
[... 4665 lines cut ...]
/**
 * Advanced Print Counter settings
 */
#if ENABLED(PRINTCOUNTER)
  #define SERVICE_WARNING_BUZZES  3
  // Activate up to 3 service interval watchdogs
  //#define SERVICE_NAME_1      "Service S"
  //#define SERVICE_INTERVAL_1  100 // print hours
  //#define SERVICE_NAME_2      "Service L"
  //#define SERVICE_INTERVAL_2  200 // print hours
  //#define SERVICE_NAME_3      "Service 3"
  //#define SERVICE_INTERVAL_3    1 // print hours
#endif

// @section develop

/**
 * M43 - display pin status, watch pins for changes, watch endstops & toggle LED, Z servo probe test, toggle pins
 */
//#define PINS_DEBUGGING

// Enable Marlin dev mode which adds some special commands
//#define MARLIN_DEV_MODE
```
</details>

gloomyandy, 2019-07-02:
@VanessaE This is what should happen with your configuration files as posted (you have things setup to shared the onboard reader between USB and Marlin) and an SD card in the onboard reader (please don't use the "free" SD card that comes with the board). Boot your printer. You should see dmesg activity on your Linux system about the SD card over USB, it may or may not get automounted, but should be available for mounting. If you go to the LCD then the SD card should not be visible, but the InitSD menu item should be there. If you choose InitSD then you should see the SD card become unavailable from Linux and available to Marlin. Release SD should reverse this process. In this mode the SD detect pins are not used so removing/inserting the SD card is probably not a good idea. With this configuration the LCD card reader should not be active (and it should probably not have a card in it).

That's what I see with rPi 3B+ (and the equivalent on a WIndows PC). I don't have a system running Buster. 

If you change SDCARD_CONNECTION to be LCD then the onboard SD reader should be made available to your PC (again the SD detect pin will not be used, so best to start the printer with the card inserted). The LCD reader should be used by Marlin (and the SD detect pins for that reader should be active). In this mode the SD card menu options should have no effect on the card seen by the PC. 

I assume you have not made any other changes other than to Configuration.h and Configuration_Adv.h files? In particular with the current version of Marlin do not comment out the calls to MSC_SD_Init(0).
Is USB working at all to your PC from the board? Do you see the CDC serial device get created and can you talk to the board via it?

VanessaE, 2019-07-02:
Ok, I feel stupid now.  It was just a cabling error (no data was getting through).  Easy fix, sorry for the false report.

`#define SDCARD_CONNECTION LCD` indeed works as you describe (the PC sees/mounts the onboard card, and Marlin sees the LCD's card), so you can ignore the entirety of my previous comment (deleted anyway to reduce clutter).

However, I discovered a new issue:

In the above mode, if I boot with the LCD SD card inserted, the beeper first sounds its normal long tone while it boots (you guys really ought to get rid of that if possible), then as soon as the Marlin title card is replaced by the main info screen, it begins beeping rapidly for a few seconds, then throws "MINTEMP BED" and tells me to reset (and sounds the usual long tone).

If I leave the LCD's card out at boot, then insert it once Marlin's up, it behaves as it ought to.  It doesn't matter if the SKR's SD card is inserted at boot or not.  Sounds like it's getting caught in a loop and triggering the watchdog... 

> I assume you have not made any other changes other than to Configuration.h and Configuration_Adv.h files?

Don't really need to answer this now since I fixed my derp, :smiley: but, for this round of tinkering, I started from a clean clone, then imported my Configuration*.h settings, my thermistor table (see #14258), and the usual one-line change to platformio.ini that makes Atom's "-->" button work.  There are no other changes.

gloomyandy, 2019-07-02:
Glad you have resolved the SD issue! As to the others I think you should create a new issue for each of them. I've not seen either problem (but I have a different configuration), but my thoughts are....

Beeping, none of my setups have ever beeped on startup (even when using the same display as you). I suspect that it may be a combination of the exact version of the display (which beeper is used, any pullups etc.), the pin the board has selected for the beeper and possibly the bootloader (it may be setting the state of that pin differently to others). But it sounds annoying and should really be addressed if possible!

The MINTEMP Bed issue. I am pretty sure this is down to the way that the LPC176x HAL has additional filtering on the ADCs to reduce noise. The way this is currently implemented means that you need to have performed a certain number of reads for the temperature to settle at the actual value. I think in the past the checks on temperature were not activated until you printed, but a few months ago they got turned on when the printer boots. When the change was first made lots of folks hit problems. A sort of fix is currently in place using THERMAL_PROTECTION_GRACE_PERIOD which means that there is a short delay (at boot) before readings are delivered. However this is not really a very good fix (IMO) as it does not really address the actual problem (because it is based on time, not the number of actual readings taken). This means that if something else (like reading the SD card) uses up that grace period (and so the extra readings don't happen), you get this error again. You could try changing the setting from 1000 to say 5000 to see if that helps (the setting is in HAL_LPC1768/hal.h I think).

AnHardt, 2019-07-02:
Beeping.
Could be.
Most bootloaders use a LED for making their state visible (debugging). If that pin is connected to the beeper one could get the described pattern - until Marlin takes over and inits its beeper.

ManuelMcLure, 2019-07-02:
The long beep at startup is unfortunately a hardware issue. The beeper will sound until the LCD gets initialized, which doesn't happen until the bootloader gives control to Marlin, so there's nothing Marlin can do to disable the beeper.

However, there is a simple hardware change that can be done to disable the beep on startup - see #13965 for details.

VanessaE, 2019-07-02:
Sounds good, then we're just down to that bogus "crash" if booting with the LCD SD card inserted.

VanessaE, 2019-07-02:
Sounds good, then we're just down to that bogus "crash" (see #14485)

cinealfa, 2019-07-22:
This thread veered somewhat, is the issue resolved with mounting the sd card?

gloomyandy, 2019-07-22:
As far as I can tell there is not currently a problem when using Linux or Windows, there could possibly be a problem when using a Mac, but I have no way of testing that. 

It could just be that the Mac does not like the SD card disappearing on it. It could be that you need to eject the card from the Mac before trying to use it via Marlin (probably a good idea anyway). It may be that there is some magic USB MSC status that we could return that would keep the Mac happy (or at least convince it to try mounting the card again later, which both Windows and Linux seem happy to do), but as far as I can tell we are returning the correct status (we basically say that the card has been removed). 

So ideally this needs more input from a Mac user (and perhaps from a Marlin developer with a Mac).

lrpirlet, 2019-07-23:
Hi @hudja 

> Hi, sorry for the late reply. Took little bit longer to get a new card than expected. I am using new 16Gb sandisc ultra formatted in FAT.

In order for a mac to share a SD disk with windows, the format MUST be exFAT... Please confirm that your SD is exFAT formated... or that your sc card is read write on BOTH MAC and Windows, implying exFAT…

_FAT16 formated implies an SDSC (Secure Digital Standard Capacity) card with a capacity up to 2 GB maximum..._ see https://en.wikipedia.org/wiki/SD_card

hudja, 2019-07-23:
Sorry for the silence, my mac is broken and I am waiting for the replacement now. Cannot confirm nor check anything immediately, will test latet.

lrpirlet, 2019-07-26:
OK, what I mean is the following…. exFAT is NOT supported in MARLIN (Marlin support FAT32)… So, you should be able to read in the printer an SD card formatted under windows using FAT32, 
Hopefully your mac is able to write on this sd card…

DO NOT use a card formated for or in a camera (it uses exFAT format)… Do use a machine running a Microsoft OS to format the card (as FAT32 is Microsoft defined  and documented standard)

boelle, 2019-09-24:
@hudja is the issue still there?

tedukes, 2019-09-25:
I am experiencing this problem using Win10 Pro. The version of Marlin is one I d/l from github on July 8, 2019. It does not use #define LPC_SD_ONBOARD but rather #define SDCARD_CONNECTION ??

boelle, 2019-09-25:
line 1071 explains things pretty good :-D

     * :[ 'LCD', 'ONBOARD', 'CUSTOM_CABLE'

tedukes, 2019-09-25:
I was just pointing out the version I'm using does not contain LPC_SD_ONBOARD. I have mine set to define SDCARD_CONNECTION ONBOARD . It will not init the SD card.

boelle, 2019-09-25:
i assume you have #define SDSUPPORT, its a strange one.

But then again i also have the re-arm (not a clone/ripoff) so its a totally different case

tedukes, 2019-09-25:
Yes, I have that defined as well.

Evg33, 2019-10-02:
I use
`#define STARTUP_COMMANDS "M22"`
in Configuration_adv.h

boelle, 2019-10-12:
@hudja is this still an issue?

boelle, 2019-10-24:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

ToaoF999, 2019-11-03:
Hi there
Got the same issue with last marlin 2.0 build (02 Nov around 10pm) and SKR 1.3 (from bigtreetech)
Just make it working
**configuration.h**
line 1622 #define SDSUPPORT
**configuration_adv.h**
line 1095: #define SDCARD_CONNECTION ONBOARD

Patag, 2019-11-03:
Same here. @boelle please stop closing issue report when it is not fixed ! You are killing Marlin !

boelle, 2019-11-03:
as said there is the option of reopen

there is no need to have solved issues open right?

boelle, 2019-11-03:
> Just make it working

so is it working or not?

p3p, 2019-11-03:
As far as I know the only reason a card will not show up as a USB Mass Storage device is that it is currently in use by Marlin, if you unmount the card `M22` or directly from the lcd menu it should become available to the OS.

[11 more replies]

closed 2019-11-03 by boelle