MarlinFirmware/Marlin #14320: [2.0.X] BigTreeTech SKR 1.3 SDCard not initializing from LCD

peppekerstens, 2019-06-18:
### Description

The SD card reader does not initialize when in stand-alone mode and calling menu item 'Init. SD card'.

### Steps to Reproduce

1. A BigTreeTech SKR 1.3 board
2. Default configuration.h and configuration_adv.h 
3. Enable #define SDSUPPORT in configuration.h

**Expected behavior:** 
A working SD card reader in both stand-alone mode (no USB connected) as well as in USB connected mode

**Actual behavior:**
- Cardreader is available as a drive when connected via USB to PC.
- Initialization fails when calling menu item "Init. SD card"

peppekerstens, 2019-06-18:
I made this issue for confirmation/documentation purposes. 

The fix is:

In Configuration_adv.h:
1. DISABLE: #define SD_DETECT_INVERTED (>line 896)
2. ENABLE: #define LPC_SD_ONBOARD (under #ifdef TARGET_LPC1768) (>line 1035)
3. ENABLE: #define USB_SD_ONBOARD (under #ifdef TARGET_LPC1768) (>line 1035)

On 1:
Remarks says:
**// This setting should be disabled unless you are using a push button, pulling the pin to ground.**
If so, why is this enabled in default configuration_adv.h...? 

On 2/3:
Remarks says:
**// LPC-based boards have on-board SD Card options. Override here or defaults apply.**
I assumed by setting #define SDSUPPORT in configuration.h I would get 'default behaviour'. Maybe I misinterpreted what is described in configuration.h, but i expected what I described under 'expected behavior'?

Issue traceback which led me to a solution; https://github.com/MarlinFirmware/Marlin/issues/12127

gloomyandy, 2019-06-18:
If you are using (and sharing) the onboard SD card reader then the SD detect state really should make no difference (and never has done until now), as the SD detect pin is disabled in this case (see the pins file https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/pins_BIGTREE_SKR_V1.3.h#L278). Are you sure that the change to SD_DETECT_INVERTING is actually required if the other settings you have changed are also made?

The default configuration for this board is to use the SD card reader that is part of many LCD displays. Personally I don't think that this is correct, but that is the way it currently is (and  will probably not change), other LPC176x boards default to use the onboard reader (whichnI think is the most sensible default).

However if you are using a recent version of Marlin it seems that some recent changes to the way that the SD card is initialised may have broken the SD card sharing code...
https://github.com/MarlinFirmware/Marlin/issues/14318
I'm about to start investigating what is going off in this case...

thinkyhead, 2019-06-18:
Duplicate of #14318

closed 2019-06-18 by thinkyhead

gloomyandy, 2019-06-18:
Are you using a recent version of Marlin (last day or two)? If so is the SD usable from the LCD on boot or do you need to select initialise SD to gain access to it? If you do have access to it does release SD make it available to your PC? Oh and what sort of PC are you using?

peppekerstens, 2019-06-18:
@gloomyandy forked on 12-6-2019; https://github.com/peppekerstens/Marlin/commit/26e7e33698a3624614f2c0c1be260be9990943f6
able and willing to get and build latest if needed (after my printer finished printing that is... :) )

@thinkyhead this is not the same case/bug. It is actually reversed: in my case USB connection is just fine, it is stand-alone functionality that is/was not working. In my point of view this is due to the fact that documentation/remarks are not clear...

peppekerstens, 2019-06-18:
@gloomyandy : on SD_DETECT_INVERTING: no, I did not test if this is mandatory for 'expected functionality'.
Will create both a new build on current fork as well as latest bits if need be..

peppekerstens, 2019-06-18:
@gloomyandy , sorry forgot to mention: Microsoft Windows 10.0.18356

gloomyandy, 2019-06-18:
@peppekerstens I'd keep the version you have at the moment. Please keep an eye on the other thread as I will probably need folks to test things once I have worked out what the problem is. At the moment the latest version of Marlin goes into a boot loop for me!

Silithas, 2019-08-31:
Same issue here. Ender 3 pro with skr 1.3 and 2209 drivers, unable to get access to the sd card at all.

Here's my config if anyone got theirs working and can tell what i got wrong. https://www.dropbox.com/s/f4gtcewkc7q05i2/Marlin.zip?dl=0

gloomyandy, 2019-08-31:
@Silithas What SD card reader are you trying to use? Do you have one attached to the LCD? If you are trying to use the onboard reader you need to set the following...
#define SDCARD_CONNECTION ONBOARD
If you want access to the card from Marlin you will need to select InitSD (or whatever it is called) from the menu. By default the onboard SD card will  be made available as a mass storage device via USB.

BTW this is an old (and closed issue) and will not be viewed by most people.

Silithas, 2019-09-01:
I'm using the sd card reader on the skr board itself.

I tried the onboard setting. Didn't work.

But i tried insane automation branch of marlin 2.0, it was able to have it read, but now i can't invert stepper direction as it now homes to the wrong directions. Tried both settings that can invert them, but nothing works.

And i did click init media, but nothing happen. But did in the insane automation branch.

Here's his branch https://github.com/InsanityAutomation/Marlin/tree/Creality_2.0_Bleeding

LuckHermsen, 2020-03-18:
> @Silithas What SD card reader are you trying to use? Do you have one attached to the LCD? If you are trying to use the onboard reader you need to set the following...
> #define SDCARD_CONNECTION [...]

This worked for me. SKR1.4 with marlin bugfix 2.0. (Ender 3)

Grimmy347, 2020-03-22:
> @Silithas What SD card reader are you trying to use? Do you have one attached to the LCD? If you are trying to use the onboard reader you need to set the following...
> #define SDCARD_CONNECTION [...]

What do you mean by select InitSD?  In Octoprint?

Grimmy347, 2020-03-22:
> This worked for me. SKR1.4 with marlin bugfix 2.0. (Ender 3)

You are able to access the SD card through octoprint to do things like update firmware?

thinkyhead, 2020-03-24:
> What do you mean by select InitSD? In Octoprint?

Marlin LCD > Main Menu > Mount Media

Ed-EE-Eng, 2020-05-20:
I tried to follow this thread as my SKR1.4 Turbo running Marlin 2.0.5.3 is screwed up.  
1) Can't INIT SD or see print files on SD card.
2) I CAN connect through serial link to Octoprint and run/print from Octo.
3) When I just plug in SKR into my Win 10 PC, it does NOT show up as an SD Drive.
    a) I can not upload new firmware.bin files to the SD card in this manor
4) When I put SKR SD card into an adaptor and plug it into PC I can see FIRMWARE.cur, etc. I have updated and recompliled Marlin and put it on the SD card (w/adaptor) as firmware.bin.
5) Put SD card back into SKR board, reboot and NOTHING.
   a) Marlin does not update
   b) fimware.bin remains and FIRMWARE.CUR seems unchanged (file date doesn't change)
   c) File update does NOT occur.  
ANY IDEAS AND WHAT I CAN DO NEXT OTHER THAN GET A NEW SKR BOARD?

Silithas, 2020-06-02:
> I tried to follow this thread as my SKR1.4 Turbo running Marlin 2.0.5.3 is screwed up.
>
> 1. Can't INIT SD or see print files on SD card.
> 2. I CAN connect through serial link to Octoprint and [...]

Can you try the build insanityautomation has on marlin 2.0? his branch made my skr 1.3 with 2209 drivers work fine

Ed-EE-Eng, 2020-06-05:
I don't think the problem is MARLIN but rather something wrong with the SKR 1.4 T board, probably the SD card socket.  Upon further inspection and review, I DO SEE the SKR board in windows (when plugged in), but no SD card.  Just a message that says insert memory card or something like that.  I can run the board via octoprint so it's not the com port either.

Suspecting a bad SD socket, I have put one on order.  In the mean time, I am looking into other ways to reprogram/update the SKR board using the Software Diag port.

div1603, 2020-06-06:
The same problem on skr 1.3. Does not see the SD card and does not load the firmware.

ellensp, 2020-06-06:
The board will always load the firmware.bin from onboard sdcard as long as board is running and bootloader has not been destroyed. In both cases it has nothing to do with Marlin.