MarlinFirmware/Marlin #14323: [BUG] [bugfix-2.0.x] [Viki2+Re-ARM] Write to SD failed with "Error:No Checksum with line number, Last Line: 0"

syurev, 2019-06-18:
Latest bugfix-2.0.x
Re-ARM
Viki2 display
SD card reader enabled on that display: #define LPC_SD_LCD 

**Writing a file to SD on Viki2 through UART failed with error:
'Error:No Checksum with line number, Last Line: 0'**

Example:
> M28 ttt.gco
> echo:Now fresh file: ttt.gco
> Writing to file: ttt.gco
> G92
> Error:No Checksum with line number, Last Line: 0
> G92
> Error:No Checksum with line number, Last Line: 0
> G92
> G92
> Error:No Checksum with line number, Last Line: 0
> Error:No Checksum with line number, Last Line: 0
> M29 ttt.gco
> Done saving file.

After that there is an empty file on the SD.
Reading file list and printing from this SD works just fine.
Write to the same SD via direct access with Panucatt WiFi Backpack also works good.
[config.zip](https://github.com/MarlinFirmware/Marlin/files/3302874/config.zip)

gloomyandy, 2019-06-18:
What program are you using to send the file to the SD card? I'm pretty sure that when you use the M28 command Marlin expects all following commands (that are to be written to the file) to have a line number and checksum (other than the terminating M29). You can see the code that enforces this here... https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/gcode/queue.cpp#L677

syurev, 2019-06-19:
I've tried to upload files via interface of Simplify3D and latest ESP3D-WEBUI.
Both with same error in console.

gloomyandy, 2019-06-19:
Do those programs have options to specify line numbers and checksums when sending code to a printer you may need to enable that. I'm pretty sure that Octoprint, Repetier host and pronterface work with M28. 

@thinkyhead should M28 allow uploads without line numbers and checksums?

thisiskeithb, 2019-06-20:
Simplify3D doesn't enable checksums by default, so you'll have to go into `Tools` -> `Firmware Configuration` -> `Communication` -> `Advanced` and check the `Include line numbers and checksums (for ASCII G-Code only)` box.

[image]

syurev, 2019-06-20:
Thank you.
Simplify3D works good with this setting for now. File uploaded without errors.
But with the old Marlin 1.1 everything worked without these settings.
I could not find any mention of checksums and line numbers in the command description at http://marlinfw.org/docs/gcode/M028.html
Seems this was not a bug but a feature. ;)

closed 2019-06-21 by syurev

thinkyhead, 2019-06-21:
> should M28 allow uploads without line numbers and checksums?

No. It is meant to capture print jobs, not to expose a general SD file utility.