MarlinFirmware/Marlin #14326: [BUG] Enabling SDCARD_SORT_ALPHA can cause watchdog reset

gloomyandy, 2019-06-18:
### Description
During boot Marlin will mount the SD card (by calling CardReader::initsd()) and call CardReader::beginautostart(), both of these call setRoot() which in turn will call presort() (if SDCARD_SORT_ALPHA is enabled) . If the sort limit has been raised and there are a large number of files then these two sort operations can exceed the watchdog time period. In my tests a single call to presort() was taking just under 3 seconds.

### Steps to Reproduce
1. [First Step]
Set the following options:
```C
  #define SDCARD_SORT_ALPHA

  // SD Card Sorting options
  #if ENABLED(SDCARD_SORT_ALPHA)
    #define SDSORT_LIMIT       128    // Maximum number of sorted items (10-256). Costs 27 bytes each.
    #define FOLDER_SORTING     -1     // -1=above  0=none  1=below
    #define SDSORT_GCODE       false  // Allow turning sorting on/off with LCD and M34 g-code.
    #define SDSORT_USES_RAM    true   // Pre-allocate a static array for faster pre-sorting.
    #define SDSORT_USES_STACK  false  // Prefer the stack for pre-sorting to give back some SRAM. (Negated by next 2 options.)
    #define SDSORT_CACHE_NAMES true   // Keep sorted items in RAM longer for speedy performance. Most expensive option.
    #define SDSORT_DYNAMIC_RAM false  // Use dynamic allocation (within SD menus). Least expensive option. Set SDSORT_LIMIT before use!
    #define SDSORT_CACHE_VFATS 2      // Maximum number of 13-byte VFAT entries to use for sorting.
                                      // Note: Only affects SCROLL_LONG_FILENAMES with SDSORT_CACHE_NAMES but not SDSORT_DYNAMIC_RAM.
  #endif
```
2. [Second Step]
Make sure the SD card has > 100 files on it in the root directory

**Expected behavior:**
System should boot normally

**Actual behavior:**
System has a watchdog reset (which on an LPC176x based system like mine results in a boot loop).

#### Additional Information
One possible fix is to reset the watchdog timer on a call to presort()...
``` C
  void CardReader::presort() {
    // Large sorts can cause the watchdog to timeout, so reinit it here
    #if ENABLED(USE_WATCHDOG)
      watchdog_reset();
    #endif

    // Throw away old sort index
    flush_presort();
```
A better solution might be to avoid the double call to setRoot or to speed up the sort (but I suspect that it may still be possible to trigger a reset given that the configuration I'm using should result in a relatively fast sort). Arguably the watchdog reset should be placed in the sort loop to allow for very slow sorts (or very large directories), but this is a little more risky as a problem with the sort would not then trigger the watchdog.

NOTE: I have these options set because without them scrolling through filenames on my display is slow/jerky. The caching provided by the sort solves this problem.

My configuration files are here:
https://github.com/gloomyandy/Marlin/tree/myskr1.3/Marlin

firepower9966, 2019-08-04:
Get same using only 40 files on Arduino Mega2560

gloomyandy, 2019-09-26:
So looking at https://github.com/MarlinFirmware/Marlin/issues/15353 it seems this is still a problem in some situations.

thinkyhead, 2019-09-27:
I will add a watchdog reset to the bubble sort so it won't time out after 4 seconds. And, I will see about optimizing the directory reading functions, as they are very wasteful.

Meanwhile, pleas make a vow never to have 100 files in a folder. Be smart and use sub-folders to organize them so you don't have to scroll for an hour everytime you want to print a file from the LCD.

closed 2019-09-27 by thinkyhead

extesy, 2019-09-27:
@thinkyhead Why use bubble sort? It is the least efficient sorting algorithm around.

gloomyandy, 2019-09-27:
@thinkyhead while I agree that keeping the number of files down,this problem can occur with far fewer than 100 files (see  #15353). In addition to the sort taking a relatively long time, it seems that having other options set can mean that a substantial part of the watchdog timeout period may have already expired before the sort is called and the time taken for the sort just pushes it over the limit. It may be worth taking a look at the startup code and considering adding further watchdog resets into that sequence.

InsanityAutomation, 2019-09-27:
Also the creality touchscreens currently do not parse subdirectories for display.... Its on my list to rewrite that code still before I get it ready to merge but for now those users are stuck...

thinkyhead, 2019-09-27:
> I'd recommend to use a shell-sort

Perhaps. Note that we can't maintain more than 2 filenames in SRAM unless there's lots of SRAM.

> this problem can occur with far fewer than 100

Sure enough. For a listing of 40 files the number of SD fetches in the current code (when not caching filenames in SRAM) is `n * (n+1)` (1640). The number of SD fetches in the new PR is `(n * (n+1) / 2) + n` (860).

> adding further watchdog resets

Or, don't init the watchdog until the end of init.

extesy, 2019-09-27:
It looks like you need a sorting algorithm to minimize a number of comparisons because that's what drives SD fetches. Shell sort looks like a great candidate from this perspective.

thinkyhead, 2019-09-29:
The shellsort is a good optimization in-general, and it's worth a try.

I rewrote a bunch of CardReader recently to better encapsulate the common working-directory actions and to get a look at how it could be better optimized.

One thing I found is that whenever you want to get an item by index from the SD card's directory, the directory has to be scanned from the start until that entry is reached. Avoiding hitting that method is always a good thing.

If you've looked at the sorting code you can see it's not hitting the directory in sequence for the sort, but picking items according to the current sort index. On the other hand, it only sorts the sort index. Anyway, it is an eccentric approach given the way that directory item fetching works. Since the directory has to be iterated, it would make most sense under iteration to find the biggest file in the current batch, put it at the bottom, and then iterate less and less for each batch.

OTOH — If we gathered an array of the unfiltered indexes of the directory items first, we could fetch any entry in the directory using `seekset(item_index * 31)` instead of going through the loop. That extra index would only cost one byte per item, but would accelerate things hugely. To be more succinct, the array really only needs to be a list of how many items to skip/keep/skip/keep... as you loop through the directory.

We'll definitely keep experimenting with all these ideas.