MarlinFirmware/Marlin #14328: [BUG] 3D-touch M401 and M402 not working.

reepotah, 2019-06-19:
### Description

Machine: Ender-3 Pro
Motherboard: SKR v1.3
Probe: Triangle labs 3D-touch connected to Zmax

The issue is, M401 and M402 commands do nothing. 
Also G28 with safe homing enabled homes to Zmin endstop without deploying the probe. Not sure if it should, but then how I'm supposed to set up it all. 

#### Also:
1. Send M401 command, state changes to "deploy: 1", but nothing actually happens.
2. Deploy pin through M280 P0 S10 or LCD menu.
3. Send G28.
4. While X is homing, blue led lights up on probe, and it actually _stows_ the pin.
5. Z axis homing proceeds to slam into the bed, ignoring Zmin endstop.

Probe can be controlled just fine with M280 command or from BL-touch menu. Machine correctly (I suppose) performs G48, G29 (linear), G29 T.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3306311/Marlin.zip)

BigIronGuru, 2019-06-20:
I don't know if this will help, but:

1. I just tried the M401 and M402 commands and my probe did deploy and stow, so there is something deeper to dig - especially as you say: **Probe can be controlled just fine with M280 command or from BL-touch menu**

2. I usually configure the probe as being my ZMIN endstop and use `#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` and connect the probe **instead** of the endstop switch.

In your case, you have separated **probing** from **homing** and you need to go step by step:

**HOMING** G28

Homing in this case DOES not use the probe. It uses the ENDSTOP. Or should.

You reported:

> Also G28 with safe homing enabled homes to Zmin endstop without deploying the probe. Not sure if it should. 

and

> While X is homing, blue led lights up on probe, and it actually stows the pin.

Yes, G28 does STOW a probe and there is no DEPLOY - a sensible thing to do when going down to god knows where.

**PROBING** G48, G29, G33

Probing uses the probe (seems logical) and you report

> Machine correctly (I suppose) performs G48, G29 (linear), G29 T

That seems to show that the probe will DEPLOY and STOW correctly from those GCODE routines.

----------

That leaves the following things to investigate:

**G28 seems to ignore the ZMIN pin**, right?
Try to activate 
```
/**
 * M43 - display pin status, watch pins for changes, watch endstops & toggle LED, Z servo probe test, toggle pins
 */
//#define PINS_DEBUGGING
```
and one could try to debug further from there. You can activate the ZMIN switch manually as a test, no need to home for testing this first step.

**M401 and M402 don't do anything**
I think that there is a bug in `probe.cpp` that needs further analysis. I think as a work-around you could enable `BLTOUCH_HS_MODE` and see if M401 and M402 work after changing that. They should work **without** BLTOUCH_HS_MODE but that definition would help diagnose the problem. You might like `BLTOUCH_HS_MODE` for other reasons though.

reepotah, 2019-06-20:
@BigIronGuru Thanks for detailed response. I just tested your suggestions, so...

`BLTOUCH_HS_MODE` Really helped, it now deploys and stows with M401/M402.

Still, If I deploy the probe with M401, then perform G28, G28 ignores Zmin endstop. If I do M401 -> M402 -> G28, then G28 is performed correctly.

BigIronGuru, 2019-06-20:
So, I know what's going wrong with **M401 and M402 don't do anything**. It just needs to be fixed, even though there is a work-around by using `BLTOUCH_HS_MODE`.

Back to the G28:
> Still, If I deploy the probe with M401, then perform G28, G28 stows the probe and ignores Zmin endstop. If I do M401 -> M402 -> G28, then G28 is performed correctly.

I understand, I hope. Are you saying: 

a. If you deploy and then stow the probe YOURSELF (M401 then M402), before doing the G28, it works. Please tell me: What triggers? The probe or your endstop switch? 
b. If you do only a deploy (M401), the G28 will not work. How far does it go?
c. If you do NOTHING and just do a G28, what happens?

Please for each case (you have DEBUGGING LEVELING turned on), so please show us the output of a G28 Z if there is any.

reepotah, 2019-06-20:
@BigIronGuru The printer is now working, so I'll try to provide all the logs later.
Anyway:
> a. If you deploy and then stow the probe YOURSELF (M401 then M402), before doing the G28, it works. Please tell me: What triggers? The probe or your endstop switch?

Zmin endstop triggers, ofc. Probe is stowed in any case, so it can't trigger.

> b. If you do only a deploy (M401), the G28 will not work. How far does it go?

All the way down. X-axis frame starts bending my poor limit switch, stepper is skipping, etc.

> c. If you do NOTHING and just do a G28, what happens?

Everything is fine, just as it should be.

Also, in any case of G28 there's slight delay before Z homing. There was none before. Though I can't say before what exactly - as I tried to install 3d-touch I updated firmware to the most recent repo clone.

BigIronGuru, 2019-06-20:
Ok. For now, don't worry about the logs. Your answers are sufficient. I'll keep you up to date on what I find out.
One more question though:

>    c. If you do NOTHING and just do a G28, what happens?
Everything goes fine, just as it should be.

So you are saying, if by any chance a poor user would do a few M40x commands by mistake, sort of, he will break the G28?

But if you turn on the printer and don't ever do a M40x unhappy sequence, the G28 is ok (and all the other stuff also).

> G28 there's slight delay before Z homing

That's probably the nearly invibile init() sequence for the probe.

reepotah, 2019-06-20:
>So you are saying, if by any chance a poor user would do a few M40x commands by mistake, sort of, he will break the G28?

M401, yes. I deliberately tried to deploy the probe before Z homing to see where(if) it will trigger.
 
I tried to figure this out by myself, and M401 and M402 are the only two commands which output smth like "deploy: 1" and "deploy: 0", so maybe it's related to the cause. That "deploy: 1" can never be set to 0 when probe is stowed by other means (not M402) or smth. So it behaves like it's Z-homing on probe. Idk, just my guess.

BigIronGuru, 2019-06-20:
Ok.
I know why M401 and M402 do not work if you don't set `BLTOUCH_HS_MODE`. I will open up a PR to fix this as work in progress as the needed action is not a simple change. For now, the work around is the `BLTOUCH_HS_MODE` and this mode is actually quite desirable.
The second issue: So this stays open, could you change the title to something like "Using M40x makes a subsequent G28 fail"?

reepotah, 2019-06-20:
@BigIronGuru Done.

#### To clarify for newcomers once again:

1. Start the printer.
2. Execute M401.
3. Execute G28.

**Expected behavior:** Homing proceeds as usual.
**Actual behavior:** Z homing slams all the way down, ignoring Zmin endstop.

**Misc:** Probe is connected to Zmax, stock Zmin endstop is present.

BigIronGuru, 2019-06-21:
Is there any chance you could (just for a quick test) do the following:

Make a backup of configuration.h and the your pins-file so you can go back easily.

Then in `configuration.h`

`#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`

comment out: `//#define Z_MIN_PROBE_PIN P1_24`

and set the Z_MIN pin number in your pins-file to P1_24: `#define Z_MIN_PIN         P1_24`

Verify the behaviour is now changed or the same.

Note, with this change the ZMIN switch is "unused" and the probe will function as ZMIN switch.

BigIronGuru, 2019-06-21:
On the other issue: You can look at PR #14352 and see the very few changes to `probe.cpp`. If you want, you can put them in right now and not wait for the PR to be commited, and report if the M401 and M402 work in your case also (even when you disable BLTOUCH_HS_MODE).

reepotah, 2019-06-21:
@BigIronGuru 
>Verify the behaviour is now changed or the same.

1. If **G28** is used after **M401**, Probe pin is stowed, then deployed again, then homing proceeds normally.
2. If just **G28** is executed, probe pin deploys and homing proceeds normally.

In both cases probe is being used by **G28** as Zmin correctly.

Don't have time to test the PR today though, will try tomorrow.

BigIronGuru, 2019-06-21:
So now I must reconfigure one of my machines, just as you have done, only in the other direction, and I will try to reproduce your problem. I will keep you up to date.

BigIronGuru, 2019-06-22:
I have identified where the problem lies and am working on a fix.

BigIronGuru, 2019-06-25:
This is fixed by #14381, you can close this

reepotah, 2019-06-25:
@BigIronGuru Thanks.

commit f090a927 (2020-04-12): Reapply BLTouch deploy/stow patch

- Fixes #14328
- Originally #14352
- Reverted in 2580104
closed 2019-06-25 by reepotah