MarlinFirmware/Marlin #14330: [BUG] G29 after G28 on delta tries to move to the side of the bed before moving down

MrMabulous, 2019-06-19:
### Description

When running G29 (configured for Bilinear auto bed leveling) on my delta in homed position, the printer first tries to move in the XY plane (which is not possible on a delta while at the top point). For successful bed leveling, I first need to manually Lower the height of the printhead down to a position where it can move on the XY plane unconstrained.

### Steps to Reproduce
1. execute steps on a DELTA printer
2. enable AUTO_BED_LEVELING_BILINEAR in the firmware
3. Home printer with G28
4. run Auto bed leveling with G29

**Expected behavior:** [What you expect to happen]

I would expect, that the printer first moves down to a height on which the printhead actually can reach the first probe point in the XY plane

**Actual behavior:** [What actually happens]

Before lowering height, the printhead immediately tries to position itself over the first probe point on one of the corners of the grid. Since it is in homed position, two of the carriages try to move past their endstops.

#### Additional Information

I can resolve the problem by first manually moving the head down to a safe height before running G29, but this should not be required. This is on 2.0.x bugfix, I did not have this problem on 1.1.9

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

mylife4aiurr, 2019-06-26:
Would you happen to know what commit your running??
On my delta I cant even get G29 bi-linear bed leveling running at all. Firmware compiles fine, doesn't warn about bed size or probe boundary issues, then after firmware is up and running throws "(L,R,F,B) out of bounds" when issuing G29
https://github.com/MarlinFirmware/Marlin/issues/14347

I've tried:
3d9d72e
07dac31
8e23e9b

I dont know how long bi-linear for Delta's has been broken since I was previously running unified bed leveling and decided to give bi-linear a try.

mylife4aiurr, 2019-06-28:
Tried 39c0c2ae last night, still broken. Gives no compile error but then complains as user tries to execute bilinear bedleveling G29.
@LVD-AC please help

LVD-AC, 2019-06-28:
Again sorry...

mylife4aiurr, 2019-06-28:
Ok thanks anyway, thanks for responding

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle