MarlinFirmware/Marlin #14331: [BUG] auto0.g waits on 'resume print' to execute

shitcreek, 2019-06-19:
### Description

Latest commit initiates SD card automatically and loads up Auto0.g but waits until you press 'Resume print' before executing

### Steps to Reproduce

1. Get latest Bugfix 2.0 (https://github.com/MarlinFirmware/Marlin/commit/5d96a6d983c065403216ed52a8c92ed22dc268ba)
2. Boot up with SD card containing auto0.g file inserted
3. See 'auto0.g' displayed

**Expected behavior:** Auto0.g executes as expected

**Actual behavior:** Nothing happens until 'Resume print'

#### Additional Information

CR-10s with default config

brisma, 2019-06-20:
I also confirm this bug.

just-jason, 2019-06-20:
Yes,I have the same problem

robbycandra, 2019-06-20:
Confirmed, its from PR #14236 .

shitcreek, 2019-06-24:
UPDATE: Scott's just patched the bug. 
https://github.com/MarlinFirmware/Marlin/commit/5ef105b2b06f2bdc82da25e858a3c374d3250c2c

closed 2019-06-24 by shitcreek