MarlinFirmware/Marlin #14332: SCARA homing moves both joints but only responds to one limit switch

corlovsky, 2019-06-20:
### Description

I am using Marlin 2.0 with SCARA type configuration with Pronterface to send the commands. 
The limit switches show up as open and not TRIGGERED when I run M119, and show TRIGGERED when I press them manually so those appear to be working.

I changed all parameters in Configuration.h according to my SCARA robot:

- MAKERARM_SCARA
- SCARA_LINKAGE_1, SCARA_LINKAGE_2
- SCARA_OFFSET_X, SCARA_OFFSET_Y
- LIMIT SWITCH PLUGS
- ENDSTOPPULLUPS
- DEFAULT_AXIS_STEPS_PER_UNIT
- HOMING DIRECTION
- Bed size

### Steps to Reproduce

1.  Use SCARA (MAKERARM) configuration
2. Try to home with limit switch configuration as X_MAX, Y_MIN
3. Pressing Home X or Home Y individually does nothing
4. The robot will only attempt to home when pressing Home All

I expect the SCARA elbow to hit its limit switch and continue homing the shoulder joint and stop when the shoulder joint has been reached

**Actual behavior:**

Both arms move at the same time, but will only respond to the Y_MIN limit switch. When I press the X_MAX limit switch, it does nothing.

[configuration_files.zip](https://github.com/MarlinFirmware/Marlin/files/3308315/configuration_files.zip)

sandrewdahl, 2019-06-20:
Yeah, I have the same problem. Any progress?

boelle, 2019-09-24:
@corlovsky is this still an issue?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle