MarlinFirmware/Marlin #14334: Thermal runaway protection does not seem to be working, at least when printer is not printing?

XOIIO, 2019-06-20:
Hello, so I've been trying to test marlins thermal runaway protection, however for some reason it seems to not be working, and I'm not sure if that's because the printer is not actively printing or because I perhaps didn't enable it properly.

Basically, I set the bed temperature to 60 degrees and held a hair dryer close to the center, and the temperature got up to 68 but did not trigger however this was very slow.

 I then set my bed temp to 25 and did the same and it rose rapidly, up to 47 before I stopped, and it still did not trigger a thermal runaway, this rise happened within 20 seconds.

Here is my current configuration.h file

`/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel 
 * Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H
#define CONFIGURATION_H_VERSION 010109

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
[... 3778 lines cut ...]
 * Sponsored by TrinityLabs, Reworked by codexmas
 */

/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Leave undefined or set to 0 to entirely disable the servo subsystem.
 */
//#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command

// Delay (in milliseconds) before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

// Only power servos during movement, otherwise leave off to prevent jitter
//#define DEACTIVATE_SERVOS_AFTER_MOVE

#endif // CONFIGURATION_H`

I'm running version 1.1.9, and everything else seems to work fine in regards to the temperature readout and control, I'm just not sure why this isn't triggering a shutdown.

I'm running on an Anet A8 Plus with the version 1.7 board. I do have octoprint hooked up but I don't see why that would interfere with this functionality.

AnHardt, 2019-06-20:
That was worked on. Please try bugfix-2.0.x

XOIIO, 2019-06-20:
Alright, I'll give that a shot, hopefully the setup isn't too different to the older version, probably forgotten a couple small details for my printer lol.

XOIIO, 2019-06-20:
Ok, so slight issue, previously I had to set the bed's temp sensor as 99 for it to work, but in the new version if I set the bed temperature to 99 I get this error output when trying to upload.

Arduino: 1.8.9 (Windows 10), Board: "Anet V1.0 (Optiboot)"

In file included from sketch\src\inc/MarlinConfigPre.h:32:0,

                 from sketch\src\inc/MarlinConfig.h:28,

                 from sketch\src\Marlin.h:24,

                 from sketch\src\Marlin.cpp:31:

sketch\src\module/thermistor/thermistors.h:147:22: error: 'temptable_99' was not declared in this scope

 #define _TT_NAME(_N) temptable_ ## _N

                      ^

c:\users\xoiio\appdata\local\temp\arduino_build_869958\sketch\src\core\macros.h:189:37: note: in definition of macro 'COUNT'

 #define COUNT(a)            (sizeof(a)/sizeof(*a))

                                     ^

sketch\src\module/thermistor/thermistors.h:148:21: note: in expansion of macro '_TT_NAME'

 #define TT_NAME(_N) _TT_NAME(_N)

                     ^

sketch\src\module/thermistor/thermistors.h:211:25: note: in expansion of macro 'TT_NAME'

   #define BED_TEMPTABLE TT_NAME(THERMISTORBED)

                         ^

sketch\src\module/thermistor/thermistors.h:212:35: note: in expansion of macro 'BED_TEMPTABLE'

   #define BED_TEMPTABLE_LEN COUNT(BED_TEMPTABLE)

                                   ^

sketch\src\module/thermistor/thermistors.h:233:11: note: in expansion of macro 'BED_TEMPTABLE_LEN'

   &&      BED_TEMPTABLE_LEN < 256 &&  CHAMBER_TEMPTABLE_LEN < 256,

           ^

sketch\src\module/thermistor/thermistors.h:147:22: error: 'temptable_99' was not declared in this scope

 #define _TT_NAME(_N) temptable_ ## _N

                      ^

c:\users\xoiio\appdata\local\temp\arduino_build_869958\sketch\src\core\macros.h:189:48: note: in definition of macro 'COUNT'

 #define COUNT(a)            (sizeof(a)/sizeof(*a))

                                                ^

sketch\src\module/thermistor/thermistors.h:148:21: note: in expansion of macro '_TT_NAME'

 #define TT_NAME(_N) _TT_NAME(_N)

                     ^

sketch\src\module/thermistor/thermistors.h:211:25: note: in expansion of macro 'TT_NAME'

   #define BED_TEMPTABLE TT_NAME(THERMISTORBED)

                         ^

sketch\src\module/thermistor/thermistors.h:212:35: note: in expansion of macro 'BED_TEMPTABLE'

   #define BED_TEMPTABLE_LEN COUNT(BED_TEMPTABLE)

                                   ^

sketch\src\module/thermistor/thermistors.h:233:11: note: in expansion of macro 'BED_TEMPTABLE_LEN'

   &&      BED_TEMPTABLE_LEN < 256 &&  CHAMBER_TEMPTABLE_LEN < 256,

           ^

exit status 1
Error compiling for board Anet V1.0 (Optiboot).

XOIIO, 2019-06-20:
Ok so problem with marlin 2.0, I've set the settings the same, temps and whatnot work, but homing is screwed up.

all of my end stops are set up as minimum, so x minimum, y minimum, and a bltouch.

I have to invert a couple of my motor directions and on malrin 1,1.9 everything works perfect.

#define INVERT_X_DIR true
#define INVERT_Y_DIR false
#define INVERT_Z_DIR true

#define X_HOME_DIR -1
#define Y_HOME_DIR -1
#define Z_HOME_DIR -1

the problem is when I set this up in 2.0 bugfix, my manual controls work the right way, but my x and y home in the wrong direction, opposite to the manual controls, but they work the right way if I un-invert the manual controls.

I can't change the home directions as everything is set up as minimum stops and it throws a bunch of errors.

XOIIO, 2019-06-20:
Ah, apparently I had inverted the endstops for the old config that's probably what's going on.

XOIIO, 2019-06-20:
Alright seems good, I'll be able to update in hopefully around 14 hours on this, can't use a hair dryer at this sort of hour.

XOIIO, 2019-06-22:
Well, gave it a try and still doesn't give an error when the bed temp changes by a fair margin when it's not actively printing.

boelle, 2019-09-24:
@XOIIO is this still an issue?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle