MarlinFirmware/Marlin #14337: Marlin 2.0 filament runout sensor error[BUG] (short description)

wonderland134, 2019-06-20:
### Description

I downloaded marlin 2.0.x ver. about one week ago.
And I upload it with filament runout sensor function.
I check filament runout sensor will be fine via pronterface.
I send the m119 command and I check open - triggered sign well.
But in the print something, filament runout script, m600 is occured when the printing started.
Eventhough, filament loaded into extruder.

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

BigIronGuru, 2019-06-20:
Absolutely sure `#define FIL_RUNOUT_INVERTING false` maybe needs to be `true`?

wonderland134, 2019-06-20:
@BigIronGuru
Yeah I'm sure.
I use mk3s filament runout sensor.
And I checked it.
filament loaded - m119 result will be filament open
filament unloaded - m119 result will be filament triggered.

If I change #define FIL_RUNOUT_INVERTING false to true, Then can get inverted result.

So I think tha value will be false.

Additionaly. I tried change that value to true.
As I saying above, pronterface m119 result inverted.
Also, print posed and filament changing problem will be occured.

talldonkey, 2019-06-20:
The MK3 and to an lesser extent MK3S are known sometimes to have unreliable filament sensors since they're optical/IR and not a physical switch.    Have you ruled out its not a hardware problem/the sensor actually giving false runout errors?  That seems more far more likely than a marlin bug at this point, until that is ruled out.  What you can try is removing filament sensor and shorting/forcing closed the filament sensor input to rule that out.

wonderland134, 2019-06-20:
@talldonkey
I know about the prusa mk3 optical sensor error issue. So, mk3s filament sensor is solve undetected problem. right??

If the my problem caused by filament sensor it self, M119 result will be wrong??

I get the result, fine result for pronterface m119 value. open and triggered

talldonkey, 2019-06-21:
MK3S is still not 100% reliable.  For example, if you print in a filament
invisible to the IR sensor, such as Esun PETG Black which surprisingly
happens to be invisible to IR then the MK3S sensor will not be reliable
either.

My point is, rule out the filament sensor so you can track it down to a
marlin code bug.

wonderland134, 2019-06-21:
@talldonkey
But, Also I have mk3s.
By prusa mk3s sensor is operate correctly.
effective differences are different board and different firmware. I use same extruder model and color.

GMagician, 2019-06-21:
> MK3S is still not 100% reliable. For example, if you print in a filament invisible to the IR sensor, such as Esun PETG Black which surprisingly happens to be invisible to IR then the MK3S sensor will not be reliable either. 

That's not correct. MK3S IR sensor doesn't detect anymore the used filament but it detect a extruder moving part (that should be printed with black PETG AFAIK).

talldonkey, 2019-06-21:
Yes, thanks for correcting me - I meant to say print the detection lever in
a filament invisible to the IR sensor.

drewzh, 2019-06-24:
I'm actually seeing issues with the latest Marlin. It would appear my filament sensor (standard cheap mechanical one, known to work) requires a inverted configuration - however FIL_RUNOUT_INVERTING is having zero effect. Status is always 'triggered' when no filament and 'open' when there's filament loaded. It's driving me mad.

BigIronGuru, 2019-06-25:
Here's a tip if anyone is having problems with `FILAMENT_RUNOUT_SENSOR`:

There is a `#define FILAMENT_RUNOUT_SENSOR_DEBUG` which will put out more information.

wonderland134, 2019-06-25:
@BigIronGuru
 I can't find information about #define FILAMENT_RUNOUT_SENSOR_DEBUG.
Can I hear about that?

BigIronGuru, 2019-06-25:
Just define it. Look at the code `runout.h` for details.
```
/**
 * Filament Runout Sensors
 * Mechanical or opto endstops are used to check for the presence of filament.
 *
 * RAMPS-based boards use SERVO3_PIN for the first runout sensor.
 * For other boards you may need to define FIL_RUNOUT_PIN, FIL_RUNOUT2_PIN, etc.
 * By default the firmware assumes HIGH=FILAMENT PRESENT.
 */
#define FILAMENT_RUNOUT_SENSOR //§§§§
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  //#define FILAMENT_RUNOUT_SENSOR_DEBUG //§§§§
  #define NUM_RUNOUT_SENSORS   1     // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.
  #define FIL_RUNOUT_INVERTING true  // set to true to invert the logic of the sensor.
  #define FIL_RUNOUT_PULLUP          // Use internal pullup for filament runout pins.
  //#define FIL_RUNOUT_PULLDOWN      // Use internal pulldown for filament runout pins.

  // Set one or more commands to execute on filament runout.
  // (After 'M412 H' Marlin will ask the host to handle the process.)
  #define FILAMENT_RUNOUT_SCRIPT "M600"

  // After a runout is detected, continue printing this length of filament
  // before executing the runout script. Useful for a sensor at the end of
  // a feed tube. Requires 4 bytes SRAM per sensor, plus 4 bytes overhead.
  #define FILAMENT_RUNOUT_DISTANCE_MM 35 //§§§§

  #ifdef FILAMENT_RUNOUT_DISTANCE_MM
    // Enable this option to use an encoder disc that toggles the runout pin
    // as the filament moves. (Be sure to set FILAMENT_RUNOUT_DISTANCE_MM
    // large enough to avoid false positives.)
    //#define FILAMENT_MOTION_SENSOR
  #endif
#endif
```

AnHardt, 2019-06-25:
@wonderland134 

> I can't find information about #define FILAMENT_RUNOUT_SENSOR_DEBUG.

Where did you look? 
This is a open source project. That means you can find all relevant information in the source code. Just grep (a directory wide search following sub-directories) for `FILAMENT_RUNOUT_SENSOR_DEBUG` and look up what it does.

> Can I hear about that?

No. Most text to speak tools fail miserably when trying to tell you 'c source code'. Reading is the better option.

wonderland134, 2019-06-25:
My marlin2.0 didn't contain about DEBUG.
Then this is reason why, filament sensor dosen't works properly?

And I think last one also means Marlin configuration.h configuration_adv.h dictionary is didn't necessity.(Also about comment in programming)
I'm just one some simple introduction about that.

BigIronGuru, 2019-06-25:
Just put the line
```
#define FILAMENT_RUNOUT_SENSOR_DEBUG
```
into `Configuration.h` where I posted it above.

Unless your Marlin bugfix-2.0.x is very old or you are using some strange fork, **I do not believe** your version does not contain `FILAMENT_RUNOUT_SENSOR_DEBUG` in the source code `/src/feature/runout.h`

wonderland134, 2019-06-25:
[image]

As you can see here.
I didn't have it.
Maybe I downloaded this version from http://marlinfw.org/meta/download/ few days ago.
In this root caused download old version?

But I can find #define FILAMENT_RUNOUT_SENSOR_DEBUG at /src/feature/runout.h .

So, Return about question, I can find why M600 present eventhough filament sensor detect filament correctly here?
I didn't catch key point.

FanDjango, 2019-06-25:
So ADD it in Configuration.adv. And VIEW the debug messages coming

AlfiQue, 2019-06-25:
Hello,

I have about the same problem

I have a machine with the MK3S head, I added on the sensor a led for a status indication

But when the machine starts, it starts purging and then stops and asks me to continue. (M600 ?)

Then at each layer change the LED of the sensor is activated and the machine stops, all I have to do is "Resume" then "Continue". (M600 ?)

I switched from a MKS base to an SKR and I tests with activating pullup and pulldown, the result is same

govido, 2019-06-30:
I can also confirm that filament runout is currently broken.
I tested SKR 1.3 board with mechanical endstop switches.
M119 reports 100% correct, manual print pausing works, but the switch is doing nothing :(

mtmello2003, 2019-07-01:
+1 !
I have the same issue here...
Using today (latest) 2.0 bugfix and the runout sensor does nothing while printing.
M119 can confirm that my two mechanical endstops are working okay.

I have two extruders with two mechanical endstops, PIN config are ok, tested with M119 with success. 

During the print runout sensor does nothing.

wonderland134, 2019-07-02:
@govido 
@mtmello2003 
Agree.
I change motherboard mks gen 1.4 to skr 1.3  But, same problem was occur.
Problem is still continue.
I look up the runout.h for filament sensor debug.
But it is not thing to help to solve the problem.
It just print filament out or in even though, I'm already test by m119 commend correctly.

govido, 2019-07-02:
@thinkyhead 
Any idea how we could track down the issue?
I would guess the sensor/endstop is polled for status change during printing, is that right?
And if the state changes (and gets recognized) it will trigger the pause script.
Is there a debug function that would help diagnosing this?

thinkyhead, 2019-07-03:
The issue can be tracked down by finding the most recent commit that still works and then examining the changes made in the following commit.

govido, 2019-07-03:
Hi,
if anyone wants to search with me:
I can confirm so far, that it was working on May 1st (9372e26) and is not working on June 1st (e4260a4).

govido, 2019-07-03:
After a few hours of testing I narrowed it down to one specific day: May 26
Between 3c179bb and 5f42d3a must be the breaking commit.
https://github.com/MarlinFirmware/Marlin/commits/bugfix-2.0.x?after=6a2a10a154c9db39d5b3732e66685953e5f945e7+226

wonderland134, 2019-07-04:
You means, If I download via to "https://github.com/marlinfirmware/marlin/archive/e1bf34b.zip", I can get the success to using optical filament sensor which is conclude prusa mk3s filament sensor??

govido, 2019-07-04:
Yes ;)

govido, 2019-07-04:
I found the breaking commit:
49e83dd
This adds BLTouch support for 3.1.
There are lots of changes inside. Maybe someone with good coding skills can take a look.
(fyi: I'm using a BLTouch with default settings in my config)

The most recent version with filament sensor working is currently 14fe41f.:
https://github.com/marlinfirmware/marlin/archive/14fe41f.zip
Problems with this version: Maxtemp error on SKR 1.3 on start, need to disable temperature checks, dual Z is broken and needs newer stepper_indirection.h file.
@thinkyhead 
This is as far as I can go to provide assistance for this (with my knowledge for now).
Hope it helps bringing Marlin 2.0 forward.

wonderland134, 2019-07-04:
@govido
What a greatness!
@thinkyhead
Always I appreciate for your effort to forwarding marlin firmware.
I hope this problem fix as soon as possible.
I want to use my sensor OTL.

Kachidoki2807, 2019-07-05:
Hi,
I had a look into the code to help @AlfiQue with this issue.

If I rely on the assumption of @govido that the breaking commit is https://github.com/MarlinFirmware/Marlin/commit/49e83dd7c81cffef7462a0a53a70a125562de83a and assuming that the BLTouch shouldn't have anything to do with the runout feature, then the only thing I can see so far that can have a side effect on the other features is configuration_store.cpp.

Maybe someone could to do an EEPROM clear to reset all the values to default and give another try.

But once more, this analysis assumes that the previous assumptions were correct.

govido, 2019-07-05:
@wonderland134 
Maybe you could try this one:
https://github.com/marlinfirmware/marlin/archive/49e83dd.zip
And report back if your runout sensor is stil working, on my machine it didn't.
That way we could double check my theory.
Thanks

wonderland134, 2019-07-05:
@govido
Yep, I will check that maybe in 2 days!
And... before the 49e83dd, any other problem exist for skr 1.3?

wonderland134, 2019-07-07:
Same problems.

InsanityAutomation, 2019-07-07:
If the eeprom state was setting it to disabled, M412 would report filament sensor off, can you check that state? Also it would likely be reporting an eeprom version mismatch on power up over the serial console. If so M502 followed by M500 would initialize it. Set M412 S1 then M500 and see what happens.

mtmello2003, 2019-07-07:
May 25 also OK (e1bf34b) ---> confirmed here: working perfectly.

boelle, 2019-07-21:
@wonderland134 and you are using bugfix-2.0.x ?

wonderland134, 2019-07-21:
Yep.
I know about that, In progress fixing.

Minims, 2019-07-21:
Hello, I have the same issue here with my mk3s sensor and MMU2. 
But it works on my other printer with a mechanical switch. 
Both are on the same latest Marlin 2.0 branch.

wonderland134, 2019-07-25:
In fact, anyone who can use the IR sensor with marlin 2.0 recent ver exist??
I guess, No.

sl1pkn07, 2019-09-08:
same error here 

https://github.com/MarlinFirmware/Marlin/issues/15200

metabug?

boelle, 2019-10-12:
@wonderland134 still having issues?

boelle, 2019-10-21:
will close this one as it seems to be a problem with hardware and not software

> And I checked it.
> filament loaded - m119 result will be filament open
> filament unloaded - m119 result will be filament triggered.

we can always reopen if it in fact is a marlin issue

closed 2019-10-21 by boelle

recoil6, 2019-12-05:
Could we please reopen this issue? there is not fix that I can find for it, I tried all the suggestions I can find but none fixed the issue.
I did try to turn on the sensor after the print had started but immediately the printer stopped  and asked to load the filament. If someone was able to find a fix please let me know.

mrbrito, 2019-12-19:
> Could we please reopen this issue? there is no fix that I can find for it, I tried all the suggestions I can find but none fixed the issue.
> I did try to turn on the sensor after the print had started but immediately the printer stopped and asked to load the filament. If someone was able to find a fix please let me know.

Are you able to use the runout sensor at all?  I ran into this issue because I am looking for a solution to why every time I start a new print, it automatically triggers a warning that there is no filament, insert, and press the button to continue.  However, once I press the button, it continues the print.  The sensor otherwise works properly, except it always starts with triggering first.

I'm using a Makerbot Mechanical Switch Endstop as the detector, when it has filament, it's very obvious because the LED is lit.  I'm also using the SKR Mini E3.  I'm wondering if this is the same, or different issue as I am experiencing.

Again, once it starts going, the sensor does function normally, if the filament runs out, the print pauses and resumes properly after load, it just always starts asking for filament to be loaded.

tpruvot, 2019-12-19:
try to invert the value of
  #define FIL_RUNOUT_INVERTING true  // Set to true to invert the logic of the sensor.

mrbrito, 2020-01-01:
> try to invert the value of
> #define FIL_RUNOUT_INVERTING true // Set to true to invert the logic of the sensor.
> 
> the runout code defy a bit the logic :p

I see what you mean about defying logic.  I assumed that when it runs out, it should "trigger" but it also makes sense that "triggered" will let it know that there is filament present.  Also, what was confusing was that it would trigger M600 when the state changed, and work properly.
Your suggestion to invert the logic corrected the issue I was having, thank you! I think I understand the filament runout logic a bit more now.

evad973, 2020-04-26:
I realise this thread is closed but it didn't seem to get resolved. I had exactly the same problem, M119 would say triggered and open when the mechanical switch was activated and M412 would say Filament runout is on, however the print wouldn't stop when the sensor was activated. 
The solution I found was to go to the configuration menu on the printer. Runout Sensor was stated as off despite M412 confirming otherwise. As soon as I activated it on the printer it worked perfectly. The board is an SKR mini E3 V1.2.

cederron, 2020-05-05:
Just to let you know guys, I was having issues with a mechanical filament runout sensor in a skr 1.3.
Read a few threads but nothing fixed it.
Finally I simply did a store settings in the LCD and boom!  working.

wonderland134, 2020-05-06:
@cederron 
What is the store settings in the LCD?
Maybe, is that LCD control store setting exactly same with M500 command?
In that case I did, but I can't get proper result.
Now I'm give up and only use other printer.

thinkyhead, 2020-05-06:
Try these commands from your favorite host:
```gcode
M412    ; sensor enabled?
M119    ; sensor triggered?
M412 S1 ; enable the sensor
M412    ; enabled?
M119    ; triggered?

M502    ; restore defaults
M412    ; enabled?
M412 S0 ; disable the sensor
M412    ; enabled?
M119    ; triggered?

M412 S1 ; enable the sensor
M500 ; Save to EEPROM
```
reboot, and go through the commands again.

creatron, 2020-05-12:
I can confirm that in the release version 2.0.5.3 this is a problem. I check M412 and filament detection is 'On' , the GUI confirms it, and M119 confirm the hardware is 100%.  I tried both true/false on the xxx_inverting, and Marlin still prints with or without filament ..

anupchavda1, 2020-05-19:
I too am having this issue on 2.0.4.4 As above, M119 is detecting correctly the filament sensor state, So too does the "Change Filament" process, as if i try to bypass the sensor, i get an option to deactivate the sensor. when i dont bypass the sensor, this menu does not appear when changing filament. However when printing from SD card (i'm eliminating the USB for now, and i trigger the Sensor, printer continues with no issue. Checked the code and i have the #define FILAMENT_RUNOUT_SCRIPT "M600"

TheDungBeetle, 2020-05-19:
I have the same problem:
- Checked the runout switch is electrically triggering as expected with multi-meter
- Confirmed cables aren't flakey with continuity testing with multi-meter
- Checked expected voltage on signal pin both with and without filament
- Created my own g-code M777 to check raw pin values i.e. without all the conditionals and inverting. This was just to quickly mis-configuration by me.
- Checked with M119 and it also reports state changes with and without filament correctly

All the above checks out good and as expected, however Marlin fails to run my FILAMENT_RUNOUT_SCRIPT

My FILAMENT_RUNOUT_SCRIPT is set to "#define FILAMENT_RUNOUT_SCRIPT "M118 A1 action:paused". But that shouldn't matter as it never gets run.

anupchavda1, 2020-05-19:
How do we go about reopening this issue? It’s obviously a software issue. Someone has even traced it back to which commit it got broken in. Marlin just doesn’t seem to trigger the action.

creatron, 2020-05-20:
I re-flashed my system with 1.1.9.. works fine, must be software..

andreslopez93, 2020-06-07:
Is this problem solved? I can't get the filament sensors to work, for a printer with mix and vtool, if M600 T0 is placed in the scrip, it works well for printers of an extruder, but not for mix printers

cskozlowski, 2020-06-07:
I have an SKR 1.4T with Mk3s IR sensor:  

M119 with no filament:  'open' status  (this is a good thing)
M119 with filament:  'TRIGGERED status (this is a good thing)
M412 S1:  filament sensor enabled
so, it appears that the filament sensor state is being read correctly; however,  it does not cause Marlin to go and execute M600 (which I have defined in config.h as #define FILAMENT_RUNOUT_SCRIPT "M600")  
So,  my sensor works properly but the script is not getting executed ... that is a problem.

andreslopez93, 2020-06-07:
469/5000
The same thing happens to me, but with the original board of my Geeetech A10M printer.
If you send the command via pronterface "M600", it will return that the tool "T" is missing, then in the scrip put like this:
#define FILAMENT_RUNOUT_SCRIPT "M600 T0"
then it works, but for printers with more than 1 extruder like mine it cannot be placed in the "T0" script since it will never change the second extruder.

check this, it may be a partial solution for you.

wat3000, 2020-06-09:
same problem

andreslopez93, 2020-06-11:
well finally the 2 filament sensors work in my A10M, I have changed the code of the files runout.h and runout.cpp

I leave the files here to help with the topic.
replace directly from the base of the Marlin and its sensors work.

note:
1_My printer is a Geeetech A10M with color mixer, and virtual tools activated, and you need to replace the 2 files to run the 2 sensors

2_If your printer fails to activate the filament sensor and it is 1 extruder only, my recommendation is to use the #define FILAMENT_RUNOUT_SCRIPT "M600 T0" directly

[runout.h + runout.cpp.zip](https://github.com/MarlinFirmware/Marlin/files/4767216/runout.h.%2B.runout.cpp.zip)

[16 more replies]