MarlinFirmware/Marlin #14343: [FR] Anycubic Trigorilla 1.1 mainboard pin definitions

arkhnchul, 2019-06-20:
new 1.1 revision of Trigorilla 8-bit AVR boards found in Anycubic printers have servo pins different from boards with no marked revision, which use RAMPS pins and have predefined board named TRIGORILLA14 in both Marlin branches.
1.1 designations: 
S0 - pin 5
S1 - pin 4
S2 - pin 11
S3 - pin 6

more details here:
https://github.com/MNieddu91/Marlin-AI3M-BLTouch/wiki/Changes-for-Trigorilla-1.1
https://github.com/MNieddu91/Marlin-AI3M-BLTouch/commit/2b4c936c1a479911b7cf25a99fce4dabfcc3c7c5

thisiskeithb, 2019-06-22:
See https://github.com/MarlinFirmware/Marlin/pull/14357.

thisiskeithb, 2019-06-22:
This has been merged, so you can now use `BOARD_TRIGORILLA_14_11` for the 1.4, Rev 1.1 board.

boelle, 2019-07-21:
closing this as the support has been added by @thisiskeithb

closed 2019-07-21 by boelle