MarlinFirmware/Marlin #14346: [BUG] on 2.0.x bugfix z_min_endstop_inverting has no effect

69kevinutah69, 2019-06-21:
I have changed the inverting flag for the z min endstop and m119 shows triggered no matter what (im using bltouch) however, if i short the endstop leads it says open. 

[Configuration - Copy.txt](https://github.com/MarlinFirmware/Marlin/files/3313300/Configuration.-.Copy.txt)
[Configuration_adv - Copy.txt](https://github.com/MarlinFirmware/Marlin/files/3313301/Configuration_adv.-.Copy.txt)

69kevinutah69, 2019-06-21:
this works as expected on 1.1.9 and 1.1.x bugfix

BigIronGuru, 2019-06-22:
The "inverting flag": I can see that the flag is forced to be "false" if you define a BLTOUCH. Look at `Conditionals_LCD.h`. It reflects the fact that the value NEEDs to be false for a BLTOUCH. This forcing was introduced some time ago in Bugfix-2.0.x.

If you short the endstop, you are setting the input pin to 0V or GND, which has the meaning: Not triggerer = OPEN. 

M119 will show the state of the probe - if the pin is up, it will show triggered and if the pin is down, it will show "open".

And now a question for you: What is the actual problem?

BigIronGuru, 2019-06-22:
@thinkyhead I think this has been discussed before - but shouldn't something this intransparent be removed and replaced by a sanity check on the order of "if bltouch and inverting is not false, then fail?"

69kevinutah69, 2019-06-22:
the problem is that no matter what i do m119 will show zmin as triggered when the circuit is open and open when the circuit is closed. this behaviour is the opposite of what it should be.

69kevinutah69, 2019-06-22:
and so what happens when i close the circuit is that the probe deploys. maybe im attributing the actual problem i was having with the continuous triggered although no previous version of marlin shows my zmin as triggered when my probe is stowed. the problem i was having is that as g29 is about to start the bed leveling fails and i assumed it was because it immediately detects zmin as being triggered instead of open

BigIronGuru, 2019-06-23:
Please note that the behavour of M119 concerning the probe was heavily modified some weeks ago (#13814). This changed the output of M119 to CORRECTLY mirror the status of the pin (UP, DOWN). Previously, sind the output of the probe is normally only a short pulse, M119 would always report OPEN.
If the pin is UP, the circuit is actually open, and M119 will report TRIGGERED. If the pin is DOWN, the circuit is closed, and M119 reports OPEN. I agree we should modify the text. I will check up what can be done.

69kevinutah69, 2019-06-23:
yes i assumed this is what was done however i believe this to be unnecessarily discontinuous from previous versions of marlin. in regards to the core issue of the failed bed leveling it randomly stopped appearing after messing around with a few different versions and then switching back to 2.0.x bugfix

boelle, 2019-07-21:
Please test with the [`bugfix-2.0.x` branch](https://github.com/
MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) so we're all on the same
page here. That branch contains all the latest patches.

boelle, 2019-09-24:
@69kevinutah69 is the issue there in 2.0.x ?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle