MarlinFirmware/Marlin #14347: [BUG] on 1.1.x bugfix and 1.1.9 a probe offset of -35 causes ther error of (L,R,F,B) out of bounds

69kevinutah69, 2019-06-21:
setting Y_PROBE_OFFSET_FROM_EXTRUDER = -35 in the config.h causes a runtime error when calling g29. the error is  (L,R,F,B) out of bounds. This is on the 1.1.x bigfix and the 1.1.9 .

i would expect this to be within bounds however i cannot seem to find where this is checked to disable the bounds check myself.

mylife4aiurr, 2019-06-23:
Bugfix 2.0 3d9d72e
same error. ? (L,R,F,B) out of bounds.

compiles ok
G33 works also, but throws error during runtime not in compiler :(

[Configuration - Copy.txt](https://github.com/MarlinFirmware/Marlin/files/3318036/Configuration.-.Copy.txt)
[Configuration_adv - Copy.txt](https://github.com/MarlinFirmware/Marlin/files/3318037/Configuration_adv.-.Copy.txt)

mylife4aiurr, 2019-06-25:
Delta configuration
Marlin 2.0 bufix 07dac315

Probe out of bounds warning after compiling done and firmware uploaded, then printer warns out of bounds when trying to use G29 , bilinear.

InsanityAutomation, 2019-06-25:
The default limits check for it to be under what default goes to, not -1 it seems. I bumped into this the other day and ill look at getting a fix in when I have some open time to poke at it again. Keep in mind just 2.0 will get the fix, not 1.1.x

InsanityAutomation, 2019-06-25:
Quick shot, 
src/module/motion.h line 258

change
return HYPOT2(rx, ry) <= sq(DELTA_PRINTABLE_RADIUS - inset);

to return HYPOT2(rx, ry) <= sq((DELTA_PRINTABLE_RADIUS + slop) - inset);

And see if that helps, as it should now pass in situations which it is probing exactly on the limit.

mylife4aiurr, 2019-06-25:
Thank u sir, I will try it tonight after work. Marlin treat Delta like step child.... (Not directed at u)

InsanityAutomation, 2019-06-25:
Because most dev's dont have any Delta machines. If people donated them as often as bed slinger cartesians are donated it might be different :)

mylife4aiurr, 2019-06-26:
8e23e9b bugfix 2.0
In motion.h, i made the suggested change. no compile errors with change. no warnings or probe boundaries complaints. 
G33 completes ok

Send: G29
Recv: ? (L,R,F,B) out of bounds.

mylife4aiurr, 2019-06-28:
Still broken on 39c0c2ae
@LVD-AC Delta guru, please help

LVD-AC, 2019-06-28:
Auto calibrate (G33) guru yes, but not auto levelling (G29) guru. Sorry...

mylife4aiurr, 2019-06-28:
Ok well thanks for responding anyway.

mylife4aiurr, 2019-07-04:
gonna try 41b7f16 and see if this is fixed

mylife4aiurr, 2019-07-04:
41b7f16 works for me now. G33 and G29

closed 2019-07-11 by thinkyhead