MarlinFirmware/Marlin #14348: TMC2209 SLAVE_ADDRESS

Gerggg, 2019-06-21:
Hello,
Can you help me understand how MS1 and MS2 jumpers work for TMC2209 in configuration_adv.h?

Four TMC2209 drivers can use the same HW/SW serial port with hardware configured addresses.
Set the address using jumpers on pins MS1 and MS2.
Address | MS1  | MS2
      0 | LOW  | LOW
      1 | HIGH | LOW
      2 | LOW  | HIGH
      3 | HIGH | HIGH

When I am the setpoint the number of steps of the drivers is not good.

Thank you

Gerggg, 2019-06-21:
I found "The slave address of the TMC2209 is selected by the pins MS1 (bit 0) and MS2 (bit 1). (SLAVEADDR=0 on TMC2208 as default)" in https://learn.watterott.com/silentstepstick/faq/

But I do not understand the difference between MS1 and MS2 for the steps in MS1 and MS2 for the slave address. There can be 2 uses for the same jumpers ...
Someone can help me ?

gloomyandy, 2019-06-21:
One (microsteps) will be in standalone mode the other (slave address) will be in UART mode. So yes the same pins can have different meaning depending upon other pin settings.

But what board do you have? You probably don't need to set different slave addresses unless you are using a single UART connected to multiple devices. Boards like the SKR V1.3 have in effect a UART per driver.

Gerggg, 2019-06-21:
I have a SKR 1.3 with active uart, so if I understood correctly, no need to configure the slave addresses. In revamp, I have to configure the steps with jumper. Without jumper I do not have the right move.

Thank you very much for your help.

gloomyandy, 2019-06-21:
Err no I didn't say that. If you are using UART mode then you probably need to configure the address and make sure it is set correctly in the Marlin configuration. But you can use the same address for each driver. Once you have the UART control working you can then use that to set your microstepping etc. 

Do you have the UART connection to your drivers working? Test with M122. If they are not working try changing the slave addess in Marlin so they are all 0, if that does not work try setting them all to 3. If you are using UART then you should probably not have any jumpers under the drivers on the SKR V1.3

closed 2019-06-25 by Gerggg