MarlinFirmware/Marlin #14356: [BUG] Bigtree SK1.3 + MAX31855 forces constant rebooting

Grogyan, 2019-06-22:
### Description

Bigtree SKR 1.3 + MAX31855 causes board to constantly reboot

### Steps to Reproduce

1. Build on 21st June 2019
2. Setup configs for Bigtree SKR1.3 + Dummy thermistor + Full Graphic  
3. Compile and upload
4. Restart board
5. Board is stable
6. Setup configs for Bigtree SKR1.3 + MAX31855 + Full Graphic 
7. Compile, upload and restart board 
8. Boards is constantly rebooting.
9. Tried again, with MAX31855 enabled, but not connected
10. Board is constantly rebooting

**Expected behavior:** [What you expect to happen]
1. Board reports on the full graphic display the current temperature read from the MAX31855 when the board is wired to the SKR 1.3
2. Board reports on the full graphic display MAXTEMP when the MAX31855 is wired to the SKR 1.3

**Actual behavior:** [What actually happens]
1.  Board is stuck constantly rebooting when the MAX31855 is wired up
2.  Board is stuck rebooting when the MAX31855 is not wired up
3.  Board is verified as working prior to setting up for MAX31855

#### Additional Information
SKR 1.3 is setup with setting the Z+ (P1_24) as the Slave Select for the MAX31855, mechanism is copied from the ReARM
Simarly with the ReARM #13854 which is also not working with the MAX31855

Not prepared to use Marlin 2.0 for RAMPS 1.4 on the MEGA2560, curretly in working condition on my I3 clone as I have lost the configuration files when set up under Marlin 1.19, as my old computer died, and the hard drive was not recoverable, last back up did not include the config files.

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
Zip also includes the bigtree skr 1.3 pins file, which has the pins setup for the slave select for the MAX31855
[SKR1-3.zip](https://github.com/MarlinFirmware/Marlin/files/3316483/SKR1-3.zip)

Grogyan, 2019-06-23:
Further information,
Board has no drivers physically installed

Downloaded the Bigtree SKR1.3 github.
Enabled the Full Graphic option only, not the MAX31855
Compile and upload
Board works, does not boot loop

Set to a dummy thermistor
Compile and upload
Full Graphic display shows the correct temperature and board does not boot loop

Compiled Marlin Bugfix as of Build on 21st June 2019 with the dummy thermistor
Upload and run
Full Graphic Display reads correctly and board does not boot loop

Change to MAX31855 (not wired to the board)
Compile, upload and run
Board begins to go into boot loop

Marlin Discord and Bigtree FaceBook have come up blank on this issue

AnHardt, 2019-06-23:
> Change to MAX31855 (not wired to the board)
Compile, upload and run
Board begins to go into boot loop

As far i can remember Marlin does not test if there is a reliable SPI communication to the MAX chips.
So it likely gets only zeros ore ones on the MiSo pin when it is not connected. That's likely to cause a Min- or Max- temp error - causing a kill - causing on some systems a reboot.
But even if we'd check if there is communication i'd call a kill when i can't find a configured sensor board. It's a catastrophic error when missing a temperature sensor.
Maybe one can play with the order in the initialization sequence, to get at least an error message.

Grogyan, 2019-06-23:
Without the board, I would expect to see MAXTEMP reported.

It is better at the moment to eliminate the board itself at this point as the source

With the board, it does the same thing, and constant boot loops.

No errors can be given as it boot loops before establishing a connection to a computer in either case.

AnHardt, 2019-06-23:
> With the board, it does the same thing, and constant boot loops.

Probably means - even than we don't have a proper SPI connection.

Grogyan, 2019-06-23:
SPI connection does work on Marlin 1.19 to the MAX31855, as that is what is currently running on my Mega2560+RAMPS setup

As there have been a lot of changes to SPI since Marlin 2.0 HAL, this suggests that the problem is there.

I dare not try Marlin 2.0 on my current 8bit Mega2560+RAMPS board, as I no longer have backups due to my old computer dying before a backup was made, in case it all goes wrong, in which case I would have no functional printer.

Grogyan, 2019-07-16:
Update on this

I have had other things to deal with these past couple of weeks, and have not been able to dig into it.
Still having issues with the BMP JTAG

Grogyan, 2019-07-21:
Requires review after #14698 is pushed to the main branch

Quick testing of only this change, shows that the boot loop is fixed.
I need to test this against an up to date version of 2.0 bugfix

Grogyan, 2019-08-27:
Tested today.

AND

I am excited to say that it is working

closed 2019-08-27 by Grogyan