MarlinFirmware/Marlin #14358: [BUG] Compile failure with latest bf2

oneofnine, 2019-06-22:
Using the latest bugfix 2 in Atom/platformio with TMC2208's getting ..

"error: 'X_SLAVE_ADDRESS' was not declared in this scope"

error (and Y_SLAVE etc) s showing up when trying to compile.

```cpp
Marlin\src\module\stepper_indirection.cpp:328:27: error: 'X_SLAVE_ADDRESS' was not declared in this scope

TMC_UART_DEFINE(SW, X, X);

^

Marlin\src\module\stepper_indirection.cpp:167:140: note: in definition of macro 'TMC_UART_SW_DEFINE'

#define TMC_UART_SW_DEFINE(IC, ST, L, AI) TMCMarlin<IC##Stepper, L, AI> stepper##ST(ST##_SERIAL_RX_PIN, ST##_SERIAL_TX_PIN, ST##_RSENSE, ST##_SLAVE_ADDRESS, ST##_SERIAL_RX_PIN > -1)

^~

Marlin\src\module\stepper_indirection.cpp:173:41: note: in expansion of macro '_TMC_UART_DEFINE'

#define TMC_UART_DEFINE(SWHW, ST, AI) _TMC_UART_DEFINE(SWHW, ST##_DRIVER_TYPE, ST, AI##_AXIS)

^~~~~~~~~~~~~~~~

Marlin\src\module\stepper_indirection.cpp:328:7: note: in expansion of macro 'TMC_UART_DEFINE'

TMC_UART_DEFINE(SW, X, X);

^~~~~~~~~~~~~~~

Marlin\src\module\stepper_indirection.cpp:328:27: note: suggested alternative: 'I2C_SLAVE_ADDRESS'

TMC_UART_DEFINE(SW, X, X);

^

Marlin\src\module\stepper_indirection.cpp:167:140: note: in definition of macro 'TMC_UART_SW_DEFINE'

#define TMC_UART_SW_DEFINE(IC, ST, L, AI) TMCMarlin<IC##Stepper, L, AI> stepper##ST(ST##_SERIAL_RX_PIN, ST##_SERIAL_TX_PIN, ST##_RSENSE, ST##_SLAVE_ADDRESS, ST##_SERIAL_RX_PIN > -1)

^~

Marlin\src\module\stepper_indirection.cpp:173:41: note: in expansion of macro '_TMC_UART_DEFINE'

#define TMC_UART_DEFINE(SWHW, ST, AI) _TMC_UART_DEFINE(SWHW, ST##_DRIVER_TYPE, ST, AI##_AXIS)

^~~~~~~~~~~~~~~~

Marlin\src\module\stepper_indirection.cpp:328:7: note: in expansion of macro 'TMC_UART_DEFINE'

TMC_UART_DEFINE(SW, X, X);

^~~~~~~~~~~~~~~

Marlin\src\module\stepper_indirection.cpp:342:27: error: 'Y_SLAVE_ADDRESS' was not declared in this scope

TMC_UART_DEFINE(SW, Y, Y);

^

Marlin\src\module\stepper_indirection.cpp:167:140: note: in definition of macro 'TMC_UART_SW_DEFINE'

#define TMC_UART_SW_DEFINE(IC, ST, L, AI) TMCMarlin<IC##Stepper, L, AI> stepper##ST(ST##_SERIAL_RX_PIN, ST##_SERIAL_TX_PIN, ST##_RSENSE, ST##_SLAVE_ADDRESS, ST##_SERIAL_RX_PIN > -1)

^~

Marlin\src\module\stepper_indirection.cpp:173:41: note: in expansion of macro '_TMC_UART_DEFINE'

#define TMC_UART_DEFINE(SWHW, ST, AI) _TMC_UART_DEFINE(SWHW, ST##_DRIVER_TYPE, ST, AI##_AXIS)

^~~~~~~~~~~~~~~~

Marlin\src\module\stepper_indirection.cpp:342:7: note: in expansion of macro 'TMC_UART_DEFINE'

TMC_UART_DEFINE(SW, Y, Y);

^~~~~~~~~~~~~~~

Marlin\src\module\stepper_indirection.cpp:342:27: note: suggested alternative: 'I2C_SLAVE_ADDRESS'

TMC_UART_DEFINE(SW, Y, Y);

^

Marlin\src\module\stepper_indirection.cpp:167:140: note: in definition of macro 'TMC_UART_SW_DEFINE'

#define TMC_UART_SW_DEFINE(IC, ST, L, AI) TMCMarlin<IC##Stepper, L, AI> stepper##ST(ST##_SERIAL_RX_PIN, ST##_SERIAL_TX_PIN, ST##_RSENSE, ST##_SLAVE_ADDRESS, ST##_SERIAL_RX_PIN > -1)

^~

Marlin\src\module\stepper_indirection.cpp:173:41: note: in expansion of macro '_TMC_UART_DEFINE'

#define TMC_UART_DEFINE(SWHW, ST, AI) _TMC_UART_DEFINE(SWHW, ST##_DRIVER_TYPE, ST, AI##_AXIS)

^~~~~~~~~~~~~~~~

Marlin\src\module\stepper_indirection.cpp:342:7: note: in expansion of macro 'TMC_UART_DEFINE'

TMC_UART_DEFINE(SW, Y, Y);

^~~~~~~~~~~~~~~

Marlin\src\module\stepper_indirection.cpp:356:27: error: 'Z_SLAVE_ADDRESS' was not declared in this scope

TMC_UART_DEFINE(SW, Z, Z);

^

Marlin\src\module\stepper_indirection.cpp:167:140: note: in definition of macro 'TMC_UART_SW_DEFINE'

#define TMC_UART_SW_DEFINE(IC, ST, L, AI) TMCMarlin<IC##Stepper, L, AI> stepper##ST(ST##_SERIAL_RX_PIN, ST##_SERIAL_TX_PIN, ST##_RSENSE, ST##_SLAVE_ADDRESS, ST##_SERIAL_RX_PIN > -1)

^~

Marlin\src\module\stepper_indirection.cpp:173:41: note: in expansion of macro '_TMC_UART_DEFINE'

#define TMC_UART_DEFINE(SWHW, ST, AI) _TMC_UART_DEFINE(SWHW, ST##_DRIVER_TYPE, ST, AI##_AXIS)

^~~~~~~~~~~~~~~~

Marlin\src\module\stepper_indirection.cpp:356:7: note: in expansion of macro 'TMC_UART_DEFINE'

TMC_UART_DEFINE(SW, Z, Z);

^~~~~~~~~~~~~~~

Marlin\src\module\stepper_indirection.cpp:356:27: note: suggested alternative: 'I2C_SLAVE_ADDRESS'

TMC_UART_DEFINE(SW, Z, Z);

^

Marlin\src\module\stepper_indirection.cpp:167:140: note: in definition of macro 'TMC_UART_SW_DEFINE'

#define TMC_UART_SW_DEFINE(IC, ST, L, AI) TMCMarlin<IC##Stepper, L, AI> stepper##ST(ST##_SERIAL_RX_PIN, ST##_SERIAL_TX_PIN, ST##_RSENSE, ST##_SLAVE_ADDRESS, ST##_SERIAL_RX_PIN > -1)

^~

Marlin\src\module\stepper_indirection.cpp:173:41: note: in expansion of macro '_TMC_UART_DEFINE'

#define TMC_UART_DEFINE(SWHW, ST, AI) _TMC_UART_DEFINE(SWHW, ST##_DRIVER_TYPE, ST, AI##_AXIS)

^~~~~~~~~~~~~~~~

Marlin\src\module\stepper_indirection.cpp:356:7: note: in expansion of macro 'TMC_UART_DEFINE'

TMC_UART_DEFINE(SW, Z, Z);

^~~~~~~~~~~~~~~

Marlin\src\module\stepper_indirection.cpp:363:27: error: 'Z2_SLAVE_ADDRESS' was not declared in this scope

TMC_UART_DEFINE(SW, Z2, Z);

^~

Marlin\src\module\stepper_indirection.cpp:167:140: note: in definition of macro 'TMC_UART_SW_DEFINE'

#define TMC_UART_SW_DEFINE(IC, ST, L, AI) TMCMarlin<IC##Stepper, L, AI> stepper##ST(ST##_SERIAL_RX_PIN, ST##_SERIAL_TX_PIN, ST##_RSENSE, ST##_SLAVE_ADDRESS, ST##_SERIAL_RX_PIN > -1)

^~

Marlin\src\module\stepper_indirection.cpp:173:41: note: in expansion of macro '_TMC_UART_DEFINE'

#define TMC_UART_DEFINE(SWHW, ST, AI) _TMC_UART_DEFINE(SWHW, ST##_DRIVER_TYPE, ST, AI##_AXIS)

^~~~~~~~~~~~~~~~

Marlin\src\module\stepper_indirection.cpp:363:7: note: in expansion of macro 'TMC_UART_DEFINE'

TMC_UART_DEFINE(SW, Z2, Z);

^~~~~~~~~~~~~~~

Marlin\src\module\stepper_indirection.cpp:363:27: note: suggested alternative: 'I2C_SLAVE_ADDRESS'

TMC_UART_DEFINE(SW, Z2, Z);

^~

Marlin\src\module\stepper_indirection.cpp:167:140: note: in definition of macro 'TMC_UART_SW_DEFINE'

#define TMC_UART_SW_DEFINE(IC, ST, L, AI) TMCMarlin<IC##Stepper, L, AI> stepper##ST(ST##_SERIAL_RX_PIN, ST##_SERIAL_TX_PIN, ST##_RSENSE, ST##_SLAVE_ADDRESS, ST##_SERIAL_RX_PIN > -1)

^~

Marlin\src\module\stepper_indirection.cpp:173:41: note: in expansion of macro '_TMC_UART_DEFINE'

#define TMC_UART_DEFINE(SWHW, ST, AI) _TMC_UART_DEFINE(SWHW, ST##_DRIVER_TYPE, ST, AI##_AXIS)

^~~~~~~~~~~~~~~~

Marlin\src\module\stepper_indirection.cpp:363:7: note: in expansion of macro 'TMC_UART_DEFINE'

TMC_UART_DEFINE(SW, Z2, Z);

^~~~~~~~~~~~~~~

*** [.pioenvs\LPC1768\src\src\module\stepper_indirection.cpp.o] Error 1

 [ERROR] Took 151.80 seconds 
```

TheNitek, 2019-06-22:
Did you do a "pio lib update"?

teemuatlut, 2019-06-22:
The issue template suggests uploading your configuration files.

oneofnine, 2019-06-22:
aahhh, I see you've sneeked some TMC slave address settings into the config adv file ;) ..

```ccp
  /**
   * Four TMC2209 drivers can use the same HW/SW serial port with hardware configured addresses.
   * Set the address using jumpers on pins MS1 and MS2.
   * Address | MS1  | MS2
   *       0 | LOW  | LOW
   *       1 | HIGH | LOW
   *       2 | LOW  | HIGH
   *       3 | HIGH | HIGH
   */
  #define  X_SLAVE_ADDRESS 0
  #define  Y_SLAVE_ADDRESS 0
  #define  Z_SLAVE_ADDRESS 0
  #define X2_SLAVE_ADDRESS 0
  #define Y2_SLAVE_ADDRESS 0
  #define Z2_SLAVE_ADDRESS 0
  #define Z3_SLAVE_ADDRESS 0
  #define E0_SLAVE_ADDRESS 0
  #define E1_SLAVE_ADDRESS 0
  #define E2_SLAVE_ADDRESS 0
  #define E3_SLAVE_ADDRESS 0
  #define E4_SLAVE_ADDRESS 0
  #define E5_SLAVE_ADDRESS 0
```

Added them to my config file and all is now OK.

Sorry about that, my bad for not updating my config files.

I highly suspect I'll be swapping the TMC2208's with TMC2209's or TMC5160's shortly .. when money allows.

closed 2019-06-22 by oneofnine

oneofnine, 2019-06-22:
> 
> 
> The issue template suggests uploading your configuration files.

That you for highlighting the problem.

On another note ..

I notice you specify that the TMC capable drivers that can do sensorless homing do it in spreadCycle mode, but from what I read the new TMC2209 does it stealthChop mode (new ability).

I guess the TMC2209 can do sensorless homing in both modes ?