MarlinFirmware/Marlin #14362: SKR 1.3 with marlin 2.0 E0 didn't work

wonderland134, 2019-06-22:
Hi
Finally I got a SKR 1.3 32bit board.
And I upload marlin 2.0 which download via to marlin github, marlin homepage, and bigtree github.(At the end, these three cases are didn't work)

Everythings are work fine. Just exist E0 stepper.

X, Y, Z axis stepper works well.
But JUST E0 STEPPER DIDN'T WORK!!

I think, special case of E0 phenomenon is below.
By LCD, motion - move axis - extruder and turn dire to move +30mm and click dire.
I remember, in normal case, extruder axis is +30mm when I reclick the motion - move axis - extruder.
But in my case, again going to 0mm.

What is the problem??
It just product problem?

I will attach the marlin 2.0 file which I rewrite base on bigtree github.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3316825/Marlin.zip)

boelle, 2019-06-22:
so this marlin you use is not from https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x ??

wonderland134, 2019-06-22:
@boelle
yep.
But I also tried from https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

boelle, 2019-06-22:
i have looked and think the board is not implemented yet, at least i cant find an example config
and i cant find the pins file either

InsanityAutomation, 2019-06-22:
The board and pins exist. Ive used them. BOARD_BIGTREE_SKR_V1_3 - 
Linear advance is not compatible with TMC2208 drivers in stealthchop. It triggers a bug in the driver ic putting it into shutdown. You need to set an OTP bit to force them to spreadcycle permanently or utilize the UART control, the board support it. For more assistance on this I recommend taking it to discord, as thats not what the issues queue is for.

gloomyandy, 2019-06-22:
Did you heat the extruder? By default E0 will be disabled unless the extruder is hot.

But as above this is a very popular board and  many, many folks are using it happily so it is unlikely that this is a Marlin bug. Try the Marlin Dsicord channel or possibly better still the Bigtreetech facebook group (which has lots of users of your board) for more help.

wonderland134, 2019-06-22:
@InsanityAutomation
You means, in this situation, the linear advanced provide the problem??

wonderland134, 2019-06-22:
@gloomyandy
yeah.
I heated hot end.
I can see the message, hotend is so cold when I didn't heat the hotend and press the move axis - extruder.

wonderland134, 2019-06-23:
InsanityAutomation
I solve the problem But I want some correct reason.
Simply I rewrite E0 driver define to DRV8825.
What about in the case of TMC2130 standalone mode? Is the same phenomenon exist??

LastDragon-ru, 2019-06-23:
AFAIK `MINIMUM_STEPPER_PULSE 1` (or 2) required for TMC2208, seems it is not set in your config.

Possible duplicate of #13535

MarkMan0, 2019-06-25:
Had a similar issue, with TMC2130 in SPI mode. The motor turned on, but it refused to move.
Try playing with MINIMUM_STEPPER_DIR_DELAY, MINIMUM_STEPPER_PULSE and MAXIMUM_STEPPER_RATE. I'm using 20, 1, 400000.

sjasonsmith, 2019-06-26:
Check MANUAL_FEEDRATE in your Configuration_adv.h to make sure there is a value set in the 5th element.

This was set to zero until a few days ago in the Ender 3 example configuration. If you happen to have started from that config, it was fixed in #13996.

wonderland134, 2019-06-27:
I'm done about this problem via to correct MINIMUM_STEPPER_DIR_DELAY, MINIMUM_STEPPER_PULSE, MAXIMUM_STEPPER_RATE.
In the case of mks gen 1.4 is normally works when didn't correct those values.
But Skr 1.3 isn't.

closed 2019-06-27 by wonderland134