MarlinFirmware/Marlin #14364: [BUG] [bugfix-2.0.x] Probing infos on LCD incorrectly shown

Bergerac56, 2019-06-22:
When building a UBL mesh (MKS SBASE with Graphic LCD) the last line of the LCD shows the probing point position xxx/xxx (right end of the last line)

There is an issue when going above point 99. The last digit does not appear.

Before 99
[image]

And when reaching 100:
[image]

Obviously the last digit would have to be 4 (111/144 and not 111/14 )

Mainly a cosmetic issue but...

boelle, 2019-09-24:
@Bergerac56 i assume this has not been fixed?

Bergerac56, 2019-09-24:
@boelle Correct ;). It is still there. Not really an issue but not clean...

Bergerac56, 2019-10-16:
It has been solved by the last bugfix version

closed 2019-10-16 by Bergerac56