MarlinFirmware/Marlin #14365: [BIG BIG BUG] Issue management (SPAM)

Patag, 2019-06-22:
I'm receiving 200+ messages saying "bad label",  "need to be closed", blablabla from @boelle !
So, even if it's a standard process to manage open issues, please, find an another way to do that. You are drowning every pertinent messages in your unnecessary comments.
Thanks in advance.
Michel

boelle, 2019-06-22:
oh bummer did not think that far

will open an issue and list issues instead

Patag, 2019-06-22:
No worries. I can understand this necessary cleaning process, but it was really "boring". I close this one right away.

closed 2019-06-22 by Patag

boelle, 2019-06-22:
problem is that only a few have the rights to close and add/remove labels

and github does not allow user rights that can do just that

but well i thought i could look through and comment what i thought could be closed etc so that the work with the upcomming RC would be more easy since the list would be clean and only have active bugs and feature  requests

Patag, 2019-06-22:
No worries again @boelle . I also think it's a prerequisite to make a clear image of Marlin issues before any official release. Sorry if my comment made you upset. 
By the way, Can't you gain these rights ?

boelle, 2019-06-22:
nope, it's either full rights or no rights (pretty much)

github does not have granular rights so each user can get just the rights they need

Patag, 2019-06-22:
OK
Bye