MarlinFirmware/Marlin #14372: [BUG] [bugfix-2.0.x] Start print from SD Card requires extra step resume print

han9988, 2019-06-22:
### Description

Start print from SD Card intruduces now a dialog wether to print or cancel.
After confirmation to print nothing happens.
Klick rotator to enter menu, select resume print and confirm selection with rotator klick starts print.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]
Print starts after confirmation of "print" in dialog.

**Actual behavior:** [What actually happens]
Nothing happens, unless you manually select resume print in menu.
#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2019-06-22:
See https://github.com/MarlinFirmware/Marlin/issues/14331

InsanityAutomation, 2019-06-22:
Try a snapshot from today. A fix was merged last night.

boelle, 2019-06-22:
and if it fails attach 2 config files as a zip file, just in case :-)

han9988, 2019-06-22:
Thanks for the quick response. I'll try tomorrow, 
there is currently a longer print running.

alexzkter, 2019-06-22:
> Try a snapshot from today. A fix was merged last night.

The latest snapshot makes my printer reboot when confirming the file to print.

Reprapdiscount full graphic and MKS GEN L on an Ender 3.

han9988, 2019-06-23:
OK, tried the latest download but problem persists.
Is there a special location for the snapshot download? I downloaded from Tab "code" select 2.0 branch and downloaded zip.

han9988, 2019-06-23:
[config files.zip](https://github.com/MarlinFirmware/Marlin/files/3317676/config.files.zip)
Here are my config files as requested earlier.

InsanityAutomation, 2019-06-23:
Just flashed todays snapshot on a machine and started an SD print without an issue. Ill review the config files above and see if anything jumps out.

InsanityAutomation, 2019-06-23:
> 
> 
> [config files.zip](https://github.com/MarlinFirmware/Marlin/files/3317676/config.files.zip)
> Here are my config files as requested earlier.

Looks like youre dealing with a REARM @han9988 but I dont see why anything would be different. Very basic SD config, no sorting ect. Id make sure the build environment is cleaned and reflash to make sure nothing was stale and try again.

Minims, 2019-06-23:
Same issue here with arduino2560, RAMPS1.6 and 12864 LCD. I'm using master branch with last commits. On dialog box, I select PRINT, and then it reboots.

shitcreek, 2019-06-23:
Confirming that the latest commit resets the machine when starting a print job via SD card. In my case with an Auto0.g file on the sd card, the machine resets itself again upon reboot then freezes unless I pull out the SD card in time.

InsanityAutomation, 2019-06-23:
What platform are you using @shitcreek?

shitcreek, 2019-06-24:
AVR

shitcreek, 2019-06-24:
UPDATE: Scott's just patched the bug.
5ef105b

han9988, 2019-06-24:
Can confirm, Scott's patch did the trick. Works now.
Thanks @thinkyhead and everybody for the quick fix.

closed 2019-06-24 by han9988