MarlinFirmware/Marlin #14374: TMC2208 error compiling error

andreashb2, 2019-06-22:
ERROR: "TMC2208 on Z2 requires Z2_HARDWARE_SERIAL or Z2_SERIAL_(RX|TX)_PIN." when compiling. Makes it impossible to use TMC2208 in UART mode with dual z-axis and mks gen v1.4

### Steps to Reproduce
1.#define MOTHERBOARD BOARD_MKS_GEN_13
2. #define Z2_DRIVER_TYPE TMC2208
3. #define Z_DUAL_STEPPER_DRIVERS
4. #define Z2_SERIAL_TX_PIN 58
#define Z2_SERIAL_RX_PIN 57
OR
4. #define Z2_HARDWARE_SERIAL Serial1

**Expected behavior:**
Should compile just fine

**Actual behavior:**

Error when compiling

Added conf, conf_adv, pins_ramps, pins_mks
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3317201/Marlin.zip)

teemuatlut, 2019-06-23:
Built fine when I just tried it with up to date bugfix-2.0.x.
Maybe someone else can verify.

Ludy87, 2019-06-23:
the same as @teemuatlut

andreashb2, 2019-06-24:
Reinstalled Atom.io. Started with a new clean marlin file. Worked. Solved!

closed 2019-06-24 by andreashb2