MarlinFirmware/Marlin #14379: #define Z_SAFE_HOMING +  #define HOMING_BACKOFF_MM == G28 will crash nozzle.

forthereg, 2019-06-23:
Will keep it as short as i can :
#define Z_SAFE_HOMING  is set to probe middle of the bed ( works wonderful).
#define HOMING_BACKOFF_MM  set to move the nozzle to the middle after XY home ( works fine).
I have to Do a XY home and then a Z home afterwards desperately. this works fine.

But if i give it a G28 , It will XY home-> move probe to middle [homing backoff] --> then do another diagonal move for [Z safe home] to move the probe to the middle even tho its ALREADY IN THE MIDDLE.
this will crash the nozzle into opposite sides of x0y0.  it will not even respect bed limits. 

so G28 needs an update in the middle ? or an absolute positioning for [safe z home] ?

btw , i LOVE the new heated chamber support , Would be nice to have it also display on LCD like bed/nozzle temps :)     
thank you for everything you've done and continue to do so   <3 <3 <3

BigIronGuru, 2019-06-23:
True, I just tested it

AnHardt, 2019-06-23:
For me this is looking like a misunderstanding of what HOMING_BACKOFF_MM does or is meant for.
For me HOMING_BACKOFF_MM is to free the endstops bz a few steps or 1/10mm. Just enough to free the (normal) endspops and or give the TMC sensorless endstops some room to accelerate before the steppers hammer the carriages against the limits again (they do need some speed to reliably detect the stall). The software endstops are set at the back upped place by purpose to avoid the hardware endstops are triggering before the software endstops.
Backing of to the middle of the bed does not make any sense to me. At least not with HOMING_BACKOFF_MM.

BigIronGuru, 2019-06-24:
I fixed it. Do we want a PR on this?

boelle, 2019-07-21:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

forthereg, 2019-07-28:
Thank you . So once again , this will do a homing backoff  AFTER the move to middle of bed for Z probe home.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3439000/Configuration.zip)

FanDjango, 2019-07-28:
I confirm that the described behaviour (**a homing backoff AFTER the move to middle of bed**) happens.

But according to @AnHardt this is not a problem because it **is looking like a misunderstanding of what HOMING_BACKOFF_MM does or is meant for**. 

Either disallow such a combination in the sanity check then, or maybe fix the code (it is a small change only).

forthereg, 2019-07-28:
True, I imagine its just a matter of changing the move commands from relative to absolute. But then again i dont know much about marlin code.
But i will say having the nozzle always rest in the middle is a super nice feature as the cables are highly strained  at max/min :)

InsanityAutomation, 2019-07-29:
My first thought for a change is to move the homing backoff into the actual homing command for an individual axis so it happens right after the switch is triggered. Maybe somewhere around line 1601 at the end of the homeaxis function in G28. For z this should check probe values to see if the raise for the probe is higher than the backoff and use the greatest number.

InsanityAutomation, 2019-07-31:
Can you test with the above PR @forthereg ?

@AnHardt im reading that as the home bump function where this is more of post home clearance. Regardless, I think this positioning for the code is more correct.

thinkyhead, 2019-08-01:
Question is whether to allow for deltas to back off on individual towers (or if this is needed) for the benefit of sensorless homing, etc. Current code assumes the back-off on Delta will only apply to Z. The new PR could allow for deltas to back off individually, and I think this would not harm delta homing, but the three backoff values must match or it breaks delta positioning (so perhaps only use the Z and ignore the XY on delta).

thinkyhead, 2019-08-01:
Question settled, and PR merged.

InsanityAutomation, 2019-08-01:
Now that the dust has settled, care to test with a current snapshot @forthereg ?

Roxy-3D, 2019-08-01:
> Question is whether to allow for deltas to back off on individual towers 

Huh?   Of course a Delta should be able to back off any tower it thinks makes sense.

InsanityAutomation, 2019-08-06:
At this point if OP doesnt get back to us, based on whats been done id call this closed.

forthereg, 2019-08-09:
Im sorry for my total noobness but i dont even know what PR means ! is it the current built/ fork ?
I tried the current download marlin 2.0.x  if that is the one you are asking.
the issue is resolved but now its acting broken in another way  :D 
Before it did a X home ,  then Y home  then a quick diagonal dash to homing back-off location. ( as it should)
i have quick Home BTW.
now  X clicks ,  and then a slow X Backoff with HOMING_BUMP_DIVISOR still applied  ( this takes its sweet time ). then goes back to finishing Y home , and then Y backoff with  Y HOMING_BUMP_DIVISOR still applies.  ( taking its sweet time again ) .
But im glad to report the primary issue was solved .
just correct feedrate on backoff and maybe have them as in one move if possible, thank you.
wish i could have shown you a video :)

forthereg, 2019-08-09:
The feedrate issue seems to go away if printer is given a move command Before homing.
But if first command is a home command then it will do this (which i cant even wrap my head around ) .
actually you can call this closed if you like. this reminds me of the old saying that you fix a bug and it creates 3 superbugs

InsanityAutomation, 2019-08-09:
Reapplying the feedrate is an easy one. I'll try to remember to get it this weekend. As for backing off multiple together, that was breaking the feature on idex. If feedrate is fixed so it's a fraction of a second, shouldn't be a concern anymore.

InsanityAutomation, 2019-08-20:
FYI, I made it back around to the feedrate adjustment for the backoff and thats merged now.

boelle, 2019-08-26:
this one should be fixed by now and will close

can always reopen it if it should not be the case

forthereg, 2019-08-27:
thank you Sir ... Sirs

closed 2019-08-27 by forthereg