MarlinFirmware/Marlin #14382: U8g Error

SkuterGaming, 2019-06-23:
**Help!**

[Marlin 2.0]  When I try install: "RepRapDiscount Full Graphic Smart Controller",
 shows me a error:
" [.pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_hw_spi.cpp.o] Error 1 "

**What to do ?**

gloomyandy, 2019-06-23:
I think you need to post the full output around that error for anyone to be able to help.

boelle, 2019-07-21:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

and yes full output

boelle, 2019-09-24:
@SkuterGaming is the issue still there?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle