MarlinFirmware/Marlin #14383: [Possible BUG???] Non-Atomic variable access inside Stepper::isr()

Roxy-3D, 2019-06-23:
@AnHardt @Bob-the-Kuhn @thinkyhead 

I've been looking at various things trying to find the cause of the Layer Shift that some people are seeing during long prints.
I'm wondering if we have a bug in the Stepper:isr() function.    The code block below is collapsed to make it easier to see the logic.   What you see below is the code block the compiler will process for an AVR with Linear Advance turned off.   Excessive comments have been removed just to collapse the logic so it can all be seen on one page.

What I'm concerned about is the updating and usage of the variables nextMainISR, interval, and next_isr_ticks.  These three variables are updated and used within the do {...} while loop with interrupts enabled.  They are 4 byte variables so it is possible to get an interrupt in the middle of their update or usage. 

Can somebody explain to me why this is safe?    Maybe the answer is any particual isr() can't recursively nest even if interrupts are re-enabled?  Or maybe the answer is ENABLE_ISRS() only enables higher priority interrupts and until a RETI instruction is exectuted this interrupt can't refire.

Can somebody help me out on this?
```
void Stepper::isr() {

  HAL_timer_set_compare(STEP_TIMER_NUM, HAL_TIMER_TYPE_MAX);
  hal_timer_t   next_isr_ticks = 0;
  uint8_t       max_loops = 10;
  hal_timer_t   min_ticks;

  do {
    ENABLE_ISRS();

    if (!nextMainISR) Stepper::stepper_pulse_phase_isr();                  // <---<<< Interrupts are enabled here.  Is this safe???

    if (!nextMainISR) nextMainISR = Stepper::stepper_block_phase_isr();    // <---<<< Interrupts are enabled here.  Is this safe???

    uint32_t interval = nextMainISR;                                       // <---<<< Interrupts are enabled here.  Is this safe???
    NOMORE(interval, uint32_t(HAL_TIMER_TYPE_MAX));                        // <---<<< Interrupts are enabled here.  Is this safe???
    nextMainISR -= interval;                                               // <---<<< Interrupts are enabled here.  Is this safe???
    next_isr_ticks += interval;                                            // <---<<< Interrupts are enabled here.  Is this safe???

    DISABLE_ISRS();

    min_ticks = HAL_timer_get_count(STEP_TIMER_NUM) + hal_timer_t( 8 * (STEPPER_TIMER_TICKS_PER_US));

    if (!--max_loops) next_isr_ticks = min_ticks;

  } while (next_isr_ticks < min_ticks);

  HAL_timer_set_compare(STEP_TIMER_NUM, hal_timer_t(next_isr_ticks));
  ENABLE_ISRS();
}
```

gloomyandy, 2019-06-23:
Well firstly all of those vars are local vars on the stack, so even if you do get a recursive reentry into this code, you will just get a second copy of them (on the stack), not the same vars being accessed by both invocations of the code. But the most important thing is that the only time you can get re-entry into this isr is from the timer and way back at the start of the isr is this line...

Roxy-3D, 2019-06-23:
Thank You!   I agree!

closed 2019-06-23 by Roxy-3D