MarlinFirmware/Marlin #14384: Anet A8 Plus  Marlin FW Erros

Celil27, 2019-06-23:
I´m trying to add the Marlin 2.0 bugfixes to my Anet A8 Plus, but if click on Check i get always the same Error....

C:\Users\Celil\AppData\Local\Temp\arduino_build_493420\sketch\src\module\temperature.cpp: In static member function 'static void Temperature::PID_autotune(const float&, int8_t, int8_t, bool)':

C:\Users\Celil\AppData\Local\Temp\arduino_build_493420\sketch\src\module\temperature.cpp:461:54: error: 'M_PI' was not declared in this scope

                 const float Ku = (4.0f * d) / (float(M_PI) * (max - min) * 0.5f),

                                                      ^

C:\Users\Celil\AppData\Local\Temp\arduino_build_493420\sketch\src\module\temperature.cpp:465:36: error: 'pf' was not declared in this scope

                 tune_pid.Kp = Ku * pf;

                                    ^

C:\Users\Celil\AppData\Local\Temp\arduino_build_493420\sketch\src\module\temperature.cpp:466:45: error: 'Tu' was not declared in this scope

                 tune_pid.Kd = tune_pid.Kp * Tu * df;

                                             ^

C:\Users\Celil\AppData\Local\Temp\arduino_build_493420\sketch\src\module\temperature.cpp:466:50: error: 'df' was not declared in this scope

                 tune_pid.Kd = tune_pid.Kp * Tu * df;

                                                  ^

Bibliothek Robot_Motor in Version 1.0.3 im Ordner: C:\Program Files (x86)\Arduino\libraries\Robot_Motor  wird verwendet
Bibliothek SparkFun_MAX31855K_Thermocouple_Digitizer in Version 1.0.1 im Ordner: C:\Users\Celil\Documents\Arduino\libraries\SparkFun_MAX31855K_Thermocouple_Digitizer  wird verwendet
Bibliothek SPI in Version 1.0 im Ordner: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI  wird verwendet
Bibliothek U8glib im Ordner: C:\Users\Celil\Documents\Arduino\libraries\U8glib (legacy) wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Anet V1.0 (Optiboot).
[image]

shitcreek, 2019-07-12:
Are you using the Marlin provided Anet A8 Plus example configuration files?

boelle, 2019-07-21:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Celil27, 2019-07-21:
Which one is the provided configuration files?

Holen Sie sich Outlook für Android<https://aka.ms/ghei36>

boelle, 2019-07-21:
you need to send us your config files

boelle, 2019-09-24:
is the issue still there?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle