MarlinFirmware/Marlin #14389: [BUG] Error compiling for LPC1768

rmcbc, 2019-06-24:
#
# NXP LPC176x ARM Cortex-M3
#
[env:LPC1768]
platform          = https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/master.zip
framework         = arduino
board             = nxp_lpc1768
build_flags       = -DTARGET_LPC1768 -DU8G_HAL_LINKS -IMarlin/src/HAL/HAL_LPC1768/include -IMarlin/src/HAL/HAL_LPC1768/u8g ${common.build_flags}
# debug options for backtrace
#  -funwind-tables
#  -mpoke-function-name
lib_ldf_mode      = off
lib_compat_mode   = strict
extra_scripts     = Marlin/src/HAL/HAL_LPC1768/upload_extra_script.py
src_filter        = ${common.default_src_filter} +<src/HAL/HAL_LPC1768>
monitor_speed     = 250000
lib_deps          = Servo
  LiquidCrystal
  U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/dev.zip
  TMCStepper@<1.0.0
  Adafruit NeoPixel=https://github.com/p3p/Adafruit_NeoPixel/archive/master.zip
  https://github.com/mikeshub/SailfishLCD.git

### Description

After adding "https://github.com/mikeshub/SailfishLCD.git" in platformio gives an error:

SailfishLCD/SailfishLCD.h:91:10: fatal error: avr/pgmspace.h: No such file or directory

*** Temporary workarround

Remove "https://github.com/mikeshub/SailfishLCD.git" from lib_deps of LPC1768

thisiskeithb, 2019-06-24:
Try changing it to https://github.com/mikeshub/SailfishLCD/archive/master.zip.

rmcbc, 2019-06-24:
I use PlaformIO and it's the latest version with all libraries needed updated.

BigIronGuru, 2019-06-24:
I have been reverting that commit since yesterday and also reported this in the commit comments because I have the same problem for LPC1768.

Moyster, 2019-06-24:
Hmm, maybe I did something wrong but it compiles here (with your config and platformio.ini @rmcbc ) .  

Looking at this commit tho :  
https://github.com/mikeshub/SailfishLCD/commit/e35d62cfcf84b1b0541971dd9aedc91bf7d5cb65  

Make sure your libs are up-to-date ;) ( platformio home > left-side tabs > libraries > updates , or just delete the sailfish lib in your marlin folder and clean / rebuild)

kAdonis, 2019-06-24:
I can confirm its working now, after updating libraries

spilz87, 2019-06-25:
I had the same problem, it was solved after uninstall sailfishlcd library and let it install it automatically when compiling.

closed 2019-07-12 by shitcreek