MarlinFirmware/Marlin #14392: M281 not documented

Reywas62, 2019-06-24:
# NO SUPPORT REQUESTS PLEASE

Support Requests posted here will be automatically closed!

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. See https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question.

For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org Marlin Forum http://forums.reprap.org/list.php?415
- Tom's 3D Forums https://discuss.toms3d.org/
- Facebook Group "Marlin Firmware" https://www.facebook.com/groups/1049718498464482/
- Facebook Group "Marlin Firmware for 3D Printers" https://www.facebook.com/groups/3Dtechtalk/
- Marlin Configuration https://www.youtube.com/results?search_query=marlin+configuration on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a Bug Report at https://github.com/MarlinFirmware/Marlin/issues/new/choose).

RudyBenner, 2019-06-25:
gcode.h has a bit of info

sjasonsmith, 2020-11-11:
I see the following text in Configuration.h:
`// Edit servo angles with M281 and save to EEPROM with M500`

This seems adequate to me. It directs the user to the command number, and they can look it up in the GCode reference.

closed 2020-11-11 by sjasonsmith