MarlinFirmware/Marlin #14396: [BUG] FIL_RUNOUT_INVERTING is not affecting filament state logic

drewzh, 2019-06-24:
### Description

I'm having a problem using the filament sensor feature in the very latest bugfix-2.0.x branch (up to date as of right now). It would appear my filament sensor (standard cheap mechanical one, known to work - https://www.aliexpress.com/item/32917462715.html?spm=a2g0s.9042311.0.0.3a4b4c4dmUNKUA) requires an inverted configuration - however FIL_RUNOUT_INVERTING is having zero effect. Status is always 'triggered' when no filament and 'open' when there's filament loaded. Wiring has been double checked and is correct. State is reported ok but the logic is in reverse.

### Steps to Reproduce

Enable filament sensor. Observe that the filament sensor switches between 'open' and 'triggered' but the states are switched/reverted. Attempt to invert the logic in the Marlin config.

**Expected behavior:** [What you expect to happen]
Logic should switch so that filament loaded becomes 'triggered' rather than 'open'.

**Actual behavior:** [What actually happens]
Logic stays the same. Filament loaded still reports as 'open' rather than 'triggered'.

#### Additional Information

Please see my current config here: https://github.com/drewzh/Marlin/tree/bugfix-2.0.x

timmmoore, 2019-06-25:
I was looking at the runout code the other day and had a question about it. The answer may resolve this bug
In endstops.cpp you see this code
      **print_es_state(READ(FIL_RUNOUT_PIN) != FIL_RUNOUT_INVERTING, PSTR(MSG_FILAMENT_RUNOUT_SENSOR));**
so it handles FIL_RUNOUT_INVERTING
in runout.h you have
    **static inline uint8_t poll_runout_states() {
      return poll_runout_pins() ^ uint8_t(
        #if DISABLED(FIL_RUNOUT_INVERTING)
          _BV(NUM_RUNOUT_SENSORS) - 1
        #else
          0
        #endif
      );
    }**
Since in cofnguration.h you have
  **#define FIL_RUNOUT_INVERTING false // Set to true to invert the logic of the sensor.**
I think the line in runout.h
        #if DISABLED(FIL_RUNOUT_INVERTING)
will always fail, FIL_RUNOUT_INVERTING is either true or false if is never DISABLED.
Doesn't that line in runout.h need to be changed to
        #if FIL_RUNOUT_INVERTING == true

BigIronGuru, 2019-06-25:
@drewzh When I change the inverting flag from true to false, the M119 output duly reverses.

Prove to yourself that the configuration change, the compile and then the flash is actually happening by changing your `#define CUSTOM_MACHINE_NAME "Ender 3 Pro"` to some other name, maybe, as a first step.

BigIronGuru, 2019-06-25:
@timmmoore Works for me.

But here's a tip if anyone is having problems with `FILAMENT_RUNOUT_SENSOR`:

There is a `#define FILAMENT_RUNOUT_SENSOR_DEBUG` which will put out more information

drewzh, 2019-06-25:
OMG, where is the 'delete and not speak about this again' button??

I've been flashing over Octoprint and the flashing has been broken for the past few days! I'm such a numpty.

2 good things however:

1) There's no issue
2) I've finally got the filament sensor working!

I'll close this issue and hide my head in shame. Thanks for the responses everyone.

closed 2019-06-25 by drewzh

thinkyhead, 2019-06-26:
> will always fail, `FIL_RUNOUT_INVERTING` is either `true` or `false`… [it] is never `DISABLED`.

If something is `false` then it is also `DISABLED`. Using `#if !FIL_RUNOUT_INVERTING` would also work, even if it is not defined. Preprocessor rules are special.

timmmoore, 2019-06-26:
Thanks, I chased all the way through the macros and I think I understand how they work now