MarlinFirmware/Marlin #14399: 1.1.9 Last stable TMC 2130 No-Movement

Wdfitzpat007, 2019-06-25:
Just finished installing TMC2130's following this guide: https://toms3d.org/2017/12/09/tmc2130-guide/

I have double and triple checked that my wiring matches the attached pictures (in the zip).

I have a: "#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER" with encoder knob and an SD card slot. However it does not use the AUX 3 port on my MKS Gen L 1.0 board. Since I control the printer via computer I commented out the above line, and unplugged the screen from the board even though it does not use my AUX 3 port. I tried an M122, which shows that none of the 2130's are communicating via SPI.

Had to do this on short notice because one of my drivers failed while I was filling a contract and I could not find any documentation relevant to my particular problem. Please help.

Connection initialization to printer:

```
Connecting...
start
Printer is now online.
echo:Marlin 1.1.9
echo: Last Updated: 2018-08-01 | Author: (none, default config)
echo:Compiled: Jun 25 2019
echo: Free Memory: 4897  PlannerBufferBytes: 1232
echo:V55 stored settings retrieved (655 bytes; crc 62494)
echo:  G21    ; (mm)
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z4000.00 E500.00
echo:Maximum feedrates (units/s):
echo:  M203 X150.00 Y150.00 Z5.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P1500.00 R3000.00 T3000.00
echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 Q20000 S0.00 T0.00 X5.00 Y5.00 Z0.30 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00
echo:Stepper driver current:
echo:  M906 X1000 Y1000 Z1000
  M906 T0 E1000
echo:Sensorless homing threshold:
echo:  M914 X8 Y8 Z8
```

M122 gives:
```
SENDING:M122
		X	Y	Z	E0
Enabled		false	false	false	false
Set current	1000	1000	1000	1000
RMS current	550	550	550	550
MAX current	776	776	776	776
Run current	17/31	17/31	17/31	17/31
Hold current	8/31	8/31	8/31	8/31
CS actual		31/31	31/31	31/31	31/31
PWM scale		255	255	255	255
vsense		1=.18	1=.18	1=.18	1=.18
stealthChop	true	true	true	true
msteps		0	0	0	0
tstep		4294967295	4294967295	4294967295	4294967295
pwm
threshold		0	0	0	0
[mm/s]		-	-	-	-
OT prewarn	true	true	true	true
OT prewarn has
been triggered	false	false	false	false
off time		15	15	15	15
blank time	54	54	54	54
hysteresis
-end		12	12	12	12
[... 249 lines cut ...]
   *   stepperY.interpolate(0); \
   * }
   */
  #define TMC_ADV() {  }

#endif // TMC2130 || TMC2208
```

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Wdfitzpat007, 2019-06-25:
Just tried latest bug fix.

Patag, 2019-06-25:
Your 2130 don't communicate properly with your board. You shouldn't get only binary 1 in the drivers registers in M122 report.
```
  #define TMC_USE_SW_SPI
  #define TMC_SW_MOSI       -1
  #define TMC_SW_MISO       -1
  #define TMC_SW_SCK        -1

```
MOSI, MISO and SCLK are undefined (-1). Try to comment out these ones assuming they are defined in your pins_MKS_GEN_L.h file

Wdfitzpat007, 2019-06-25:
AUX 3 is inverted on the MKS Gen L 1.0, after looking at someone with the same problem I moved pins and the problem went away. I have a new problem with /kill getting called when sensorless homing X and Z, but not Y in addition to a ton of busy responses during homing on pronterface. For those looking for a solution here is a PDF of the MKS Gen L 1.0. Happy printing! *salute* 07

[MKS_Gen-L_TMC2130_SPI_Sensor-less_Homing_Wiring_Diagram.pdf](https://github.com/MarlinFirmware/Marlin/files/3326957/MKS_Gen-L_TMC2130_SPI_Sensor-less_Homing_Wiring_Diagram.pdf)

closed 2019-06-25 by Wdfitzpat007