MarlinFirmware/Marlin #14403: Beaglebone black + Replicape Rev B hardware support

buttigieg94, 2019-06-25:
Beaglebone black + Replicape Rev B board is not supported by Marlin. Would really be nice considering the hardware capabilities it can provide.

marcio-ao, 2019-07-17:
I have actually pondered this idea myself. WIt would not make sense to run Marlin on the ARM core, since that normally runs Linux; but it would be amazing if Marlin could run on one of the PRU, as those are essentially micro-controllers. So basically you would have a Linux and host software on the ARM CPU and Marlin on the PRU. It would be kinda awesome.

But this is kind of a longshot for a feature request. Somebody with a lot of knowledge about the AM3358 would have to do the porting work. People with the requisite skills would probably support another project like [redeem](https://www.thing-printer.com/redeem/) rather than spending time with Marlin. But we can hope...

EvilGremlin, 2019-07-17:
Just quick look at it tells that there couldn't be good solution: [http://beagleboard.org/support/bone101/#headers](http://beagleboard.org/support/bone101/#headers)
There isn't enough PRU pins to run printer, and most them are taken by HDMI already. And even without HDMI there is two PRUs. And no ADC for PRUs.
Great task for sport programming though :)

gladio145, 2019-07-17:
@EvilGremlin 
you're forgetting the Replicape Rev B board that is connected with the beagle bone. You need to look at that for pins related to running a printer and not beagle bone board alone

marcio-ao, 2019-07-17:
@EvilGremlin: However, the BeagleBone does have analog inputs. It's not clear how accessible they are from the PRUs, but I did find this:

https://ln1985blog.wordpress.com/2015/10/21/beaglebone-black-read-adcs-from-pru/

So, it's possible I guess? At the worst case, it might require some damon running in the ARM side of things to read temps and shuffling them to the PRUs in some manner. The lack of pins is probably a greater issue, but how many are needed anyway? I'm thinking a STEP/DIR per axis (8-10), endstops (0-3), heaters (2-3) and maybe SPI for everything else (4-5). So 14 pins on the low end, to maybe 21 on the high end.

It seems like one PR has exactly 14 pins, so just barely for the minimum configuration. You would need to use a Trinamic chip so you could use sensorless homing and not need endstops. With the Trinamic chips, it would even be possible to configure the stepping direction via SPI, although I do not know how performant that would be.

marcio-ao, 2019-07-17:
@gladio145: The Replicape is not designed for having the entire printer firmware running on the PRUs, it was designed for redeem. The way redeem works (AFAIK) is that the PRUs are only used for the steppers, everything else (motion planning, endstops, temperature control) runs on the ARM chip. Certainly Marlin could be split up into two parts like that, but that would be even a more ambitious undertaking than simply cramming the whole thing into a PRU and calling it good.

marcio-ao, 2019-07-17:
Anyhow, I think @EvilGremlin has a point. Someone would have to do this for "sport" rather than practicality. It would take someone with a heavy understanding of AM3358, with an interest in 3D printing and a willingness to take the hard path of porting Marlin rather than building on redeem. Multiply those probabilities out and I think there is maybe one person in the whole world who may be able to do this. We just need to find him or her...

EvilGremlin, 2019-07-17:
I'm aware. But we're talking about Marlin here. Replicape is closer to Klipper, but kinda spread out. ADCs and PWMs are run by linux, endstops should be as well. So PRU handles a little more than step/dir. And all that with ton of port expanders.

EvilGremlin, 2019-07-17:
Ahaha, i'm writing too slow :)