MarlinFirmware/Marlin #14404: fan and heat bed works together

barisyalcinfree, 2019-06-25:
Hello everyone. i have marlin on arduino mega with ramps 1,4. the problem is when i open the fan heat bed starts heating automaticly while printing or none. i send m106 code on pronterface bed start to heating up to 100 degrees and more. Never stops. Please help.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3326765/Marlin.zip)

barisyalcinfree, 2019-09-24:
Yes. İ connected fans external power supply    
And on print setting im closing cooling

boelle, 2019-09-24:
have you checked that everything is connected correct?

https://raw.githubusercontent.com/wiki/revarbat/snappy-reprap/v2.0-RAMPS_Wiring_For_Snappy.png

its for the snappy but its good enough, you can disregard the MAX switches

barisyalcinfree, 2019-09-24:
Yes same connection

boelle, 2019-10-14:
your ramps pins file is not the one from https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/ramps/pins_RAMPS.h

where did you get marlin from?

boelle, 2019-10-14:
and the beginning of configuration.h does not look like the start of https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/Configuration.h they should be the same down to line ~40