MarlinFirmware/Marlin #14405: SKR 1.3 #define PIDTEMPBED causes crash (bugfix-2.0.x)

RideOn123, 2019-06-25:
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/3326866/Config.zip)

### Description

Board shows up as USB Stick, disappears after about a minute.
Windows shows an error, that the last connected USB Device doesn't work as intended.
Pronterface cannot connect, shows the same error.
I cannot upload new firmware via USB, I have to use an external cardreader.

### Steps to Reproduce

Enable PIDTEMPBED

**Expected behavior:** [What you expect to happen]
It working

**Actual behavior:** [What actually happens]
See Description
[image]

ManuelMcLure, 2019-06-25:
How old is your checkout of the bugfix-2.0.x branch? There were some recent (a couple weeks ago) fixes to solve that sort of issue.
Is the bed thermistor connected correctly?

RideOn123, 2019-06-26:
> How old is your checkout of the bugfix-2.0.x branch? There were some recent (a couple weeks ago) fixes to solve that sort of issue.
> Is the bed thermistor connected correctly?

It happend first with the version of June 9th, but i downloaded the newsest yesterday and copied my config files. Still the same error. I think, the thermistor is connected correctly, i will check tomorrow, but even if it isn't, this should not produce this kind of error, right? Since it worked with PIDTEMPBED disabled.

ManuelMcLure, 2019-06-26:
You should get an error, but if you don't have an LCD to see the error it might be impossible to see in the console because the error might occur before serial communications start up.

As a test, you can add
```
#define BOGUS_TEMPERATURE_FAILSAFE_OVERRIDE
```
to your ``Configuration.h`` which should allow you to connect and use ``M105`` to see what your thermistors are reporting. Note that this is just a _**temporary**_ change to help debug this - you do _**not**_ want to leave this set as it will disable thermal protection!!!

RideOn123, 2019-06-27:
I got my old Display out and connected it. You are right. I didn't plug my thermistor all the way in. Thanks a lot!

closed 2019-06-27 by RideOn123

john4242, 2019-09-18:
Just a comment: I had a short circuited thermistor (zero ohms). Effect was the same: no USB drive showed up, no communications. (I didn't have an LCD either.)