MarlinFirmware/Marlin #14410: [BUG] (short description) Z homing and bilinear leveling grid non coherent

spilz87, 2019-06-26:
### Description

There is something weird between Z homing and bilinear leveling grid.

### Steps to Reproduce

0. Turn off and on the printer (to reset everything)

1. Check the Z probe offset value is ok

2. Do a homing (my Z homing is done in X100 Y100)

3. Do an autobedlevel with G29
(My configuration file :
#define AUTO_BED_LEVELING_BILINEAR
...
#define GRID_MAX_POINTS_X 3
#define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X

#define LEFT_PROBE_BED_POSITION 50
#define RIGHT_PROBE_BED_POSITION 150
#define FRONT_PROBE_BED_POSITION 50
#define BACK_PROBE_BED_POSITION 150
)

4. Check the bilinear leveling grid report on serial

**Expected behavior:** [What you expect to happen]

As Z homing is done on X100 Y100
As the 5th point of the bed leveling measurement (the center of the grid 3x3) is done on X100 Y100 too, I expect a 0

**Actual behavior:** [What actually happens]

Result : 

Bilinear Leveling Grid:
0      1       2
0 -1.343 -1.023 -0.774
1 -1.112   -0.814 -0.339
2 -0.809 -0.266 -0.034

spilz87, 2019-06-27:
I précise I did it several time and had always this phenomena, so it is not an error from a bad probing sensor

gloomyandy, 2019-06-27:
Do you have a display? If so try the following....

So when you home your machine, what reading do you see on the display after homing? Is it 100, 100 or something else? When your machine performs the G29 what does the display read when doing the centre point? It could be that one operation is using the nozzle position while the other is using the probe position. If that is the case then the probing points will be different by whatever your probe offset is.

What are your probe offsets set to?

Also please always include your configuration files in an issue report, it makes it much easier to check things like the type of printer (Cartesian, CoreXY, Delta etc) and your probe offsets, how you have specified homing etc.

spilz87, 2019-06-27:
Thanks for your reply,

It’s on an “Ultimaker like” printer. 
The probe I use is a BLTouch V2.0 (real one)

It’s not a problem with probe offset setting : 
Z homing is done on the middle of the bed (I have a grid drown on the bed) and on screen it’s not X100 Y100 because of x and y offset. 

And even if the offsets were wrong, they still are the same for bed leveling. 
I checked during bed leveling, the 5th point (center of the matrix 3x3) is done exactly at the same point on the bed than Z homing, that’s the point.

spilz87, 2019-06-27:
my config files 

[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/3334644/Configuration_adv.h.txt)

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/3334638/Configuration.h.txt)

gloomyandy, 2019-06-27:
Thanks, so it looks like you are using Marlin 2, how recent a version is this?

spilz87, 2019-06-27:
Yes, you’re right i forgot to specify it 
I did a merge 5 days ago

spilz87, 2019-06-27:
I can’t really define when this issue appaire.

gloomyandy, 2019-06-27:
I can't test this at the moment, but I did run some quick tests earlier using UBL (not bilinear) and saw no problems. A few things that may provide useful input....
1. Can you check that repeated probings on the same point give the same results using M48 (http://marlinfw.org/docs/gcode/M048.html), you will need to turn on this feature if not already enabled.
2. Try turning on debug output during levelling using M111 see http://marlinfw.org/docs/gcode/M111.html again you will probably need to enable DEBUG_LEVELING_FEATURE and rebuild etc.
Hopefully one of these will provide some indication as to what is going on.

One final thought is that there has been a lot of activity around the BLTOUCH recently. It may be worth checking that your configuration for the probe is still correct (there may be new/different settings that need to be configured). https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/Configuration_adv.h#L527

spilz87, 2019-06-27:
Test of "M48 P10 X100 Y100 V2" : 
Mean: -0.027250 Min: -0.046 Max: 0.010 Range: 0.055
Standard Deviation: 0.018108

very far of -0.814

reliability of the probe doesn't seem to be the explaination

gloomyandy, 2019-06-27:
Hmm a range of 0.055 is relatively high, but as you say nothing like -0.814. So it looks like we need to see the debug output.

spilz87, 2019-06-28:
after testing with debug output, it appeare that this wird mesures are the consequences of an other part : 
I got got this mesure if I do 2 times G29 without G28 between .

but this difference is because this part of code : 
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/gcode/bedlevel/abl/G29.cpp#L959

removing this line solved all the issue I actually have with bedlevling bilinear

do you have any idea why ??

I will maybe open a new issue for that

thanks for your help @gloomyandy

gloomyandy, 2019-06-29:
Sorry I've never really looked at the ABL code (I use UBL).

@Roxy-3D Any thoughts on this?

DaVincIan, 2019-07-01:
Dear Splitz87

My question is not relating to your "bug". Can you share with us how to configure a printer to zero in the centre of the bed (in your setup X100 Y100) and not at the traditional X-home Y-home position.

Thanks
Ian

spilz87, 2019-07-01:
By using 
#define Z_SAFE_HOMING

#define Z_SAFE_HOMING_X_POINT 100 
#define Z_SAFE_HOMING_Y_POINT 100

spilz87, 2019-09-24:
I’m not sure it’s really resolved but I found a way that it works 
I think we can close it

closed 2019-09-24 by spilz87