MarlinFirmware/Marlin #14413: [FR] Porting Marlin to the new STM32MP1 MCU series?

Gamester17, 2019-06-26:
Can I suggest that you port Marlin firmware to run the new STM32MP1 MPU series?
https://blog.st.com/stm32mp1-mpu-stm32mp157a-ev1-stm32mp157c-dk2/

I am only an end-user / technology enthusiast myself and I do not actually know if anyone is already working on a 3D-printer controller board based on the new STM32MP1 MPU series or not, however, I read that STMicroelectronics claims STM32MP1 architecture enables developers to use the same software as STM32 MCU series.

For technical specification summery checkout:
https://www.cnx-software.com/2019/02/21/stmicro-stm32mp1-cortex-a7-m4-mpu/

STM32MP157A-DK1 and STM32MP157C-DK2 Discovery kits are $99 with LCD or $69 without
https://www.st.com/en/evaluation-tools/stm32mp157a-dk1.html
https://www.st.com/en/evaluation-tools/stm32mp157c-dk2.html
STM32MP157A-EV1 Evaluation kit for PCB engineers does have a steeper price at $399
https://www.st.com/en/evaluation-tools/stm32mp157a-ev1.html

First read about this MCU/MPU in the release news about PanGu board from I2Som
https://www.cnx-software.com/2019/06/25/pangu-board-stm32mp1-sbc/

Gamester17, 2019-06-26:
Theoretically, with the STM32MP1 MPU series you run both Marlin and OctoPrint on the same board.

Gamester17, 2020-07-01:
FYI, STMicro Updates STM32MP1 family with faster 800MHz Cortex-A7 processors, though the Cortex-M4 core still runs at 209 MHz

https://www.cnx-software.com/2020/02/27/stmicro-updates-stm32mp1-family-with-800-mhz-cortex-a7-processors/

Hedda, 2021-02-12:
STM32MP157A-DK1 is still probably best board to try but some more STM32MP1 boards (and SoM modules) are now available:

Seeed Studio ODYSSEY-STM32MP157C SBC based on Seeed Studio SoM-STM32MP157C SoM 

https://www.cnx-software.com/2020/01/21/odyssey-stm32mp157c-sbc-feature-stmicro-stm32mp157c-cortex-a7-m4-soc/

Emtrion emSBC-Argon 

https://www.cnx-software.com/2019/07/18/emtrion-emsbc-argon-stm32mp1-dual-cortex-a7-m4-processor/

i2Som PanGu board from I2Som:

https://www.cnx-software.com/2019/06/25/pangu-board-stm32mp1-sbc/

MYIR MYD-YA157C development board based on their MYiR MYC-YA157C STM32MP1 CPU Module

https://www.cnx-software.com/2020/01/07/myir-launches-som-development-board-based-on-stm32mp1-microprocessor/

Avenger96 96Boards CE Extended Board Features an STMicro STM32MP1 SoM

https://www.cnx-software.com/2019/02/23/avenger96-96boards-stmicro-stm32mp1-som/

Shiratech Stinger96 board (96Boards board design)

https://www.cnx-software.com/2020/05/15/stinger96-nb-iot-emtc-and-avenger96-96boards-stm32mp1-sbcs-giveaway/

Direct Insight / Ka-Ro STM32MP1 QSMP 1570 / 157C SoM module and development kit with thier reference devboard/devkit:

https://www.cnx-software.com/2020/06/26/linux-powered-stmicro-stm32mp1-solder-down-qfn-style-som-measures-27x27mm/

Hedda, 2021-06-23:
FYI, Olimex has just announced the availability of three "STMP15X-SOM" SoM (System-on-Module) models based on STM32MP151, STM32MP153, and STM32MP157 as well as compatible evaluation board (carrior board / breakout board) called "STMP1(A13)-EVB".

https://www.cnx-software.com/2021/06/21/olimex-stm32mp1-som-and-evaluation-board-support-linux-5-10/

The evaluation board (carrior board / breakout board) is released as open source hardware with schematics on OLIMEX GitHub:

https://www.olimex.com/Products/SOM/STMP1/STMP157-SOM-512/

https://www.olimex.com/Products/SOM/STMP1/STMP1-A13--EVB/open-source-hardware

https://github.com/OLIMEX/STMP1-A13-EVB

STMP157-SOM-512-IND and STMP157-SOM-512-EXT (the SOM models based on the faster STM32MP157) cost €30, while slower SoMs based on STM32MP151 and STM32MP153 cost €20 and €25 respectivly, and the matching evaluation board cost about €20. 

STMP157-SOM-512-IND has an internal WiFi antenna and STMP157-SOM-512-EXT has an external WiFi antenna. Both of those have dual-core ARM Cortex-A7 @ 800MHz for Linux OS plus a single-core ARM Cortex-M4 microcontroller @ 209 MHz for real-time.

[image]
[image]
[image]