MarlinFirmware/Marlin #14414: Arduino generated .ELF files

Roxy-3D, 2019-06-26:
Does anybody know how to get Arduino to leave the .ELF file laying around after Marlin gets built?

I can't find it anywhere.   I might be looking in the wrong place.   Or maybe Arduino is deleting it after the build?    

If anybody has an AVR .ELF file of Marlin they could post, that would be very helpful!

Thank You!

thisiskeithb, 2019-06-26:
Here's a random compile with some Tarantula Pro firmware: [Marlin.ino.elf.zip](https://github.com/MarlinFirmware/Marlin/files/3331072/Marlin.ino.elf.zip)

If you enable verbose output, it should show you where the files are located:

```
Linking everything together...
*snip*
/var/folders/6b/2_s32tjd6p509ym4t6lhf_f00000gn/T/arduino_build_67508/Marlin.ino.elf
Sketch uses 117108 bytes (46%) of program storage space. Maximum is 253952 bytes.
Global variables use 4464 bytes (54%) of dynamic memory, leaving 3728 bytes for local variables. Maximum is 8192 bytes.
```

Roxy-3D, 2019-06-26:
@thisiskeithb    THANK YOU!!!!!

closed 2019-06-26 by Roxy-3D