MarlinFirmware/Marlin #14415: [FR] (feature request title)

HassanSaied, 2019-06-26:
### Description
I have a custom made 3D printer, with Marlin 1.17 firmware installed. It's using RAMPS 1.4 along with an arduino Mega. I wish to use it as a 3D scanner, and It is necessary that the 3D printer gives me some feedback. A simple signal to the host machine would be more than enough.
### Steps to Reproduce
So the feature should work like this
While(Stepper motor is stepping) do
          Stepper.step
          if axis = x SERIAL_ECHO(1)
          if axis = y SERIAL_ECHO(2)
          if axis = z SERIAL_ECHO(4)
          if axis = e SERIAL_ECHO(8)

Also does any body has an idea If I wish to add this feature by myself, where is the function that actually steps the stepper motor (writes high then low to the Step pin of the driver) in the firmware.

Thanks.

AnHardt, 2019-06-26:
It's in the stepper interrupt. Printing there is a bad idea.

ManuelMcLure, 2019-06-26:
If what you want is to know where the printhead is, it might be better to go the other way - command the printer to the position you want and then use M400 to wait until the printer has completed all queued moves before executing whatever scanning you want to do at that position.

HassanSaied, 2019-06-27:
Waiting for the head to be in position will either cause my xy resolution (minimum distinguishable distance) either very high, or cause the scanning process to take lots of time, also alignment errors could happen.
But If I probed the step pin, or even write its value to an unused pin on RAMPS then use something faster than an arduino like a node MCU to monitor the changes in these pins, then I will be able to monitor the steps without slowing down the interrupt

AnHardt, 2019-06-27:
Hardware counter on the step pins - scanning every n steps?

HassanSaied, 2019-06-27:
Not every n, scan every step. However, I tried to SERIAL_ECHO inside the interrupt code and it didn't seem to have broken any thing, yet I still need some End marker to know that the motion has ended, and ok isn't always sent. Where should I put this end marker.

AnHardt, 2019-06-27:
Can't remember how this was exactly done in 1.1.7 and not willing to look that up. New code should be directed to the bugfix-2.0.2 branch. Otherwise there is **NO** chance to integrate it here.

However. In a Bresenham line algorithm there is always a counter, counting from 0 to the number of steps of the largest distance axis in steps (May also count backwards to 0). When that number is reached the block has finished and the next block is attempted to begin. In bugfix-2.0.x about [here](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/stepper.cpp#L1539).

AnHardt, 2019-06-27:
'ok' is always sent. But when the move is completely planned - not when the steppers stop to step. In between there is the planer block buffer. That is needed to join the single moves to a smoother motion.

boelle, 2019-07-21:
@HassanSaied could you edit the title? [FR] (feature request title) is not a good description

phearl3ss1, 2020-12-10:
I have s Scara that came from a University surplus auction it uses 90v Dc motors with optical encoders.  (Sumtax Optucoder  JBT-048)2048 s/r )  It has the A B and Z sig  going into a driver board along with step and direction inputs from a main controller.   It also has a- and b- that appear to have been wired into the main controller (that did not come with the Scara).      I am guessing the out of phase sig was used  as feedback to confirm or trigger the next step.

thisiskeithb, 2023-07-19:
This has been implemented under `M114` with `M114_REALTIME`:

https://github.com/MarlinFirmware/Marlin/blob/885e9cc0576af2c6a79d8bfa9a4f1560c6748b27/Marlin/Configuration_adv.h#L3812-L3815

closed 2023-07-19 by thisiskeithb