MarlinFirmware/Marlin #14417: 2.0, STM32F1R compilation with ArduinoIDE

paukstelis, 2019-06-27:
I thought I would just play around with putting Marlin on a basic STM32 Blue Pill using the basic STM32F1R profile. Using ArduinoIDE (1.8.5 I think) and latest STM32_Arduino hardware support (1.6.0).

The first couple of problems I run into is with pins_STM32F1R: It seems the definition from the STM32 Arduino libraries defines STM32F1xx instead of __STM32F1__.

Modifying that, it will begin to compile, but then basically all the pin definitions are wrong; the STM32_Arduino library seems to now want underscores between port and pin numbers:

> sketch/src/inc/../pins/pins_STM32F1R.h:50:28: note: suggested alternative: 'PC_0'
>  #define X_STEP_PIN         PC0

Making all these changes for the pins, along with switching SERIAL_PORT in  Config.h to -1 will get everything to compile. However, after flashing it to Blue Pill (with DFU bootloader) I can't get anything through USBSerial. I've tested with just a basic sketch (just using Serial.begin etc.) and serial works fine, so I'm not quite sure if this is a problem or not having anything connected to the reset of pins is preventing it from booting up.

Any suggestions on how to proceed with this?

AnHardt, 2019-06-27:
For the STM32 processors there are at least 3 frequently used Arduino distributions.

(1) By [rogerclarkmelbourne](https://github.com/rogerclarkmelbourne/Arduino_STM32) based on libmaple. Supporting mainly STM32F1. Rather old and complete, but don't expect much further development here.

(2) By [danieleff](https://github.com/danieleff/STM32GENERIC) based on STM32_HAL. Much better support for STM32F4 then the above, but, also here, don't expect much further development.

(3) By [stm32duino](https://github.com/stm32duino/Arduino_Core_STM32) based on the STM32_HAL with direct access to STM32_LL, supporting, more or less, the complete pallet of STM32 processors. It's currently a bit behind the others, supports less ready Arduino libraries, but is currently the only one actively developed. This is done by a ST employee. This seems to be the future.

Marlin, more or less supports all of them by different HALs.
For what board-file is made for what HAL take a look into [platformio.ini](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/platformio.ini)
Compare the `src_filter` field of the STM32 based boards. If you find: 
* <src/HAL/HAL_STM32F1> it's for (1)
* <src/HAL/HAL_STM32F4> it's for (2)
* <src/HAL/HAL_STM32> it's for (3)

Looks as if you are trying to compile files for (1) with (3).

paukstelis, 2019-06-27:
Thanks for the response. I did recognize that there were different hardware setups. I've tried them all at various points. I can get things to compile with **3** and that does use HAL_STM32 (as you've defined). I still have the issue there that Serial doesn't work. I don't know if there are strange things going on with defintions, but when I use **1** (which should use STMF32F1) the HAL definition still goes to HAL_STM32 and throws an error:

> sketch/src/HAL/HAL_STM32/watchdog_STM32.cpp:30:25: fatal error: IWatchdog.h: No such file or directory
>    #include <IWatchdog.h>

xC0000005, 2019-06-27:
What board and part number did you select?

paukstelis, 2019-06-27:
> What board and part number did you select?

When trying to use STM32duino I tried multiple. Same watchdog error. 

> Error compiling for board Generic STM32F103R series.
> Error compiling for board Generic STM32F103C series

I've selected the 128K flash version when trying these.

xC0000005, 2019-06-27:
If you try to compile using PIO and STM32DUINO, make sure HAL_STM32 is in the - src filters so it’s not compiled. I’m not sure you can compile for STM32DUINO using the arduino IDE anymore - the only board manager I show is ST’ls core (which is a different HAL). The STM32F1R pinout uses some pins that I don’t see in the bluepill variant, which makes me think @victorpv was using some slightly different board.

paukstelis, 2019-06-27:
Thanks. I am getting closer. I started just working through the various pre-defined STM32 boards. I was able to get everything working with the MAYLANM200 profile. I can make pin modifications I need from there.

closed 2019-06-27 by paukstelis