MarlinFirmware/Marlin #14420: [BUG] tmc5160 Reporting Multiple Failures

KerseyFabrications, 2019-06-27:
### Description

I'm running a mixed stepper printer, X,Z = tmc2208, Y = tmc5160, E = A4988.
Before the recent updates, probably a month ago or longer, I had no problems running in the mode. With the tip of the tree though, I was having serial issues as well as now my tmc5160 is reporting all kinds of errors, including:
00:00: Y driver overtemperature warning! (1000mA)

Y driver error detected: 0xDF800000
overtemperature
coil short circuit

Since this driver was working fine prior to the updates, I think this is a software issue, not hardware. Here is my full output:
```
00:00: Y driver overtemperature warning! (1000mA)

Y driver error detected: 0xDF800000
overtemperature
coil short circuit
                X       Y       Z
Enabled         false   false   false
Set current     800     1000    800
RMS current     795     990     795
MAX current     1121    1396    1121
Run current     25/31   19/31   25/31
Hold current    12/31   9/31    12/31
Global scaler           133/256
CS actual               12/31   0/31    12/31
PWM scale       14      4026531840      14
vsense          1=.18           1=.18
stealthChop     true    false   true
msteps          16      0       16
tstep           max     536870912       max
pwm
threshold               79      0       658
[mm/s]          100     -       3
OT prewarn      false   true    false
OT prewarn has
been triggered  false   true    false
off time                3       0       3
blank time      24      16      24
hysteresis
-end            -1      -3      -1
-start          1       1       1
Stallguard thrs         0
DRVSTATUS       X       Y       Z
stallguard                      *
sg_result                       0
fsactive
stst            *       *       *
olb                     *
ola                     *
s2gb                    *
s2ga
otpw                    *
ot                      *
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
                X       0xC0:0C:00:00
                Y       0xEF:C0:00:00
                Z       0xC0:0C:00:00

Error:Printer halted. kill() called!
```

### Steps to Reproduce

1. Edit configuration as shown.
2. Reboot the system.
3. See error.

**Expected behavior:**
No errors.

**Actual behavior:**
Errors as shown.

#### Additional Information

Board: SKR V1.3
Driver: TMC5160-V1.0

teemuatlut, 2019-06-27:
With the limited time I have to debug, it would really help to narrow down on the version or even better the commit that changed the behavior.

Augustus22, 2019-06-27:
I understand and I've been working on that. I'll keep on it over the weekend and see what I can find. I wanted to submit the bug for reference.

KerseyFabrications, 2019-06-27:
I understand and I've been working on that. I'll keep on it over the weekend and see what I can find. I wanted to submit the bug for reference.

KerseyFabrications, 2019-06-27:
For reference, I'm running commit 2580104c8befb7bb6a363b79776f4a8ba9436686 at this time.

3sr3v3r, 2019-06-29:
Have you removed the clk pin and bidged clk to ground on the TMC5160? I'm running the same drivers on an skr1.3 (on all axis and dual extruders) and had all sorts of similar errors until I made the modifications.

KerseyFabrications, 2019-07-02:
@3sr3v3r Can you explain why this mod is necessary before I mod my board?

thisiskeithb, 2019-07-02:
Details are in the [datasheet](https://github.com/watterott/SilentStepStick/blob/master/hardware/SilentStepStick-TMC5160_v13.pdf). I had to make the same mod on my genuine Watterott 5160s.

The good news is that the next 5160 rev (at least from Watterott) will have a solder jumper so we won’t have to use a physical wire.

closed 2019-07-11 by thinkyhead