MarlinFirmware/Marlin #14421: [BUG] Function doesn't use its parameter (bugfix-2.0.x)

tigertooth4, 2019-06-27:
### Function doesn't use it's parameter

Hi, 

I'm going to connect an SH1106 OLED display to the Cohesion3d-Remix (LPC1769 based) board via i2c. I think there might be some minor issues in HAL_LPC1768>u8g>HAL_LCD_I2C_routines.c 

### Problems

1. In around line 90: uint8_t u8g_i2c_start(uint8_t sla) { ...}. This function doesn't use its parameter (sla). Variable sla should be used in line 103.
 
2. In lines 102-105:   LPC_I2C1->I2CONCLR = I2C_I2CONCLR_STAC; ... Notice that there are 3 i2c interfaces, namely LPC-I2C0, LPC_I2C1, and LPC_I2C2. It appears that user have the ability to choose which interface is going to use, by setting the value of  USEDI2CDEV_M (to 0, 1, or 2). However, in lines 102-105, the interface has been fixed to LPC_I2C1，which seems incorrect.  

3. The same issue appears in lines 157--159.

boelle, 2019-09-24:
@tigertooth4 is the issue still there?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle