MarlinFirmware/Marlin #14422: [FR] New board support

GMagician, 2019-06-27:
What has to be the minimum "feature" supported by a new HAL to be inserted in Marlin official list?

- [x] ADC
- [ ] PWM
- [x] DIO
- [ ] SD
- [ ] LCD
- [x] EEPROM
- [x] TIMERS
- [x] SERIAL
- [x] WDT

AnHardt, 2019-06-27:
I disagree with the feature centric view of this. I prefer a functionality centric view. See a proposal for details in https://github.com/MarlinFirmware/Marlin/issues/14345#issuecomment-504675913

GMagician, 2019-06-27:
Well, some of the things reported in such comment are not really related to the HAL itself

smooth movements, stepper drivers...

And also EEPROM, LCD, SD and USB are not necessary for everyone (SD or USB are mandatory but others?)

Of course I'm far from having something that works but the answer to my question will tell me when the code I have may be posted here to be "mass" debugged..

AnHardt, 2019-06-28:
I hoped to list needed functions of a HAL/board to be useful and alternative ways to achieve them.  

> And also EEPROM, LCD, SD and USB are not necessary for everyone (SD or USB are mandatory but others?)
For example:
To print anything you need either 
a host connection to receive g-code
or alternative
a way to read g-code files and to select them.
Better both, but at least one.

And it does not matter how the host connection is realized. It could be a native serial or a USB serial or a Bluetooth serial or a network telnet session or something else or many of them, but at least one.
And it does not matter on what media the files are stored. It could be a SPI connected SD-card or a SDIO connected SD-card or a USB stick or a harddisk or something else or many of them, but at least one.

AnHardt, 2019-06-28:
Or the other way around.

ADC support is not necessarily needed if the functionality of reading temperatures is realized by reading the MAX sensors over SPI. (But the MAXs should be on board then)
EEPROM support is not necessarily needed if the functionality of configuration storage is realized on SD or ...
TIMER support could not be required if the functionality of smooth stepping and scheduling tasks and   heating, and ... is reached by an other mechanisms. (That's difficult)

GMagician, 2019-06-28:
LOL, that's right!

GMagician, 2019-06-28:
Some questions:
1) What is the reason to have serial -1?
2) SAMD51 has an smarteeprom emulation integrated (it uses its flash memory in a smart way). Easy to integrate as config storage area but...it's not enabled by default and it require to burn some fuses to define its size. What's the best:
    a) let user install a custom bootloader that program fuses
    b) integrate fuses burning in HAL when e2p is enabled and no external  e2p is defined
    c) create an additional sketch to program it?
Considering that at the moment an external spi/i2c e2p is also supported (but not tested, some shields have it)

AnHardt, 2019-06-28:
> What is the reason to have serial -1?

That's best explained at a DUE. That has 2 USB connectors. The first (Programming Port) is connected to a on board USB chip - and that is connected to a native serial port (1) of the processor. The other is connected to a nativ USB port of the processor configured as a serial USB connection (-1). 
Freeing the processors hardware serial from the duty to communicate with the host makes it available for other purposes like a display, a TMC, extra debugging, ... you name it.

EEPROM Simulation in PROM?
That's a matter of taste. 
**I** prefer to use a external EEPROM if available.
Second would be c), when it's a one time thing and the same sketch can restore the original state.
Third would be b). That's most confortable, is carrying (probably) one time used code thru the lifetime of Marlin.
Least preferable to me seems to be a). Even if mostly a good solution most users are scared by unfamiliar bootloaders. They are afraid to brick their boards.

GMagician, 2019-06-28:
So, if I correctly understand, positive numbers are for communication through real serial pins (serial on chip), while negative (only -1 at the moment) is for usb decicated pins...
On AGCM4 there is only one usb connector and is connected directly to some dedicated D+/- pins hence -1 is correct number to assign to it...thanks

GMagician, 2019-06-30:
> I prefer to use a external EEPROM if available.

if I2C_EPROM/SPI_EPROM is define external eeprom is used

> Second would be c), when it's a one time thing and the same sketch can restore the original state.

Yes,  once programmed it's kept. I made a sketch (needs to be tested), where do you think it's better to store it? inside Marlin folders tree, where? somewhere else in github?

GMagician, 2019-07-01:
Development is proceeding, I 'll post a PR when something is testable

closed 2019-07-01 by GMagician

ETE-Design, 2019-12-23:
@GMagician Any news on this?

GMagician, 2019-12-23:
@ETE-Design 
Theoretically LCD, SD and eeprom has been added. Not yet done a really good test because of missing of shield board.
I'm in contact to @mralvin to get a ramps 1.4.4 to be able to consolidate developping.

GMagician, 2019-12-23:
@ETE-Design please be aware that there is a bug in Adafruit usb implementation that forget to send some bytes (aka not usable with repetier).
I already posted a fix to their repository that has been merged on 8th August, but they did not release an official release yet.

GMagician, 2019-12-30:
@ETE-Design Adafruit has just released framework 1.5.8. It should include my serial fix.
Once PIO will update its framework then USB will works has expected