MarlinFirmware/Marlin #14423: [FR] TMC2130 Hybrid Threshold with acceleration/jerk

Jmlannan, 2019-06-27:
Currently hybrid threshold can only be set with a max speed, but speed doesn't require more torque from the motors, its high acceleration and jerk that need it. 

I found that my printer could lose steps in stealth-chop mode at low speeds with high jerk and acceleration settings. 

Being able to set a hybrid threshold acceleration and jerk would allow for spread cycle only when the printer is making a quick move and allow it to remain quiet while moving quickly. Or allow the printer to remain in stealth-chop with low acceleration/jerk and high velocity settings.

AnHardt, 2019-06-27:
> but speed doesn't require more torque from the motors, its high acceleration and jerk that need it.

That's **about** right, but not the point.

The stepper motors do not need more torque to move fast, but the stepper drivers do need more power to keep the motors at the same torque at high step rates.

A brief and very simplified explanation:
Torque of stepper motors depends on the current thru the coils.
The coils are inductive devices. Once the current flows it wants to flow into the same direction even if tried to switch of.
You could eiter wait until the current stops due to internal resistance, or you can apply a opposite polarity voltage to stop the current faster.
When a voltage is applied to a inductive device it needs some time until the current has its full strength.

Switching a coil from one polarity to the other needs time. After that the full wanted current can flow until the step ends and the polarity changes again.
The time to change the polarity for a given current with a given voltage is constant.

When the step-rate increases the time of this cycle decreases.
When the time for a step decreases the time of the full current decreases.
When the time for one step becomes shorter than the time for changing the polarity the full current is not reached anymore - the torque of the stepper motor decreases.

See **any** literature about steppers and stepper drivers.

Jmlannan, 2019-06-28:
Yes the amount of torque may be besides the point. Step loss is most likely to occur while accelerating. In stealthchop the driver can't detect missed steps. Spreadcycle is still great for high speeds, but it would also be helpful at high jerks/accelerations for detecting step loss. It may not provide more torque, but it still provides utility.