MarlinFirmware/Marlin #14424: [BUG] 3DTouch on SKR 1.3 dont responding

przemeklev, 2019-06-27:
Hey

I have some problem with my 3dtouch and SKR 1.3. When I power on my printer 3dtouch get deploy and stow. Its all what i can do with my sensor. It doesn't react M280 code and LCD options.

I'm 100% sure to correct connection. 

I need some help.

shitcreek, 2019-06-27:
Without providing much information, we cannot know what is going wrong. Are you sure you've configured it properly?

That said, this Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on this page.) For best results getting help with configuration and troubleshooting, please use the following resources:

RepRap.org Marlin Forum
Tom's 3D Forums
Facebook Group "Marlin Firmware"
Facebook Group "Marlin Firmware for 3D Printers"
Marlin Configuration on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

przemeklev, 2019-06-27:
i check many resources and do everything like change softweare , add capicitor , try diffrent wire. and now I dont know what can i do.  Thanks for your help

shitcreek, 2019-06-27:
Have you configured BLTouch in Marlin? ie #define BLTOUCH
I could take a look at your config files.

Gerggg, 2019-07-02:
I also have problems with a BLtouch, from time to time, the printer tells me he does not respond. the only way to recover it is to reboot the printer. while before going into Marlin 2.0 I had no problem.

przemeklev, 2019-07-04:
My skr is reset every time when I change something. Maybe u talk about diffrent boot

shitcreek, 2019-07-05:
@charlesmike hello

boelle, 2019-07-21:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

john4242, 2019-09-22:
For anyone having problems with BLTouch or clone, enable:
`#define PINS_DEBUGGING`
Then you can issue a test command:
`M43 S`
To test the BLTouch. Beware, some clones only emit a short, 5ms signal, that is okay, it will still work, but this debug command will suggest otherwise. Ignore that "noise" warning.

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle