MarlinFirmware/Marlin #14426: Bad traveled distance

Kowy54, 2019-06-27:
Today i´ve configured marlin 2.0 for skr 1.3 and 2209. everything is working. I have 1/256 microstepping on X axis and calculated axis_steps_per_unit for this value (1600). If i hit move x to 100, real traveled distance is about 25mm. Any help?

shitcreek, 2019-06-27:
Seems like you need to reduce your steps to a 4th of that. Most people set theirs with 1/16 or 1/32 micro stepping and the trinamics would interpolate that on the fly. Follow this calculator if you haven't already: https://blog.prusaprinters.org/calculator/

That said, this Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on this page.) For best results getting help with configuration and troubleshooting, please use the following resources:

RepRap.org Marlin Forum
Tom's 3D Forums
Facebook Group "Marlin Firmware"
Facebook Group "Marlin Firmware for 3D Printers"
Marlin Configuration on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

thinkyhead, 2019-06-28:
TMC steppers don't do 256 real micro-steps. You configure them to do "16" micro-steps and that is what you should set in your configuration. The TMC stepper driver transparently interpolates the 16 micro-step moves you give it to 256 micro-steps.

closed 2019-06-28 by thinkyhead