MarlinFirmware/Marlin #14427: [BUG] Printer pauses starting a print (2.0 bugfix)

blueeagle69, 2019-06-27:
When starting a print from SD Card (https://github.com/MarlinFirmware/Marlin/issues/6164)

Printer pauses immediately when starting an SD card print, and won't start until I hit resume.

Print from SD Card

**Expected behavior:** Print should start

**Actual behavior:** Printer pauses and waits for me to select Resume Print. Also when I am in the new Print dialogue screen (2.0 bugfix), the encoder on my Smart Discount Graphic LCD is backwards despite going the correct way everywhere else.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2019-06-28:
The pause bug may have been reintroduced in https://github.com/MarlinFirmware/Marlin/commit/ce656e2a268a120ccef77a7d80cc5c1c4fcc7f7e. Does Marlin from https://github.com/MarlinFirmware/Marlin/commit/decfe3424fa4975ed9315feaa98ad7b56e719f4d work? [Download link](https://github.com/MarlinFirmware/Marlin/archive/decfe3424fa4975ed9315feaa98ad7b56e719f4d.zip).

You may need to comment/uncomment `#define REVERSE_MENU_DIRECTION` in` Configuration.h`.

Posting your configs would help troubleshoot further.

InsanityAutomation, 2019-06-28:
@thisiskeithb with whats there I dont see how that could possibly contribute to this kind of bug. A few timer lines in a rarely used function and the z probe end script subcommand processing wouldnt touch anything near this.

reloxx13, 2019-06-28:
post the gcode file to be sure there is no pause comand in it

JimGWIn, 2019-06-28:
I am also having this issue. I have to click once to activate the menu and then click resume.

[Z_endstop_adj v1_SSR_Cover_.txt](https://github.com/MarlinFirmware/Marlin/files/3339731/Z_endstop_adj.v1_SSR_Cover_.txt)

blueeagle69, 2019-06-28:
Here's my gcode, if you need it.
It's not the gcode, it was fine on 1.1.9

[Calibration Cube.zip](https://github.com/MarlinFirmware/Marlin/files/3340926/Calibration.Cube.zip)

blueeagle69, 2019-06-28:
With https://github.com/MarlinFirmware/Marlin/commit/decfe3424fa4975ed9315feaa98ad7b56e719f4d it appears that the pausing before a print has been fixed.
However the reversed encoder options still persist in the new Start Print dialogue.
If I un-comment #define REVERSE_MENU_DIRECTION then the Start Print dialogue is now correct, but the main menus are backwards.
If also have #define REVERSE_ENCODER_DIRECTION un-commented, because if I don't then value changing is backwards.
I'll just live with the Start Print dialog etc. backwards.
It's no big thing tbh.

blueeagle69, 2019-07-23:
Okay scratch that, there is still a random pause before starting an SD card print with  decfe34.
I have Advanced Pause enabled. The printer randomly triggers it after starting a print.
Weirdly though, it doesn't do it all the time. Sometimes it will start a print as expected. Other times it will trigger the advanced pause and eject my filament.

boelle, 2019-09-24:
@blueeagle69 is the issue still there?

JimGWIn, 2019-09-24:
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/MarlinFirmware/Marlin/issues/14427?email_source=notifications\u0026email_token=AMPGZMGPCIDPX2LND4HSFXTQLIMI7A5CNFSM4H4A7GKKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD7OOPNA#issuecomment-534570932",
> "url": "https://github.com/MarlinFirmware/Marlin/issues/14427?email_source=notifications\u0026email_token=AMPGZMGPCIDPX2LND4HSFXTQLIMI7A5CNFSM4H4A7GKKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD7OOPNA#issuecomment-534570932",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> }
> ]

No the issue is solved Sent from my Samsung Galaxy smartphone.

boelle, 2019-09-24:
i will close then, we can always reopen

closed 2019-09-24 by boelle