MarlinFirmware/Marlin #14431: [BUG] I2C OLED not working on LPC1768

istyszy, 2019-06-28:
Hello,

I have found this topic on the bugs list and it is closed, but the problem still exists.

[image]

I have an i2c OLED display, with marlin 2.0 (2019-06-24) + SKR V1.3, i use hardware i2c from E1. When the board starts everything is OK, display works as it should, but after 40 sec or so, the display crashes.
Did someone had this problem? I used the display with ramps+mega and never had any issues? I use u8glib_SH1160. (also tried U8GLIB_SSD1306 same result).
My diy display: https://reprap.org/forum/read.php?13,499572,500608#msg-500608

I have tried all the solutions on https://github.com/MarlinFirmware/Marlin/issues/13257
even with a 3.3V regulator.
the problem remains. display artifacts after some time.

tigertooth4, 2019-06-28:
I have exactly the same problem, connecting a sh1106 i2c oled display to the cohesion3d-remix board. The display is connected to i2c0 though.

istyszy, 2019-06-28:
I tried with 5V and 3.3V same result

istyszy, 2019-06-28:
i think i am missing the pull-up resistors for the I2C bus?

Rundercaster, 2019-06-28:
Just want to add to this, the screen is still responsive but just layer shifted (i have the same issue)

istyszy, 2019-06-28:
So problem solved, added 4.7k resistors for clock and data line( the board was not designed to use I2C on these pins so it has no pull-up resistors).

thinkyhead, 2019-07-03:
Thank you for the update! That will be useful troubleshooting information.

istyszy, 2019-07-03:
> Thank you for the update! That will be useful troubleshooting information.

The issue was closed too soon, the problem still persists but now it appears after 4 minutes or so.
i have no tools for data logging. Maybe you could recommend something affordable,

thinkyhead, 2019-07-04:
i2c is always fun to wire up. You do need a resistor to ground on the line.

[image]

thinkyhead, 2019-07-04:
Assuming the circuit is right, the next thing to look at is the quality and length of the wires, plus interference from nearby. If you have some ferrite cores, put them on the LCD leads to reduce induction. I'll take a look and see if there's an easy way to change the i2c data rate for the display. Perhaps a slightly slower transfer rate would be more reliable. You might try introducing noise intentionally to see if it induces the same artifacts. Put the wires very close to a spinning servo, or something…

Rundercaster, 2019-07-04:
I don't think it's hardware issues, as it doesn't randomly shift, or continues to shift further, also I tried 10k resistor for both lines pull up and tried pull down, didn't work for both. It's just that one line shift and it stays there forever until restart

(I didn't say anything because I thought resistors have solved the problem for istyszy)

thinkyhead, 2019-07-04:
I'm not sure exactly where to look. I'll poke around the HAL i2c code, if any, for a few minutes…
Meanwhile, here are some other options you can try just for fun:
```cpp
#define OLED_PANEL_TINYBOY2
```
```cpp
#define MKS_12864OLED
```
```cpp
#define MKS_12864OLED_SSD1306
```
```cpp
#define ULTI_CONTROLLER
```

thinkyhead, 2019-07-04:
There are some comments about timings, commented-out defines, etc. in these files that you might want to try messing around with to see if it has *any* effect (even negative) and maybe something will emerge from that.
[image]

istyszy, 2019-07-04:
Thanks for the info, i will play around.

istyszy, 2019-07-04:
LPC_I2C1->I2DAT = I2CDEV_S_ADDR & I2C_I2DAT_BITMASK
i cand find the definition / value for : I2C_I2DAT_BITMASK

tigertooth4, 2019-07-05:
> > Thank you for the update! That will be useful troubleshooting information.
> 
> The issue was closed too soon, the problem still persists but now it appears after 4 minutes or so.
> i have no tools for data logging. Maybe you could recommend something affordable,

Yes, I can confirm that problem persists. I've tried 10K and 5K pull up resistors, still have the problem.

thinkyhead, 2019-07-06:
Google says that CMSIS (core headers) define `I2C_I2DAT_BITMASK`.

tigertooth4, 2019-07-08:
I tried change the I2C_TIMEOUT today. When the value was setting to 1, the screen went more corrupted, for I2C_TIMEOUT ≥ 3， no essential improvements, the screen behaves just like description before.

istyszy, 2019-07-08:
> Google says that CMSIS (core headers) define `I2C_I2DAT_BITMASK`.
i ordered a  Logic Analyzer i have to see what is going wrong. 
There is something strange with this anomaly. 
i have tried frrite core to on the wire and makes no difference.  
Untill my equipment arrives i will play around with my 2004 LCD and rotary encoder( i got it working this way, just a temporary solution)

tigertooth4, 2019-07-12:
Today I tried disable the status screen and menu by comment out the contents of u8g main loop in ultralcd.cpp, 
[image]
by doing so, only the splash screen was showed. And there was no screen crashing then. 

Is this an indication of a u8g loop problem instead of i2c routines?

Rundercaster, 2019-07-12:
If you disable the status screen, it only sends 1 frame at the beginning and it doesn't update it after.
So technically you just cant see the line shift, because the screen have 1 frame and it is displayed indefinitely

istyszy, 2019-07-16:
@thinkyhead So, status update:
If i use the the source from: https://github.com/obertr0n/Marlin/tree/marlin_2.0.x_ssd1306/Marlin , the Oled display works perfectly, so i made the changes

Affected files:
HAL_LCD_I2C_routines.c
u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp

to a new version of of marlin... the problem appars
so i guess the problem is from somewhere else?

tigertooth4, 2019-07-17:
> @thinkyhead So, status update:
> If i use the the source from: https://github.com/obertr0n/Marlin/tree/marlin_2.0.x_ssd1306/Marlin , the Oled display works perfectly, so i made the changes
>
> Affected [...]

Great! 

Check this out: in u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp，there are two more lines in the obertr0n version of marlin: 
[image]
may be the reason. I've added these two lines to my marlin build, the display is ok for some time.

tigertooth4, 2019-07-18:
> @thinkyhead So, status update:
> If i use the the source from: https://github.com/obertr0n/Marlin/tree/marlin_2.0.x_ssd1306/Marlin , the Oled display works perfectly, so i made the changes
> Affected [...]

I see a lot changes in u8g_dev_ssd1306_sh1106_128x64_I2C.cpp file.

> 
> Great!
> 
> Check this out: in u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp，there are two more lines in the obertr0n version of marlin:
> [image]
> I've added these two lines to my marlin build (my marlin bugfix 2.0.x was forked in April, it's 508 commits behind the main stream now), the display has no problem since then.

istyszy, 2019-07-18:
I will take a look

istyszy, 2019-07-19:
u8g_dev_ssd1306_sh1106_128x64_I2C.cpp
I compared to the latest version of marlin and they are identical.

tigertooth4, 2019-07-19:
The change history to the u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp file can be find:

https://github.com/MarlinFirmware/Marlin/commits/bugfix-2.0.x/Marlin/src/lcd/dogm/u8g_dev_ssd1306_sh1106_128x64_I2C.cpp

it shows the changes happened in commits on May 7, 2019. My copy and I believe obertr0n's copy of Marlin was this commits. 

What's happens to me is that according to the hints made by obertr0n's version, I added two lines to the u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp as I showed above, I found the display became normal.

And I found some inconsistency exists before and after the commits on May 7,2019, in the u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.  For example the definition of macro
SH1106_START_LINE(N) provides a value (0x40 | (0)) when N = 0, while according to the previous version, this value should be 0x40.

I'm not sure if this leads to the crash. I may try substitute this file with the previous version or substitute in other way, to see if everything is fine or not.

istyszy, 2019-07-22:
@tigertooth4 
- u8g_dev_ssd1306_sh1106_128x64_I2C.cpp -> Identical
- HAL_LCD_I2C_routines.c -> Coped from obertr0n's version
- u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp -> Coped from obertr0n's version
The modifications in theese 2 files fixes the LCD problems. 

I think this is the fix in u8g_com_HAL_LPC1768_ssd_hw_i2c : (u8g_i2c_send_byte(arg_val ? 0x40 : 0x80))

I am sorry i am just a begginer : (

boelle, 2019-09-24:
@istyszy is the issue still there?

istyszy, 2019-09-27:
@boelle can someone work on software I2c? to see if we have the same problems?

boelle, 2019-10-21:
@istyszy you never answered my question?

boelle, 2019-10-30:
since there has been no activity since 27 sep i will close this one

if multiple users can confirm this one and we have their configs we stand a better chance

we can always reopen

obertr0n, 2020-03-15:
@istyszy, @boelle, @thinkyhead, @tigertooth4 I can confirm that the issue is still present even on the bugfix branch. I don't expect many people to report an issue, because of the limited number of users of this hardware combination.

With the patch that I have made a few months ago, the display works fine, even though **I don't understand why**.
Maybe a more capable person can dig into the internals of the I2C functions for LPC1768, because from my limited experience with this platform, I couldn't tell what is the actual problem. 
I suspect some incorrect order of transmission of the data to the display chip.

@thinkyhead, to my understanding, there is no "Wire" library fully implemented for LPC platform, right? I tried to use a similar implementation to the one for STM platforms, without any luck.

KingTomaHawk, 2020-05-12:
@boelle I can also confirm that the bug is still present (using Marlin 2.0.5.3).

istyszy, 2020-05-12:
i was playing around and... the problem appears after you define the correct prins for the rotary encoder... maybe it was a coincidence?

ajanky, 2020-05-14:
@boelle I also can confirm that this one line shift is still an issue.
with no action at all it takes up to a5min to appear, after homing it comes pretty fast.

@istyszy after commenting out my encoder pins ( -1) I hoped that helped BUT after 30+ min it appeared again. I will try to dig deeper into the problem the next days...

iotricity, 2020-06-04:
The row shifting happens after a few minutes, but it can be sped up by browsing thru menu's fast. At first the shift is one row, later on the shift was even three rows (see attached image). Since the display is not corrupted it's not a timing problem. It's more an offset problem where the start row is wrong. The SKR has more RAM available so I will change the U8Glib from paged mode to single buffer to see if that changes anything.

[image]

KingTomaHawk, 2020-07-29:
Just a reminder, that the bug still persists in Marlin 2.0.6.
I forgot to mention: I'm using a SKR 1.4 Turbo board.

sjasonsmith, 2020-07-29:
Can someone point to the hardware needed to reproduce this? I’m n it familiar with display and clearly sharing how to obtain it might help open it up to other people to debug.

KingTomaHawk, 2020-07-29:
@sjasonsmith My setup is a little bit different as I use the OLED Controller of a Wanhao Duplicator 6. 

But the OLED used is a SSD1306.

KingTomaHawk, 2020-07-30:
I found a thread in the reprap.org forum where they are talking about the same problem. Also there is a possible fix, where you have to uncomment all commented code in the Marlin/src/HAL/LPC1768/u8g/u8g_com_HAL_LPC1768_ssd_hw_i2c.ccp file. 
It didn't worked for me, but it maybe it works for one of you.

https://reprap.org/forum/read.php?13,499572,page=7
(Post by user enif posted April 26, 2020 03:27PM)

Arakon, 2020-08-10:
Getting this same issue, on a SKR 1.4 Turbo using the latest Marlin 2.0.6 stable version.

boelle, 2020-08-10:
> the latest Marlin 2.0.6 stable version.

and with  [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) ?

Arakon, 2020-08-11:
> 
> 
> > the latest Marlin 2.0.6 stable version.
> 
> and with [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) ?

Also happening with the latest bugfix revision, just tested.

Arakon, 2020-08-15:
> I found a thread in the reprap.org forum where they are talking about the same problem. Also there is a possible fix, where you have to uncomment all commented code in the [...]

This doesn't work for me, btw. Same with adding 4.7k pullups to the signal pins to reduce possible interference, no difference at all.

Arakon, 2020-08-16:
> With the patch that I have made a few months ago, the display works fine, even though **I don't understand why**.
> Maybe a more capable person can dig into the internals of the I2C functions for [...]

I tried your modifications adapted to the latest bugfix branch and it STILL doesn't work for me.. behavior is completely unchanged.

iotricity, 2020-08-17:
The I2C communication is correct. Even when the display shows the offset info. Somehow the offset of the buffer or a register is set wrong at some point in time. If timing for the I2C bus was wrong, data would be corrupted on the display and the next frame would be shown correctly since it's a full buffer transfer for each new frame.

Arakon, 2020-08-31:
To keep this issue updated: Same issue remains on 2.0.6.1.

KingTomaHawk, 2020-10-10:
FYI: I tested 2.0.7 a few days ago and the issue still exists.

atoomnetmarc, 2020-10-23:
> 
> 
> FYI: I tested 2.0.7 a few days ago and the issue still exists.

+1, also on 2.0.7
Printing works fine, even for >12hr jobs. I am able to reset the row shift by entering the menu. After returning to the status screen the row shift will happen within a minute or so. Navigating the menu will row shift within a few scrolls.

[image]

This is a ULTI_CONTROLLER with a diy LPC1768 board (which is similar to the SKR1.4).
FWIW: [Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5428668/Configuration.zip)

yet-another-average-joe, 2020-11-28:
*MAYBE* I found a solution.

I have this well known issue with a SKR 1.4 Turbo I received 3 days ago, and the known good OLEDs I have been playing with for a couple years (using u8g2, without a problem).

And I also have 2 other problems...

Issue 1 : an horizontally shifted display depending on the OLED (0.96" or 1.3"). Easily solved by editing u8g_dev_ssd1306_128x64_data_start in u8g_dev_ssd1306_128x64.c

Issue 2 : the vertical shifted display after 5mn or so : solved with a ferrite (see picture). Wires are 300mm long, the ferrite comes from the salvaged parts bin (probably a hallogen dimmer). Better if near to the OLED. Adding pullups didn't help at all (I²C is already pulled up onboard). Therefore it is probably hardware (wiring) related.

Issue 3 : the whole thing still randomly reboots from time to time (with no reproductible pattern)..

The scope shows a data line with a weird signal at the begining of every frame (? - not 100% sure), but I don't know how to interpret it. If it's an issue, it is most likely software related (or at MC level)...

[image]

yet-another-average-joe, 2020-11-29:
I gave up with the OLED and hooked up an EEPROM module : daisychained with 150mm Dupont cables from the OLED to the EEPROM. Total I²C length = 450mm between the EEPROM and the board, the OLED in between. It's been 3H30 hours since it was done. Since then, the display never shifted ! (the ferrite is still there, I will remove it at some point). I don't give up anymore :)

Crashes/reboots still occur from time to time (last 1H30 without).

What's happening is unclear. I've been playing with these OLEDs hooked to STM32s and Arduinos  with 600mm (2x 300 Dupont) lying among all sorts of cables, including (twisted !) RS485 (total 6 populated breadboards) and I²C LCD, without an issue. 

This EEPROM has 10k pullups. Therefore, the bus is now terminated with 4k7 on the SKR, and 10k on the EEPROM.

Adding 4k7 pullups on the OLEDs had no positive effect (it was worse).
Adding an EEPROM seems to make things better (by far).

Soon, I'll test without the EEPROM but with 10k pullups instead of 4k7. Just to see. The scope was showing no difference with or without (rise time). But it seemed that with the probes, display shifting took a bit longer (10 mn vs 5mn).
Also, the scope shows that at the default speed, rise time is not negilgible (order of magnitude = half the clock pulse width). Maybe the default speed is a bit optimistic with this board ? Could someone tell me where the I²C speed could be be adjusted ?

yet-another-average-joe, 2020-12-02:
No idea if this could be related, but...

The scope shows a very short "negative" spike at the begining of the initialization sequence (yellow = SDA, blue = SCL). A few tens of nanoseconds, before the signal raises again and then falls down. And shouldn't SCL be at 3.3V when idle ? Is this normal ?

[image]

yet-another-average-joe, 2020-12-04:
Solved !

No more crashes.

No more shifting.

Just remove U8G_I2C_OPT_FAST. With this option I²C frequency is 400kHz. Seems too much or bad choice for these displays.
Without the option : 100kHz.
Verified with the scope.

The display seems a bit less responsive, but not even sure...

No idea why 400kHz causes problems. The SSD1306 datasheet tells the oscillator frequency is typically 370kHz (min = 333kHz, max 407kHz). Could there be some interference between a 400kHz I2C SDA signal and the oscillator ?

The devs should add an option in Configuration.h for I2C frequency control.

atoomnetmarc, 2020-12-04:
You are onto something. But....

I modified marlinui_DOGM.h and removed U8G_I2C_OPT_FAST from the SSD1309 section. Now the display does not shift anymore. The display is now noticeably (and understandably) slow. I this recent version https://github.com/MarlinFirmware/Marlin/tree/d17db477753067982b2cbf277609f4d66b0b57a9

However, I am now unable to print anything. The printer resets just after homing and trying to prime the nozzle. Also, when the printer is idle, I noticed a sudden reset.

After reverting the change I can print again (with a shifty display) and no sudden resets occur when idle.

iotricity, 2020-12-04:
The I2C displays I've tested with can do even 800kHz SCL/SDA on an ESP32 using bit banging code. The same displays fail on the LPC1769. If data was corrupted, it would garble the display's content and the lines would jump up and down if the next display/frame update is okay. This is not the fact. The shifting is a constant process, not a glitch. 

As far as I've discovered and monitored there is a shifting in the frame buffer offset. All I2C data is correctly clocked at 400kHz, no data errors. Maybe the display parameters change during time. I haven't been able to discover where that happens.

yet-another-average-joe, 2020-12-04:
Yes, shifting looks like the 1st frame replaces the 8th one, the 8th goes to first, and so on (AFAIR, there are 8x, 8px height frames). Looks like a rotation or a ring buffer with shifted indices.

On my side, with the 62680bb commit, I had another crash, but now it's been stable for 3 hours (no shifting no reboot). Obviously, this thing is still unable to print. But was stable for one full day. It was monday and it was full moon. The next day, after some reorganization on the breadboard, shifting was back.

On u8glib Git, there's not much information about speed. Somewhere, olikraus says Wire is poorly documented about speed. Unfortunately, he does not support u8glib anymore...

Wiring : Until now, the EEPROM and the OLED got +5V. Now it's 3.3V, and everything is pulled to 3.3V instead of 5V. Reverted to U8G_I2C_OPT_FAST. But it can take up to one hour or more... Wait and see. Preparing the last commit for flashing.

yet-another-average-joe, 2020-12-08:
Definitely solved the problem using SPI. Many OLEDs are SPI/I²C ; better use them as SPI : MKS_12864OLED_SSD1306 in Configuration.h, some editing in pins_BTT_SKR_V1_4.h, marlinui_DOGM.h and marlinui_DOGM.cpp ; done !

KingTomaHawk, 2020-12-08:
> Definitely solved the problem using SPI. Many OLEDs are SPI/I²C ; better use them as SPI : MKS_12864OLED_SSD1306 in Configuration.h, some editing in pins_BTT_SKR_V1_4.h, marlinui_DOGM.h and marlinui_DOGM.cpp ; done !

Thanks that sounds great.
Can you tell me , what exactly you changed in these files? I want to test it on my printer.

yet-another-average-joe, 2020-12-08:
Not sure it is the right place for explanations about wiring a SPI OLED ; this thread is a bug report about I²C, and it isn't really a solution !
I will shoot a short video and add the link to this post and RepRap forum when it will be done (have to verify some details and shoot one or two pics). Should be more explanative because of compilation errors that have to be fixed (MKS_12864OLED_SSD1306 is not supported by Marlin with Re-ARM). Be aware that printing from LCD-SD could lead to problems : some users have been reporting stuttering with LCD Touch displays.
Will need a couple days, but will do it !

obertr0n, 2020-12-08:
> The I2C displays I've tested with can do even 800kHz SCL/SDA on an ESP32 using bit banging code. The same displays fail on the LPC1769. If data was corrupted, it would garble the display's content [...]

@iotricity if this is the case, how comes that patching the code in [Marlin/src/HAL/LPC1768/u8g/u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp](https://github.com/obertr0n/Marlin/commit/d44e700ce3f65e111e1156c960b9f23ace9e9cd3#diff-e901e33eb2d4a1d29c2a49fe21d6a2f006bb1905b9fa4cbaa38a009de73c2c59) appears to solve the issue?

I'm not an expert on I2C nor SSD1306, but I cannot understand why that code fixes the issue. Does it avoid the corruption somehow?
I am using and 0.96' OLED with an encoder as a controller for the printer for over one year with an SKR 1.1 and there was no shifting with that workaround.

@yet-another-average-joe can you give it a try with the above mentioned file patched in your build with your hw setup?

[13 more replies]

closed 2021-02-25 by github-actions[bot]